raspberrypi/firmware #567: RPI2 multiple openmax pipelines block temperature sensor and more

xlazom00, 2016-03-17:
If I open multiple openmax pipelines I can't get temperature and whole
vcgencmd measure_temp
vcgencmd measure_volts
don't work.

I know that there is some limitation to 4 connection 
But can we increase this value?

popcornmix, 2016-03-17:
I don't think it is the openmax pipelines, but probably connections to gencmd.
omxplayer for example uses gencmd. You can only run 4 apps that connect to gencmd.

First what is the application you are running that uses openmax? Does it need gencmd? Does it call bcm_host_init() (which starts the gencmd service).?

xlazom00, 2016-03-17:
I use gst omx that call bcm_host_init
https://cgit.freedesktop.org/gstreamer/gst-omx/tree/omx/gstomx.c#n111

I use 3 omx elements(per process) it still init it just once base on openmax code in rpi userspace libraries

but then I have 3 processes 
and then I call vcgencmd measure_temp vcgencmd measure_volts 
are there any reason why you can't change this limitation ?
btw why we have this limitation to 2. processor on Pi ?

popcornmix, 2016-03-18:
Can you test this firmware?
It increases the number of clients for gencmd, dispmanx, tv and cec services from 3 to 8.
https://dl.dropboxusercontent.com/u/3669512/temp/firmware_vchis.zip

xlazom00, 2016-03-18:
It is working now.
Are you planning to put this to master ?

popcornmix, 2016-03-18:
Yes, this change should be in next rpi-update firmware update (and later in apt-get) as long as no regressions are spotted.

popcornmix, 2016-03-21:
This is in current rpi-update firmware. Okay to close?

xlazom00, 2016-03-21:
It is ok now

closed 2016-03-21 by xlazom00