raspberrypi/firmware #570: perf record freezes the system

AntonKozlov, 2016-03-22:
Hi!

I'm trying to run `perf record` on following example:
```C
#include <stdio.h>
#include <unistd.h>
#include <pthread.h>

#define THREAD_N 100
#define STACK_DEPTH_INEXACT_KBYTES 320
#define WORK_DELAY 1000

void worker_fn(int depth) {
        char payload[1024];
        payload[0] = 0xff;
        if (depth > 0) {
                worker_fn(depth - 1);
        }
        
        volatile int delay = WORK_DELAY ;
        while (--delay) {
                sleep(0);
        }
}

void *thread_worker_fn(void *arg) {
        fprintf(stderr, "Starting worker 0x%08x(%d)\n", pthread_self(), (int) arg);
        worker_fn(STACK_DEPTH_INEXACT_KBYTES);
        return NULL;
}

int main() {
        pthread_t worker;

        for (int i = 0; i < THREAD_N; ++i) {
                pthread_create(&worker, NULL, thread_worker_fn, (void *) i);
                pthread_join(worker, NULL);
        }
        
        fprintf(stderr, "Done!\n");
        return 0;
}
```
(building with `cc -std=c99   test.c   -o test -lpthread`)

It start and destroy a thread 100 times, each allocates big stack (~320Kbytes). At start each thread prints own id.
Usually, it breaks:                
 * process under perf could freeze in the middle of exection, it prints as '\<defunc\>' in `ps ax`
 * process under perf could freeze in the middle of exection, but then `ps ax` freezes after printing perf PID, but before printing application PID.
 * whole system could freeze, including raspberry console (appeared on HDMI)

With default kernel settings, there are such messages in dmesg
```
[  260.768904] perf interrupt took too long (2512 > 2500), lowering kernel.perf_event_max_sample_rate to 50000
[  260.798620] perf interrupt took too long (5037 > 5000), lowering kernel.perf_event_max_sample_rate to 25000
[  260.876395] perf interrupt took too long (10014 > 10000), lowering kernel.perf_event_max_sample_rate to 12500
[  263.350533] perf interrupt took too long (20033 > 19841), lowering kernel.perf_event_max_sample_rate to 6300
```

I tried to set `kernel.perf_event_max_sample_rate` to 5000 before trying, it doesn't help, except such messages are gone.

Sometime such messages could appear in dmesg:
```
[51905.696266] INFO: rcu_sched self-detected stall on CPU
[51905.696300]  3: (2099 ticks this GP) idle=8db/140000000000001/0 softirq=198452/198454 fqs=2100
[51905.696309]   (t=2100 jiffies g=163862 c=163861 q=224)
[51905.696323] Task dump for CPU 2:
[51905.696331] test            R running      0  8055   8053 0x00000002
[51905.696351] Task dump for CPU 3:
[51905.696357] test            R running      0  8056   8053 0x00000002
[51905.696398] [<800185e0>] (unwind_backtrace) from [<80013f48>] (show_stack+0x20/0x24)
[51905.696418] [<80013f48>] (show_stack) from [<8004fbc8>] (sched_show_task+0xb4/0x108)
[51905.696438] [<8004fbc8>] (sched_show_task) from [<80052d6c>] (dump_cpu_task+0x48/0x4c)
[51905.696458] [<80052d6c>] (dump_cpu_task) from [<800764d4>] (rcu_dump_cpu_stacks+0x9c/0xd8)
[51905.696476] [<800764d4>] (rcu_dump_cpu_stacks) from [<80079e94>] (rcu_check_callbacks+0x5e0/0x928)
[51905.696493] [<80079e94>] (rcu_check_callbacks) from [<8007ed8c>] (update_process_times+0x48/0x74)
[51905.696510] [<8007ed8c>] (update_process_times) from [<80090954>] (tick_sched_handle+0x58/0x64)
[51905.696528] [<80090954>] (tick_sched_handle) from [<800909b8>] (tick_sched_timer+0x58/0x9c)
[51905.696543] [<800909b8>] (tick_sched_timer) from [<8007f990>] (__run_hrtimer+0x88/0x2a4)
[51905.696559] [<8007f990>] (__run_hrtimer) from [<80080474>] (hrtimer_interrupt+0x130/0x2bc)
[51905.696578] [<80080474>] (hrtimer_interrupt) from [<80459828>] (arch_timer_handler_phys+0x40/0x48)
[51905.696599] [<80459828>] (arch_timer_handler_phys) from [<80072b50>] (handle_percpu_devid_irq+0x80/0x194)
[51905.696617] [<80072b50>] (handle_percpu_devid_irq) from [<8006e8d0>] (__handle_domain_irq+0x94/0xec)
[51905.696635] [<8006e8d0>] (__handle_domain_irq) from [<8000933c>] (asm_do_IRQ+0x2c/0x30)
[51905.696651] [<8000933c>] (asm_do_IRQ) from [<80578174>] (__irq_svc+0x34/0x12c)
[51905.696661] Exception stack(0xb9357dc8 to 0xb9357e10)
[51905.696673] 7dc0:                   bd3535ac 00000000 0000001f 0000001d b90eddb0 00000054
[51905.696686] 7de0: 76db5000 808240b8 bbdd1370 bd3535ac 00000049 b9357e1c b9357e20 b9357e10
[51905.696697] 7e00: 80123240 80577830 20000113 ffffffff
[51905.696715] [<80578174>] (__irq_svc) from [<80577830>] (_raw_spin_lock+0x40/0x54)
[51905.696736] [<80577830>] (_raw_spin_lock) from [<80123240>] (handle_mm_fault+0x6ec/0x10c0)
[51905.696753] [<80123240>] (handle_mm_fault) from [<80578c2c>] (do_page_fault+0x2e4/0x398)
[51905.696767] [<80578c2c>] (do_page_fault) from [<800092a8>] (do_PrefetchAbort+0x44/0xa8)
[51905.696781] [<800092a8>] (do_PrefetchAbort) from [<805787e4>] (ret_from_exception+0x0/0x1c)
[51905.696789] Exception stack(0xb9357fb0 to 0xb9357ff8) 
[51905.696800] 7fa0:                                     76ea59e0 000107c0 7656b460 7656b460
[51905.696813] 7fc0: 76ea59e0 00000000 7ea8ca88 00000152 7ea8cad4 76f0f4c0 00000000 7656ae5c
[51905.696825] 7fe0: 00020908 7656ae48 00010660 76db5230 60000010 ffffffff
```

To reproduce, please run `perf record -F 1000 --call-graph dwarf ./test`
Reducing sampling frequency seems to have no effect.
Under `fp` call-graph method issue not reproduces.

OS: Raspbian GNU/Linux 8
rpi-firmware revision: `2d22e509411a946071038758c5c03cc3536e0b75`
kernel revision: `f4b20d47d7df7927967fcd524324b145cfc9e2f9` (4.1.20-v7+ #862, obtained with rpi-source script)
perf built with: `make -f Makefile.perf NO_LIBDW_DWARF_UNWIND=1 DEBUG=1`
perf build log head:
```
Auto-detecting system features:
...                         dwarf: [ on  ]
...                         glibc: [ on  ]
...                          gtk2: [ OFF ]
...                      libaudit: [ OFF ]
...                        libbfd: [ on  ]
...                        libelf: [ on  ]
...                       libnuma: [ OFF ]
...                       libperl: [ OFF ]
...                     libpython: [ OFF ]
...                      libslang: [ OFF ]
...                     libunwind: [ on  ]
...            libdw-dwarf-unwind: [ on  ]
...                          zlib: [ on  ]
...                          lzma: [ OFF ]
```

Ruffio, 2016-07-24:
@AntonKozlov it looks like you are using a really old kernel (4.1.20), could you try to upgrade to latest and try again?

Ruffio, 2016-07-29:
@AntonKozlov have you tried to upgrade to latest 4.4.y and tested that?

AntonKozlov, 2016-07-30:
Hi, Rasmus,

No, haven't tried to update since issue created.

I was able to workaround it by reducing sampling frequency, because it
seems to hang when sampling tick occur during pthread starting.

But I'll check new verstion in next few weeks, when I return from trip.

Thanks,
Anton

AntonKozlov, 2016-08-12:
Hi!

I've checked new kernel
Linux raspberrypi 4.4.16-v7+ #900 SMP Wed Aug 10 18:02:31 BST 2016 armv7l GNU/Linux

Behavior seems changed a bit, but it still doesn't work. I'm still using next line to reproduce (it could went unnoticed)
```
perf record -F 1000 --call-graph dwarf ./test
```

It could stop responding at all. Cursor on HDMI console could remain blinking, and could not. USB keyboard could affect blinking period, but no input shown.

Or I able to kill perf and application, but on next start perf starting to complain about to big mmap parameter (4G). After severals errors shown, system hangs again.

Actually, I believe it's an issue of vanilla kernel, not of raspberry build. But I need you opinion and probably directions on upstreaming this.

JamesH65, 2017-12-21:
Tried this on 4.9.59 kernel (after installing linux-perf-4.9), and I get permission denied for the command line above, even under sudo/

This is almost certainly not an issue with Raspberry Pi itself, if it indeed still exists on 4.9, so closing.

closed 2017-12-21 by JamesH65