raspberrypi/firmware #571: Question: USB input not as modules?

marcust, 2016-03-22:
I just had one of my RPi2s Kernel panic during rpi-update, which in itself is an issue, but I disregard this as a glitch. The outcome was that the kernel didn't find its modules upon boot. 
This would have been more easily fixed if I could've just attached an USB keyboard, but apparently the USB Hid layer is compiled as a module, so the keyboard wouldn't work. 
What's the reason for that? 
Shouldn't something as basic be compiled in?

Ferroin, 2016-03-22:
While I would personally tend to agree, this gets kind of tricky as:
1. There are lots of people (myself included) who run Raspberry Pi's headless.  In such a situation, the USB-HID module is just dead code as long as the system is working correctly.
2. There are actually two sets of generic USB HID modules.  First, you have the full featured one that handles regular operating modes for most keyboards and pointing devices (mice and non-multitouch touchscreens and touchpads), and then you have a pair of special ones for the HID Boot Protocol operating mode of keyboards and mice (which is the mode that gets used by the BIOS/UEFI on PC systems for USB keyboard and mouse support).  The HID BP drivers are significantly smaller, but also support a  very limited set of features (for example, the HID BP Mouse driver doesn't support absolute mode, which means that touchscreens and most touchpads won't work with it).  Because of how Linux's driver model works, you can only have o ne set or the other active on the system at a time, which means that only one or the other can be compiled into the kernel.  This ends up partly reinforcing item 1, as most people who run headless probably would prefer to use the HID BP drivers when they actually do connect a keyboard to their system, as they will have less impact on memory usage (in fact, I, and multiple other people I know, have the regular usb-hid module blacklisted on our headless systems for exactly this reason).

marcust, 2016-03-22:
Sounds reasonable

closed 2016-03-22 by marcust