raspberrypi/firmware #572: last rpi-update broke my raspbian-ua-netinst(doesn't boot anymore)

trulio, 2016-03-22:
last rpi-update broke my raspbian-ua-netinst(doesn't boot anymore)

pelwell, 2016-03-22:
Which model of Pi do you have?

What are the symptoms?

trulio, 2016-03-23:
Rpi2

trulio, 2016-03-23:
Yellow led blink once at boot then nothing

trulio, 2016-03-23:
I replaced manually prevoious 4.1.20 boot files and it worked. There is an issue with yesterday's update.

pelwell, 2016-03-23:
So far you are the only person to report a problem with the update, so either your installation failed in some way or there is something different about your setup.

What is is in your config.txt and cmdline,txt?

trulio, 2016-03-23:
here are the files content:
config.txt:
#[pi2]
#kernel=vmlinuz-3.18.0-trunk-rpi2
#initramfs initrd.img-3.18.0-trunk-rpi2 followkernel
gpu_mem=16
#arm_freq=800
#core_freq=250
#sdram_freq=400
#over_voltage=0
dtparam=i2c_arm=off
dtparam=spi=off
device_tree=
start_x=0

cmdline.txt:
dwc_otg.lpm_enable=0 kgdboc=ttyAMA0,115200 console=tty1 elevator=deadline root=/dev/mmcblk0p2 rootfstype=ext4 rootwait

pelwell, 2016-03-23:
With a similar configuration I can repeat the boot failure. It's going to take a while to fully diagnose this - not before tomorrow - but I can see that the mmc driver is unhappy.

You are booting with Device Tree disabled (and yet using dtparam to try and disable things that are already disabled by default - double futility) - that isn't going to be supported on 4.4 and beyond. Why not remove that device_tree= line and see what happens.

trulio, 2016-03-23:
ok lets do it

trulio, 2016-03-23:
it works as you said.

pelwell, 2016-03-23:
That's a useful data point, and it gives you a work-around for now and a potential path to 4.4.

I'll investigate the problem anyway, though.

pelwell, 2016-03-24:
Adding the enable_uart setting changed the dependencies between various parts of the system. In resolving those dependencies, by allowing a more flexible initialisation sequence, one dependency was missed - that of the emmc clock on the core clock. Fortunately it is simple to put that dependency back, and the next firmware will do that. Sorry for the inconvenience.

trulio, 2016-03-24:
Thanks but it was not a convenience.
I could have tried to solve the problem without noticing.
But I wanted to raise up the bug as a contribution to the community.
So thanks to you for your hard work.

pelwell, 2016-03-24:
The master firmware branch has now been updated.

inspector71, 2016-03-25:
FYI I had the same experience on my RPi2: wouldn't boot after upgrading to the latest firmware, re-applied previous firmware and it booted again as normal.

trulio, 2016-03-25:
Thanks to pelwell. everything is OK with the update 
(but i must add that i changed my config files to match the default recommanded configuration now)

closed 2016-03-25 by popcornmix