raspberrypi/firmware #573: framebuffer blocked& display freeze randomly when using xwindow

huangjiansword, 2016-03-23:
I write a qt4 program which use Xwindow of raspberry, sometimes the display(hdmi) freeze, but the machine is still alive, only display freeze.
This case appears randomly but frequently, I can't find any particular steps to reappear it.

When I login in I can see some process(include Xwindow) in uninterruptible state

```
pi@raspberrypi ~ $ ps aux | grep 'D'
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root        19  0.0  0.0      0     0 ?        D    Mar17   1:35 [kworker/0:1]
root        35  0.0  0.0      0     0 ?        S<   Mar17   0:00 [DWC Notificatio]
root      1615  0.3  0.0      0     0 ?        S    Mar17  28:24 [RTW_CMD_THREAD]
root      1681  0.0  1.0   5968  1964 ?        Ss   Mar17   0:30 /sbin/wpa_supplicant -s -B -P /var/run/wpa_supplicant.wlan0.pid -i wlan0 -W -D nl80211,wext -c /home/pi/Documents/wpa_supplicant.conf
root      2427  0.0  0.7   2304  1440 ?        S    Mar17   0:00 daemon --name=DAEMON_python -r -- /usr/bin/python3 /home/pi/Documents/run/HttpAutoUpdate.py
root      2471  0.0  0.5   1696  1032 ?        Ds   Mar17   0:00 startpar -f -- rc.local
root      2540  0.0  3.8  21612  7152 tty7     Ds+  Mar17   3:46 /usr/bin/X -nocursor -nolisten tcp :0 -auth /tmp/serverauth.aHIqu5inZd
root      2554  0.0  0.6   1760  1184 ?        S    Mar17   0:00 sh -c /usr/bin/python3 /home/pi/Documents/python/main.py
root      3960  0.0  5.2  12232  9676 ?        S    Mar21   0:01 /usr/bin/python3 /home/pi/Documents/run/HttpAutoUpdate.py
pi        4966  0.0  1.0   3552  1920 pts/3    S+   14:11   0:00 grep --color=auto D
```

I can see the stack of the processes:
```
pi@raspberrypi ~ $ sudo cat /proc/2540/stack
[<c0055f84>] down+0xc0/0xc4
[<c001e2b4>] bcm_mailbox_read+0x64/0x9c
[<c032a0f8>] bcm2708_fb_set_par+0x88/0x130
[<c0322bc4>] fb_set_var+0x150/0x350
[<c031e334>] fbcon_blank+0x1bc/0x298
[<c0349778>] do_unblank_screen+0xa4/0x1b4
[<c033ef34>] complete_change_console+0x5c/0xe4
[<c03401fc>] vt_ioctl+0x1240/0x12c0
[<c0332b34>] tty_ioctl+0x220/0xd50
[<c014c478>] do_vfs_ioctl+0x3f0/0x5d4
[<c014c6a0>] SyS_ioctl+0x44/0x6c
[<c000e8c0>] ret_fast_syscall+0x0/0x48
[<ffffffff>] 0xffffffff
pi@raspberrypi ~ $ sudo cat /proc/19/stack
[<c0055f84>] down+0xc0/0xc4
[<c005b448>] console_lock+0x1c/0x40
[<c034adc8>] console_callback+0x20/0x160
[<c0039b3c>] process_one_work+0x13c/0x444
[<c003a998>] worker_thread+0x190/0x580
[<c003edf8>] kthread+0xd8/0xf4
[<c000e988>] ret_from_fork+0x14/0x20
[<ffffffff>] 0xffffffff
pi@raspberrypi ~ $ sudo cat /proc/2471/stack
[<c0055f84>] down+0xc0/0xc4
[<c005b448>] console_lock+0x1c/0x40
[<c034b18c>] do_con_write.part.24+0x28/0x2014
[<c034d218>] con_write+0x40/0x58
[<c0334768>] n_tty_write+0x1c4/0x538
[<c0330588>] tty_write+0x10c/0x274
[<c03307cc>] redirected_tty_write+0xdc/0xe8
[<c013a680>] vfs_write+0xb0/0x1e8
[<c013ac94>] SyS_write+0x4c/0xa0
[<c000e8c0>] ret_fast_syscall+0x0/0x48
[<ffffffff>] 0xffffffff

```

And I check the code of raspberry linux, it showes it blocked at
`device_lock(dev);`

```
static inline void device_lock(struct device *dev)
{
	mutex_lock(&dev->mutex);
}
```

```
static int dev_mbox_read(struct device *dev, unsigned chan, uint32_t *data28)
{
	int rc;

	struct vc_mailbox *mailbox = dev_get_drvdata(dev);
	device_lock(dev);
	rc = mbox_read(mailbox, chan, data28);
	device_unlock(dev);

	return rc;
}
```

It seemed a race condition to get mailbox, but I can't find which module get the dev's mutex.

I then try the hello_triangle.bin( the raspberry demo), it blocked at bcm_host_init function,

```
pi@raspberrypi /opt/vc/src/hello_pi/hello_triangle $ gdb hello_triangle.bin
GNU gdb (GDB) 7.4.1-debian
Copyright (C) 2012 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>...
Reading symbols from /opt/vc/src/hello_pi/hello_triangle/hello_triangle.bin...done.
(gdb) bt main
No stack.
(gdb) b main
Breakpoint 1 at 0xa33c: file triangle.c, line 518.
(gdb) r
Starting program: /opt/vc/src/hello_pi/hello_triangle/hello_triangle.bin
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".

Breakpoint 1, main () at triangle.c:518
518        bcm_host_init();
(gdb) s
[New Thread 0xb6d53470 (LWP 5002)]
```

and also vcgencmd version blocked:
`pi@raspberrypi /boot $ vcgencmd version

`
but I know the GPU firmware version is:
```
pi@raspberrypi /boot $ vcgencmd version
Feb 14 2015 22:20:17
Copyright (c) 2012 Broadcom
version 7789db485409720b0e523a3d6b86b12ed56fd152 (clean) (release)
```
I got the gpu_firmware version from another same setting Pi.

I use raspberry A+ mode.The os version is:
```
pi@raspberrypi /boot $ uname -a
Linux raspberrypi 3.18.7+ #4 PREEMPT Mon May 11 15:39:01 CST 2015 armv6l GNU/Linux
```

I think it must be some problem about video core, could any body give me any suggestion about the issue?
Thank you in advance!!!

huangjiansword, 2016-03-23:
vcdbg result as below:
```
pi@raspberrypi ~ $ sudo vcdbg reloc stats

Relocatable heap version 4 found at 0xc000000
total space allocated is 44M, with 44M relocatable, 0 legacy and 0 offline
0 legacy blocks of size 2359296

small_allocs                  : 93
allocs                        : 420
alloc_fails                   : 0
legacy_block_fails            : 0
compactions                   : 0
discard_compactions           : 0
aggressive_compactions        : 0
aggressive_compaction_waits   : 0
aggressive_compaction_timeouts: 0
locks                         : 0
small_locks                   : 0

free list at 0xe9c7560
33M free memory in 4 free block(s)
largest free block is 33M bytes
```

popcornmix, 2016-03-23:
Your kernel is not up to date. Are you using raspbian jessie?
It looks like the gpu has died which is why gpu requests are blocking.
Are you overclocking or using any non-default config.txt settings?

huangjiansword, 2016-03-23:
config.txt setting as below:
disable_splash=1
start_x=1
gpu_mem=64
dtparam=i2c_arm=on
init_uart_clock=16000000
init_uart_baud=1000000
config_hdmi_boost=4
hdmi_cvt 800 480 60 6 0 0 0
hdmi_group=2
hdmi_mode=87
display_rotate=3
avoid_warnings=2

no overclocking.

No, I'm using wheezy.
i
```
pi@raspberrypi ~ $ uname -a
Linux raspberrypi 3.18.7+ #4 PREEMPT Mon May 11 15:39:01 CST 2015 armv6l GNU/Linux
```

popcornmix, 2016-03-23:
Could you test with Jessie?
Wheezy is no longer updated, so we can't help with that.

huangjiansword, 2016-03-24:
@popcornmix , thanks for your answer.
Haha, because about 20,000 wheezy version pi has been shipped and distributed to our customers, so I think I'd better to dig in more, try to solve it. I'll adopt Jessie version pi on new devices.
answers for 2 questions is appreciated.
1,If colorful square(low voltage) appears, could it possible to freeze GPU?some device randomly show colorful square.
2,Can I only update latest gpu firmware for Jessie, but not update kernel version? Is Wheezy kernel  compatible with latest jessie gpu firmware?

JamesH65, 2017-12-21:
@huangjiansword Is this still an issue? Did you find a fix?

closed 2018-06-29 by JamesH65