raspberrypi/firmware #574: Boot issues with the latest firmware

TheSin-, 2016-03-23:
NOTE I have a custom build kernel, though just changing the FW fixes/breaks boot.

Committee 1c4f145 boots without issue.

Just updating *.elf *.bin *.dat on my SD card to commit de7aa7e causes

```
[    0.942498] mmc0: Reset 0x1 never completed.
[    1.046913] mmc0: Reset 0x1 never completed.
[    1.086591] mmc0: Internal clock never stabilised.
[    1.111624] mmc0: Internal clock never stabilised.
[   11.162491] mmc0: Timeout waiting for hardware interrupt.
[   11.268099] mmc0: Reset 0x2 never completed.
[   11.372437] mmc0: Reset 0x4 never completed.
[   21.402493] mmc0: Timeout waiting for hardware interrupt.
[   21.508024] mmc0: Reset 0x2 never completed.
[   21.612357] mmc0: Reset 0x4 never completed.
[   21.636910] mmc0: Internal clock never stabilised.
```

Using the firmware from the repo allows it boot, what changed in the firmware that could cause a custom kernel to fail?  Is there a new Kconfig option that is required to be set with the new firmware?

pelwell, 2016-03-23:
This may be a duplicate of #572 in the sense that the cause may be the same. I'll be looking at this tomorrow.

TheSin-, 2016-03-24:
Yeah I kinda thought it might but there wasn't enough info in that ticket to know, I'm seeing it on all Pi, CM, Pi2, Pi3, but only with custom kernels which is so odd and hard to track down, spent most of the day trying different combos to try and narrow it down.  Let me know if you need anymore info or tests.

pelwell, 2016-03-24:
Adding the enable_uart setting changed the dependencies between various parts of the system. In resolving those dependencies, by allowing a more flexible initialisation sequence, one dependency was missed - that of the emmc clock on the core clock. Fortunately it is simple to put that dependency back, and the next firmware will do that. Sorry for the inconvenience.

TheSin-, 2016-03-24:
perfect thank you, as soon as it's available I'll test and report back though I'm sure it'll be great.

pelwell, 2016-03-24:
The master firmware branch has now been updated.

TheSin-, 2016-03-24:
Confirmed working on all 3 platforms, thanks Phil.

closed 2016-03-24 by TheSin-