raspberrypi/firmware #576: SPI sets erroneous GPIO states

davidperrenoud, 2016-03-24:
If I add `dtoverlay=overlays/spi-bcm2835-overlay.dtb` to my config.txt, the following states are applied to the GPIOs needed for SPI0:

    GPIO7  OUT  (SPI0_CE1)
    GPIO8  OUT  (SPI0_CE0)
    GPIO9  ALT0 (SPI0_MISO)
    GPIO10 ALT0 (SPI0_MOSI)
    GPIO11 ALT0 (SPI0_SCLK)

But the following GPIO states are also erroneously changed:

    GPIO18 OUT
    GPIO24 OUT

Adding a later overlay to set these pins back to input doesn't work.

JamesH65, 2017-12-21:
@davidperrenoud Is this still an issue?

davidperrenoud, 2017-12-21:
I don't have the hardware currently with me. Can anyone check this using wiringPi?

JamesH65, 2018-06-29:
Please update to the latest kernel which may contain a fix for this issue.

Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

closed 2018-06-29 by JamesH65