raspberrypi/firmware #577: 32-bit "boot stub" code appears to write 1 to NSACR res0 bits on RPi 3 (and probably RPi 2)

swarren, 2016-03-26:
I have booted an RPi 3 in 32-bit mode, extracted the "ARM boot stub" (that's what I'm calling the FW-provided code at address 0; is there an official name for this?) and disassembled it with the aim of determining how to implement something equivalent for AArch64. I have found what appears to be a bug:

nsacr_val:
	.word 0x63fff
...
	; NSACR: DMA, CL, all co-procs enabled from NS mode
	ldr	r1, nsacr_val
	; Write NSACR (with ARM1176JZF-S value!)
	mcr	15, 0, r1, cr1, cr1, {2}

As far as I can tell, the value written there is appropriate for the original ARM1176JZF-S CPU in the BCM2835. However, many of the bits are now res0 on the Cortex A53, so I believe that value should be changed.

popcornmix, 2016-03-26:
Yes, that's believable. I'll investigate when I get a chance.
Feel free to suggest a corrected version if you know what should be set.

swarren, 2016-03-29:
I'm not totally sure, but I'd guess the following values would be correct:

ARM1176:
0xc00
- I'm not sure what the DMA bit does.
- I'm not aware of a need for the kernel to access TLB/cache lockdown registers. That said, I suppose it's enabled now so since there really isn't any secure world SW, perhaps just leave it turned on.
- Those last two points imply it /might/ make sense to add in some/all of 0x70000 on top of the value above.
- The "c" value is for co-processor 10/11 for the FPU.

Cortex A7:
0x60c00
- I think it makes sense for NS_SMP/NS_L2ERR to be set (the "6" value).

Cortex A53 32-bit:
0xc00

JamesH65, 2017-12-21:
@popcornmix Is this still ongoing?