raspberrypi/firmware #578: 32-bit "boot stub" code appears write to wrong mailbox clear registers

swarren, 2016-03-26:
I have booted an RPi 3 in 32-bit mode, extracted the "ARM boot stub" (that's what I'm calling the FW-provided code at address 0; is there an official name for this?) and disassembled it with the aim of determining how to implement something equivalent for AArch64. I have found what appears to be a bug:

	ldr	r5, mbox_core0_mbox3_wr_set
	; Write 0 ...
	mov	r3, #0
	; ... to core's mbox3 "write to set" register (this is a no-op)
	str	r3, [r5, r0, lsl #4]
	; Now point at mbox "read" register
	add	r5, r5, #0x40
secondary_wait:
	; Read core's mbox3 read register
	ldr	r4, [r5, r0, lsl #4]
	; Is it still 0?
	cmp	r4, r3
	; Yes, spin
	beq	secondary_wait
	; Clear the value
	str	r4, [r5, r0, lsl #4]
	; and continue to boot
...
mbox_core0_mbox3_wr_set:
	.word 0x4000008c

That first str is writing "0" to a register that QA7_rev3.4.pdf states is write-to-set. That will have no effect. Of course, perhaps the docs have the write-to-set/write-to-clear registers swapped, but if so, the second store is incorrect, since it writes to the read/write-to-clear register (according to that same PDF) which if the registers are swapped will be a write-to-set operation and hence again be a no-op.

popcornmix, 2016-03-26:
I think you are right and the original store of 0 is a no-op.
It is harmless, but as 256 bytes of "boot stub" code is a bit restrictive, removing it is worthwhile.
I'll do this next time the code is touched.

Ruffio, 2016-07-24:
@popcornmix IMHO this looks like a small/quick fix. Any ETA on this?

popcornmix, 2016-07-24:
Added: https://github.com/raspberrypi/tools/commit/4cfaa4fda3e516e6ced869d9e88c20e3d9bffa51
Will be in next firmware build.

Ruffio, 2016-08-07:
@swarren have you tried to upgrade firmware. If it is fixed please close this issue.

Ruffio, 2016-08-08:
@swarren can it be closed?

swarren, 2016-08-09:
The commit in the armstub source looks sufficient to close this. Pinging twice within 24 hours is rather impatient.

closed 2016-08-09 by swarren

Ruffio, 2016-08-09:
@swarren I don't know what went wrong there. Sorry.