raspberrypi/firmware #579: 64-bit boot stub?

swarren, 2016-03-26:
If I write a "boot stub" for the VC FW to place into memory at address 0 in 64-bit mode, will you accept it into the VC FW? If so:

a) What license do you want it under? I'd be happy to make it MIT/BSD/X11 or assign copyright to the Pi Foundation; whatever is easiest for you.

b) Which address(es) does the VC FW write to? I assume at least 0x14 for the ATAGS/DTB pointer, but I'm not sure whether the initialization values for cntfrq or the Linux machine ID are written too. I'm discounting address 0x100 and on since that's just data and not part of the boot stub itself.

(As an aside, if this stub could be loaded from a separate .bin file, that might be helpful since it'd allow easy experimentation, but that would then get into a discussion of how to name the files since we'd need different files for bcm2835, bcm2836, bcm2837 32-bit, bcm2837 64-bit).

swarren, 2016-03-26:
Oh, and it might turn out useful if the boot stub code could be included in e.g. Qemu for RPi emulation - both the existing 32-bit version and any new 64-bit version. Do you have any problems with my doing that, even for the existing 32-bit binary that you wrote? I wonder if you could publish the source under an open license to help the contribution chain to Qemu.

popcornmix, 2016-03-26:
No problem with releasing the boot stub code. See: http://pastebin.com/raw/stxjiVVD
If you are happy with a similar licence for a 64-bit boot stub, we'd be happy to include it in the firmware.

Just let us know what addresses to modify. Currently we do:
```
         ((unsigned *)mem)[BCM2836 ? 5:13] = device_tree_address;
```
But we could do something different in 64-bit mode.
The machine ID and ctrfreq are not overwritten.

capnm, 2016-03-27:
Hmm, it seems to work without issues, but at least according to the A53 TRM (p.328)  the order of enabling the caches and SMP is wrong:

> Data coherency is enabled only when the CPUECTLR.SMPEN bit is set. You must set the
SMPEN bit before enabling the data cache. If you do not, then the cache is not coherent
with other cores and data corruption could occur.

```
	mrc p15, 0, r0, c1, c0, 0 @ Read System Control Register
	orr r0, r0, #(1<<2)       @ cache enable
	orr r0, r0, #(1<<12)      @ icache enable
	mcr p15, 0, r0, c1, c0, 0 @ Write System Control Register
.if !BCM2710
	mrc p15, 0, r0, c1, c0, 1 @ Read Auxiliary Control Register
	orr r0, r0, #(1<<6)       @ SMP
	mcr p15, 0, r0, c1, c0, 1 @ Write Auxiliary Control Register
```

swarren, 2016-03-27:
Writing to the SCR doesn't actually enable the icache/dache. To enable the cache, the MMU must also be configured/enabled, and the "ARM boot stub" doesn't do that (Linux/U-Boot/... does it during early boot).

capnm, 2016-03-28:
Great, thanks!  Do you know why is the SCR code in the stub at all?

I guess the next stop is the kernel entry point here
http://lxr.free-electrons.com/source/arch/arm64/kernel/head.S#L54
MMU=off is clear, but they actually mean the state of the SCR register bits, right?

```
 58  * The requirements are:
 59  *   MMU = off, D-cache = off, I-cache = on or off,
```

What about the IIRC VC/ARM **shared** L2 cache? Is it always enabled? I couldn't find anything special in the kernel code.

swarren, 2016-03-28:
Correction: System Control Register is SCTLR, whereas SCR is Secure Configuration Register.

I think the write to the SCTLR in the stub is not necessary; At least the arch/arm64 kernel repeats this and I'm sure the arch/arm kernel must too; see setup of x0 here:
http://lxr.free-electrons.com/source/arch/arm64/mm/proc.S#L178
based on data here:
http://lxr.free-electrons.com/source/arch/arm64/mm/proc.S#L229
and x0 is written to SCTLR here:
http://lxr.free-electrons.com/source/arch/arm64/kernel/head.S#L644

I suspect the comment in the kernel code refers to the logical state of the caches being disabled.  The fact this is implemented by writing bits in the SCTLR is true but simply not explicitly mentioned.

I don't know if the L2 cache can be enabled/disabled, but IIUC all control over that cache is from the VideoCore FW. I believe its state is irrelevant to ARM-based SW, except for interaction with peripherals doing DMA, since it's beyond the point of coherency/unification for the ARM cores themselves. As such I would not expect to find any code that controls it.

swarren, 2016-03-30:
A first cut is available at https://github.com/swarren/rpi-3-aarch64-demo in armstub64.S. The most-recent-but-one includes a TODO list describing the outstanding changes.

swarren, 2016-04-02:
I have pushed what I hope is a final version of the 64-bit stub to that github repo. The source is in armstub64.S.

I have tested this with U-Boot, and the test app in the same repo. I have not tested it with any Linux kernel yet, nor have I attempted to boot any Linux kernel from U-Boot. I believe @anholt will test it some time.

The DTB address should be written *as a 64-bit value* to address 0xf8.

The stub implements the standard RAM-based "spin table" method of SMP secondary CPU boot. This is a mechanism already supported by the mainline Linux kernel's arch/arm64 port, so should ease upstreaming kernel support; the arch/arm64 maintainers are unlikely to welcome a new SoC-specific SMP mechanism (besides spin table or PSCI) that uses the bcm283x mailbox registers.

I'd like to suggest creating a new git repo that contains the ARM stub source for all the Pis. I'd be happy to contribute patches to such a repo to fix the issues I filed with the 32-bit stubs if there was somewhere I could send a regular git patch against. I'd also be happy to create the initial content of such a repo if you want; just let me know (assuming the pastebin you linked above is all the versions; otherwise it might be best if you filled in the existing 32-bit stub code first).

sukantoghosh, 2016-04-02:
Do I need any custom version of boot-firmware for this or is the HEAD of github.com/raspberrypi/firmware master branch supposed to work? Somehow this is not working.

This might be worth documenting in the README as well as the firmwares shipped with 2016-03-18-raspbian-jessie don't respond to 'enable_uart=250'.

BTW, I really hope that when this stub is integrated to VC FWs it is picked from a bin file as Stephen suggested

popcornmix, 2016-04-02:
I'm happy to add support to load:
armstub.bin (when on Pi1)
armstub7.bin (when on Pi2 or Pi3 in 32-bit mode)
armstub8.bin (when booting in 64-bit mode)

at address zero (up to 0x100 bytes). If ``armstub*.bin`` is not present it will use default ones (including current armstub64.S from @swarren's repo).

Also the presence of kernel8.img will enable 64-bit mode.
I can add existing ``armstub*.S`` files to a Pi repo (possibly in tools/mkimage).
Is that okay?

swarren, 2016-04-03:
@sukantoghosh I tested with 046effa13ebc "firmware: arm_loader: emmc clock depends on core clock See: https://github.com/raspberrypi/firmware/issues/572", which was HEAD of github.com/raspberrypi/firmware master branch when I last fetched a few days ago.

swarren, 2016-04-03:
@popcornmix that all sounds good to me, thanks!

Just one query though: The stub code you posted had an ifdef for BCM2810-vs-not, so I think you'd end needing 4 FW files: ARMv6, ARMv7, ARMv8 32-bit mode, and ARMv8 64-bit mode, or am I misinterpreting something?

swarren, 2016-04-03:
@sukantoghosh that should be enable_uart=1 not =250. I'll fix my repo in a minute.

popcornmix, 2016-04-03:
Yes, you are right, the stub code is slightly different between Pi2 and Pi3 so we should have an extra stub file.

sukantoghosh, 2016-04-03:
thanks @swarren it works for me

anholt, 2016-04-04:
With @swarren's boot stub I have SMP enabled on 64-bit Linux, and I'm making it to the point of trying to NFS root mount before it fails because USB device probing isn't working.

On 2836, the boot stub was setting a 19.2Mhz CNTFREQ but not LOCAL_CONTROL/PRESCALER, so we set the LOCAL_* regs early in Linux boot.  For this boot stub, I've disabled the LOCAL_* setup so that we run with the 1Mhz it sets in CNTFREQ.  I'm wondering, though: Don't we want to run the timer at 19.2Mhz?  If so, should the firmware/boot stub take over that responsibility, if it was already setting CNTFREQ previously?

swarren, 2016-04-04:
BTW, there's some discussion re: DWC2 and broken DMA support in the kernel on page 3 (and later) of https://www.raspberrypi.org/forums/viewtopic.php?f=72&t=137963&start=50. I'm not sure if that's talking about the RPi Foundation kernel or mainline.

Where (which module) are the LOCAL_CONTROL/PRESCALER registers; I'm not familiar with those.

capnm, 2016-04-04:
Other difference that I spotted during my experiments (and I guess the
fw should take care of) is the another arm64 kernel load-adress, currently
hardcoded in the makefile

http://lxr.free-electrons.com/source/arch/arm64/Makefile#L53

TEXT_OFFSET := 0x0008_0000 # 512K
was 0x8000 / 32k

The image should be idealy loaded at adress taken from the image
header (if magic == 0x644d5241 ?), not sure what to do with the
u-boot or another images or how to pass that to the boot blob.
So probably just simply load the arm64 image at 0x00080000?

```
The decompressed kernel image contains a 64-byte header as follows:
  u32 code0;			/* Executable code */
  u32 code1;			/* Executable code */
  u64 text_offset;		/* Image load offset, little endian */
  u64 image_size;		/* Effective Image size, little endian */
  u64 flags;			/* kernel flags, little endian */
  u64 res2	= 0;		/* reserved */
  u64 res3	= 0;		/* reserved */
  u64 res4	= 0;		/* reserved */
  u32 magic	= 0x644d5241;	/* Magic number, little endian, "ARM\x64" */
  u32 res5;
```

popcornmix, 2016-04-04:
I've added the arm stubs here: https://github.com/raspberrypi/tools/commit/920c7ed2ee7c84d57feeec76ba0477bd97f36c2f
Pull requests for fixes are welcome.

Here is a (not hugely tested) firmware:
https://dl.dropboxusercontent.com/u/3669512/temp/firmware_armstub.zip
that supports, e.g.
``armstub=armstub8-32.bin``
to load a stub file from a file. The file should be 0x100 bytes (or less).

If you have a ``kernel8.img`` (and optionally an ``armstub8.bin``) they should be used and enable 64-bit mode automatically. Supported kernel names are:
``kernel8.img``,  ``kernel8-32.img``,  ``kernel7.img``,  ``kernel.img``
If one of those is loaded it will by default look for:
``armstub8.bin``, ``armstub8-32.bin``, ``armstub7.bin`` ``armstub.bin``
although they can be overridden with ``kernel``/``armstub`` in config.txt.

anholt, 2016-04-05:
drivers/irqchip/irq-bcm2836.c

swarren, 2016-04-06:
@popcornmix I've tested firmware_armstub.zip and it works great, thanks! I tested:

1) Unified SD card containing kernel.img (U-Boot rpi build), kernel7.img (U-Boot rpi_2 build), kernel8.img (U-Boot rpi_3 64-bit build). I plugged the card into a B+, 2 B, and 3 B, and all worked. I didn't test if the 2 B board picked kernel.img or kernel7.img; both would have worked. I'm pretty sure the B+ correctly picked kernel.img not kernel7.img or it would have failed to execute the ARMv7 instructions in it.

2) An SD card containing just kernel8-32.img (U-Boot rpi_3_32 build) booted on the Pi 3. I also accidentally validated that fallback to kernel7.img works on the 3 B in 32-bit mode.

I didn't use mkknlimg in any of my testing. I should probably switch to that soon:-)

swarren, 2016-04-06:
@capnm yes it looks like the FW and 64-bit ARM stub can't just hard-code the kernel address; it must be read from the kernel image header. I'll prepare a patch for the stub that:
- Defines a 64-bit location that the FW can over-write with the address of the kernel, to which the stub will jump.
- The default value at that address will still be 0x8000 for compatibility. When loading a Linux kernel, the FW will always have to read the value out of the kernel image header, so the default value isn't at all relevant for Linux, just custom other FW.
- I'll also specifically reserve a location for the CPU0 spin-table release address. This will never actually be used, but according to the DT binding documentation for ARM stream tables all CPUs need a release address defined.

swarren, 2016-04-06:
@popcornmix I've also tested using custom ARM stubs (only on RPi3 in 64-bit mode). If I use config.txt option armstub=armstub8.bin, it works great. However, if I don't put any armstub= option in config.txt, the FW doesn't seem to automatically check for and load armstub8.bin (even though I put the "kernel" in kernel8.img). I thought from your description it would?

popcornmix, 2016-04-06:
Yes, your expectations are correct. I'll push an updated firmware later today that fixes a few issues. Hopefully that will work.

Currently, not specifying ``armstub=`` and ``armstub8.bin`` being found enables 64-bit mode automatically.
Obviously ``armstub=mystub.bin`` shouldn't enable 64-bit mode.
What about ``armstub=armstub8.bin``?
What about ``armstub=mystub8.bin``?
Currently they don't, but I'm wondering if one or both should (``*8.bin`` being the wildcard to enable 64-bit mode). Thoughts?

swarren, 2016-04-07:
@popcornmix my inclination is to keep things simple. Have the FW decide whether to boot in 32- or 64-bit mode solely based on the kernel image filename, and don't take the stub filename into account. That's because the kernel is the primary piece of SW, and the stub is somewhat only there to support the kernel. Once that bit-size determination is made, use that to select the default stub filename. This way, the FW doesn't have to have rules to decide whether the bit size implied by the kernel filename overrides the bit size implied by the stub filename, or the other way around.

If someone wants to do something really unsusual and run a 32-bit OS (kernel) under a 64-bit secure monitor/OS, I assume they'd make that work by putting the secure monitor into kernel8.img or setting arm_control=0x200 (thus ensuring 64-bit booting), using a custom stub file (ths avoiding the built-in switch to HYP/EL2), and having the secure OS load the real kernel image (or perhaps concatenating it with the secure OS image). Hence, even with this simple scheme, doing something unusual is still possible.

I envisage something like:

```
bits=32
if arm_control 64-bit flag is set:
    bits=64
if kernel= not in config.txt and kernel8.img exists:
    bits=64

if bits == 64:
    armstub_fn=armstub8.bin
    kernel_fn=kernel8.img
else:
    armstub_fn=armstub8.bin
    kernel_fn=kernel8-32.img

if armstub= in config.txt:
    armstub_fn=value_from_config

if kernel= in config.txt:
    kernel_fn=value_from_config

Does that make sense?
```

swarren, 2016-04-07:
@anholt the timer frequency setup is implemented in https://github.com/raspberrypi/tools/pull/54, and tested w/ U-Boot's sleep command.

popcornmix, 2016-04-07:
@swarren there are a number of "bare metal" users who just use a single block of arm code.
Currently they use a kernel.img with kernel_old=1.
It would probably make more sense for them to just use armstub.bin and no kernel file. We could then deprecate kernel_old=1. (The size limit on armstub.bin would be removed).
In that scenario determining the bits from armstub makes sense.
In theory it would be possible for armstub.bin to be 64-bit code and kernel.img to be 32-bit (with armstub switching back to 32-bit mode), which again would mean deriving the bits from the armstub would make more sense.

popcornmix, 2016-04-07:
Here is latest test firmware. It includes latest armstub8.S from github:
https://dl.dropboxusercontent.com/u/3669512/temp/firmware_armstub2.zip

It is useful for the firmware to provide some information. Currently:
device_tree_address (dtb_ptr) and kernel_address (kernel_entry)
Possibly we'll also need device_tree_end.

Currently device_tree_address is (uint32_t *)0x34 for Pi1 and (uint32_t *)0x14 for Pi2/Pi3.
We are not currently writing it, but it looks like armstub8.S is using (uint64_t *)0xf8.

armstub8.S also uses (uint64_t *)0xd0 for kernel_entry

As armstub7.S is quite cramped in 0x100 bytes currently (we may be able to increase the size, but that could break some bare metal assumptions), using 0xd0 might be awkward.

Can we define an API for where to put these words? It seems 32-bit values will be enough as bus addresses are still 32-bits even in 64-bit mode. Using unused vectors (like 0x14) is convenient for 32-bit mode. Can we find suitable values that work for both 32-bit and 64-bit stubs?

We could always use different values for 32-bit and 64-bit, but if making them the same is possible, it simplifies things.

pelwell, 2016-04-07:
device_tree_end shouldn't be needed, since the dtb contains a length and any serious OS will unflatten it into a more convenient form ASAP.

swarren, 2016-04-07:
I was going to suggest unifying the addresses too. That would also be very useful since you plan to make armstub a "first class citizen" and the primary mechanism for bare-metal. Having consistent addresses would make life easier for people implementing their own stub. I assume that when loading just a stub, all the DT loading/processing would still happen, and the DTB pointer would still be written into the stub memory. Hopefully there will also be a simple specification (beyond just the labels in the code) indicating what addresses the FW will write to in the stub. BTW, I like the idea of allowing just a stub, since it'll resolve the issue where bare metal code needs to do odd tricks to get back into SVC from HYP mode. Removing the size limit if only loading armstub was the part I hadn't though about in my previous answer. I guess given that I'd suggest:

```
bits=32
if arm_control 64-bit flag is set:
    bits=64
if (armstub= not in config.txt) and (armstub8.bin exists):
    bits=64
if (not only loading armstub) and (kernel= not in config.txt) and (kernel8.img exists):
    bits=64
# Question: Is setting arm_control a good mechanism for people who want custom filenames,
# yet also want 64-bit boot?

if bits == 64:
    armstub_fn=armstub8.bin
    kernel_fn=kernel8.img
else:
    armstub_fn=armstub8.bin
    kernel_fn=kernel8-32.img

if armstub= in config.txt:
    armstub_fn=value_from_config

if kernel= in config.txt:
    kernel_fn=value_from_config
```

I chose 64-bit addresses for the 64-bit version for potential forwards-compatibility, just in case we ever get an RPi4/5/6 with a much larger amount of (or different location of) RAM. I suppose we could use 32-bit values instead since that's all that's necessary in practice. I obviously don't have visibility into whether this future-proofing is likely to be useful; either based on potential new HW or whether the new HW would require a completely different stub for other reasons.

Re: code size: I suspect armstub7.S can be optimized a bit, or at least the order of code and constants re-organized so we could e.g. always put the kernel address at 0x100 - sizeof(pointer). In particular, I notice it sets up MVBAR and does SVC in order to get to SVC mode to set some secure registers and/or return to HYP mode. Presumably the CPU is already in SVC mode so there's no need to invoke SVC to do that, and hence no need to set up MVBAR. Equally, getting into HYP mode should be as simple as an eret (or AArch32 equivalent). I expect that would save quite a few instructions.

popcornmix, 2016-04-07:
> I suspect armstub7.S can be optimized a bit

I found I couldn't write to CNTVOFF from reset state and needed the smc to _secure_monitor to do that. I got the code from uboot, but there may be simpler ways of achieving this.

Pull requests are welcome for a simplified armstub7. It would also be nice to get the same sort of spinning code. Currently armstub7 busy spins rather than using wfe. Not a big issue for a working kernel, but becomes a bit of a heat generator for bare metal users who only use one core.

SylvainGarrigues, 2016-04-08:
I have one question here. I am no expert, but I am under the impression that the 0x8000 jump address is hardcoded in the bootstub, right?

Is the kernel_address= config.txt parameter deprecated then? How does one select the kernel load address? FreeBSD kernels require to be loaded on a 1MB boundary, this 0x8000 value is not convenient.

swarren, 2016-04-08:
@popcornmix Ah yes, the SVC is needed to write CNTVOFF. The CPU boots in supervisor mode. CNTVOFF can only be written in monitor mode with SCR.NS set or from HYP mode. So the the CNTVOFF write can't be before switching out of supervisor mode somehow.

I'm not sure that CNTVOFF needs to be initialized though. I'd expect that to be the job of whatever code runs in HYP mode. The Linux kernel appears to set CNTVOFF to 0 if it's started in HYP mode, and I assume any actual hypervisor would also initialize it.

It looks like the SMC is also required due to the need to write SCR. I think the code that sets SCR can be simplified though. The value can be constructed from scratch, so no need for the mrc instruction. I think the only bits that actually need to be set are NS and maybe SCE; I don't think AW/FW have any effect on regular use-cases. I think NS/SCE can be set up with a single mov. That should save about 3 instructions.

_secure_monitor could be moved to address 0x8 to avoid the jump, and so save 3 more instructions.

I'm not sure that it's necessary to write VBAR in hyp mode, and not by copying the value from system mode since I don't think it points at anything in particular. That might save 2 instructions.

So, there seems to be opportunity for optimization:-) Testing is probably the largest issue; a time-sink.

I think you wanted to save instructions to add a wfe to the SMP spin loop for the A7? That might be problematic; existing kernels don't issue a sev in bcm2836_smp_boot_secondary() at the moment, and that would be needed to wake up any wfe. The Pi Foundation could probably be updated in sync with new FW pretty easily, but not mainline or other OSs. I'm not 100% sure wfe can be added in a backwards-compatible way.

I think it should be easy to rework all stubs to read 32-bit DTB pointer from 0xf8 and 32-bit kernel address from 0xfc.

pelwell, 2016-04-08:
@SylvainGarrigues Yes, the current stubs hard-code 0x8000 as the kernel entry point. The kernel_address parameter is the address to load the kernel to, so you could imagine creating a kernel where the entry point isn't the start of the file, but really kernel_address is meant for monolithic images that include stub and kernelm, specified using `kernel_old=1`. As you'll have noticed the new stubs will have a parameter indicating the entry point, so this won't be a problem in the future.

SylvainGarrigues, 2016-04-08:
@pelwell If I understand well, currently, the kernel file (kernel7.img or filename specified in kernel=) is loaded at 0x8000 and kernel_address specifies the location of the entry point, is that right? I always thought kernel_address was the address the kernel file (kernel7.img) was loaded in memory (to override the 0x8000), and so I was wrong.

pelwell, 2016-04-08:
No, it's the other way around:
>  the current stubs hard-code 0x8000 as the kernel entry point. The kernel_address parameter is the address to load the kernel to

popcornmix, 2016-04-08:
> The Linux kernel appears to set CNTVOFF to 0 if it's started in HYP mode

That is believable. Changing the stub code to set HYP mode is quite new, so previously VCNTVOFF wasn't being set by kernel. I'll check if the initialisation of CNTVOFF can be removed now.

> I think the only bits that actually need to be set are NS and maybe SCE; I don't think AW/FW have any effect on regular use-cases. I think NS/SCE can be set up with a single mov. That should save about 3 instructions.

You mean NS and HCE? That would be 0x101 which doesn't fit in a mov. We can still save two instructions by not preserving SCR (as the reset state is known).

swarren, 2016-04-08:
@popcornmix i realized later that the entire existing SCR value can be set using a movw instruction (which supports a 16 bit constant field) so I guess there's no need to modify the value at all.

popcornmix, 2016-04-08:
What do you think about putting a magic value and a version number in two words of the stub.
We'll only overwrite with device tree address and kernel address if the magic word is present.
The version may allow us to set additional words in the future.
We could perhaps use 0xf0 and 0xf4 and the firmware could set them to zero if recognised (so your spin table still works).

popcornmix, 2016-04-08:
32-bit armstubs have been updated in github. This is the firmware that I will be pushing later:
https://dl.dropboxusercontent.com/u/3669512/temp/firmware_armstub3.zip

The stub should contain:
```
.org 0xf0
.word 0x5afe570b      @ magic value to indicate firmware should overwrite atags and kernel
.word 0			      @ version
atags:	.word 0x100   @ device tree address
kernel:	.word 0x8000  @ kernel start address
```

popcornmix, 2016-04-08:
Current firmware has been pushed and can be got through rpi-udpate.
The stubs match current github ones, but the 64-bit could do with updating to the new offsets.

swarren, 2016-04-09:
The latest firmware.git content doesn't seem to be initializing the UART correctly on 64-bit Pi 3 (or at least something is going wrong); I see corrupted output. This is true even once I've updated the 64-bit stub for the new layout, copy that to the SD card, and reference it from config.txt using armstub=armstub8.bin.

With firmware_armstub.zip my test app worked fine. That's true with either the 64-bit stub built into the FW, or with config.txt armstub=armstub8.bin and the stub built from the latest firmware.git. It's also true whether or not I use mkknlimg on the app ("kernel") binary.

With firmware_armstub2.zip or firmware_armstub3.zip I see the same issue with the UART. I didn't test all the options here (e.g. external stub or not, mkknlimg use or not etc.) Sorry, I guess I forgot to pick up and test firmware_armstub2.git, and didn't get a chance to test firmware_armstub3.git.

swarren, 2016-04-09:
BTW, I've pushed the updated armstub8.S to git://github.com/swarren/rpi-tools.git branch armstub8_magic_layout, but haven't issued a pull request since I haven't successfully tested it. My test app is at git://github.com/swarren/rpi-3-aarch64-demo.git branch master directory 64/ if you want to try it.

capnm, 2016-04-09:
@popcornmix Does it still make a sense to check DTOK?
As far as I can see, the arm64 kernel wouldn't boot without the DT anyway.

pelwell, 2016-04-09:
But Linux isn't the only "kernel" - is there no 64-bit OS or loader that wants ATAGs?

I would consider changng the defaults, though.

SylvainGarrigues, 2016-04-09:
I can speak for NetBSD and FreeBSD on RPI 0/1/2/3: neither use ATAGS.
- FreeBSD is fully configured by the DTB u-boot provides its bootloader, or by the DTB pointed to by the r2 register if directly booting the kernel.
- NetBSD use VC properties to get command line and configuration info.

I wouldn’t make ATAGS the default anymore.

SylvainGarrigues, 2016-04-09:
If I may add: for non-Linux kernel, not having to rely on a 3rd-party tool (mkknlimg) to tag our DTB-aware kernel binaries to make them bootable looks a cleaner approach.

swarren, 2016-04-10:
DT only (or at least by default) on AArch64 makes sense to me too.

@SylvainGarrigues For Linux, r2 is used on 32-bit ARM, but x0 is used on 64-bit ARM. I coded the Pi "ARM stub" to follow the same. I assume that works for FreeBSD too?

zxombie, 2016-04-10:
FreeBSD has it's own boot loader. It will most likely also use U-Boot (or UEFI if someone ports it) to boot.

pelwell, 2016-04-11:
@swarren The Friday firmware release did have "issues" that may have been the cause of your problem. We've done a quick respin to fix it, but I'm still open to changing the defaults.

swarren, 2016-04-12:
@popcornmix @pelwell The latest in firmware.git (e484afecfa0a "kernel: Bump to 4.4.6") seems to work well. Thanks. I validated:

- Single SD card image with U-Boot boots on Pi B+, 2 B, 3 B (64-bit).
- Built in 64-bit ARM stub works.
- The new 64-bit stub I created that uses the new layout works.
- Loading of separate armstub8.bin, with or without armstub= in config.txt.

I think the only remaining issues for 64-bit are:

- Parse the kernel load address from the Linux image header (typically yielding 0x80000, whereas the default 0x8000 is good for U-Boot).
- Use DTB by default on 64-bit.

capnm, 2016-04-12:
I additionally validated:
+ The size limit on armstub8.bin is removed (I successfully executed a 64MB stub)

I think the default for the **'armstub8.bin only'** case could be (not tested):

if the  **0xf0: magic** is not present, don't overwrite anything,
otherwise:
 - write the DT at the address found in 
  * the **device_tree_address=xy** option (+ overwrite the atags: location)
  - at the address found in the **atags:** stub location 

- switch to ATAGS if available:
 - already documented [/boot/overlays/README](https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README#L12)
 **device_tree=** option 
 - invent a new **force_atags:** location in the stub?

pelwell, 2016-04-12:
@swarren Thanks for the feedback. Tell me about this kernel header? Is it readable before the kernel is decompressed? (One of the main reasons for the existence of the kernel trailer is the inability to see inside a compressed kernel).

capnm, 2016-04-12:
@pelwell the header is described here
https://www.kernel.org/doc/Documentation/arm64/booting.txt

The arm64 kernel supports only uncompressed or gzip compressed image that must
be decompressed by the firmware or the bootloader - before it's loaded into memory).

pelwell, 2016-04-12:
I'd like more clarity on the information flow between the kernel, stub and firmware. Initially it was simple:

    config.txt -> firmware -> stub -> kernel

We then added a back channel from the kernel trailer, but it can be thought of as a separate file and conveniently doesn't affect the placement of the kernel:

    config.txt + trailer flags -> firmware -> stub -> kernel

Now the kernel name is another input. The stub is being read, but only to determine where and whether to patch the device tree and kernel addresses.

    config.txt + trailer flags + kernel name -> firmware <-> stub -> kernel

@swarren suggests the load address of the 64-bit kernel can be read from the kernel header, which will only work for uncompressed kernels. This is awkward because it would require the kernel to be read in two pieces or potentially moved, but that isn't enough reason in itself to discount it. An easier, more portable option would be to read the load address from the stub.

@capnm suggests the device tree address may be read from the stub. This in itself wouldn't be sufficient; the firmware uses device_tree_address and device_tree end - the end is needed because DTB can grow as a result of overlays being applied, and libfdt needs to know how large it can grow. We could include the end address in the stub as another firmware input - the kernel doesn't require since the DTB includes a length - which could be overwritten later so as not to waste space. We'd probably want the version number to indicate its presence. In the event that the DTB was too large for the slot, should the firmware load it high instead?

For stub fields that can work in either direction (kernel_address, device_tree_address), the presence of a zero can indicate an input and non-zero an output.

Any thoughts?

SylvainGarrigues, 2016-04-12:
I am not in favor of reading a "kernel" loading address from the "kernel" image as "kernel" can be anything, not necessarily an ELF, and not necessarily Linux code.

I believe we should make as few assumptions as possible as to the content / format of the "kernel" binary image.

capnm, 2016-04-12:
The arm64 linux kernel header includes a magic number. For an image without that, you can default to something else.

The purpose of that all would be: you can just put the 'make Image'  linux kernel as kernel8 in the /boot folder and **it works**! Without any additional configurations, boot-loaders or special stubs ...

AFAIK the compression/decompression cannot by done by kernel itself any more, it is optional and doesn't need to be supported.

pelwell, 2016-04-12:
I'll be surprised if the 64-bit Linux entry point ever changes from 0x80000, but for the cost of some implementation effort I don't mind supporting the possibility that it might, as long as everybody is aware that this will only work with an uncompressed kernel.

> I like that idea. Currently there is no way for the stub-only based image to get the DTB.

Can you explain this? In the latest firmware a compatible stub will be patched with the DTB address - is that not sufficient?

popcornmix, 2016-04-12:
As far as I remember a compressed kernel boots significantly faster than an uncompressed kernel, so I question the value in supporting features for the uncompressed kernel that I doubt any popular distribution would make use of.

It also introduces an unexpected behaviour when switching from compressed to uncompressed kernel. It sounds like assuming 0x80000 as the default kernel address for 64-bit may be a more useful behaviour, which would still allow a kernel8.img to "just work" in most cases.

TheSin-, 2016-04-12:
I have a quick question about the new firmware code to determine 64bit.  If I've read everything right I assume that i need a file named kernel8.img (which I don't use this name in my dist) OR I need to set the stub, otherwise it has no idea I want 64bit right?  If this is true, could I request having a config.txt param to turn on 64bit in stead of using the stub or filename?  I use kernel=[customname] and I'd like to stay bout from needed a stub file and just use the firmware.

OR is the stub file not a real local file, just a switch to tell it which stub in the firmware to use, in which case that is a switch?  like being able to do bits=64 in the config.txt would be more useful IMHO.

[59 more replies]

closed 2016-04-19 by swarren