raspberrypi/firmware #580: hdmi_ignore_cec_init and disable_splash don't appear to have any effect in /boot/config.txt

DanTup, 2016-03-27:
(Apologies if this is the wrong place to post this; if so, please advise where I should re-post :))

I added the following to my boot.config (based on [this page](http://elinux.org/RPiconfig)):

```
# Disable turning TV on at boot!
hdmi_ignore_cec_init=1

# Disable rainbow splash screen
disable_splash=1
```

But neither seem to have any effect (they're both at the bottom, and there's no other instances of them in the file). I still see the square rainbow at startup, and it still turns the TV on.

The splash screen isn't a big deal; but the TV turning on when I reboot the pie if I'm not home is quite an annoyance.

I'm using [raspian downloaded with Noobs 1.9.0](https://www.raspberrypi.org/downloads/noobs/) yesterday and have run `apt-get update && apt-get dist-upgrade` prior to setting these.

Any ideas?

XECDesign, 2016-03-27:
When you use NOOBS, it first boots from the first partition, NOOBS determines which partition the OS you want to boot into is on and then reboots again. The first time, config.txt is read from the first partition, the second time it's on the partition that you've edited config.txt on. In other words, your config.txt changes have no affect when NOOBS starts.

Try doing the following:
```
sudo -i
mkdir /tmp/recovery
mount /dev/mmcblk0p1 /tmp/recovery
nano /tmp/recovery/config.txt
```
Add the config.txt options you'd like and save the file, then unmount the recovery partition:
```
umount /tmp/recovery
exit
```

DanTup, 2016-03-27:
Aha; that makes sense. I'll give this a go today. Thanks!

DanTup, 2016-03-27:
Works great; thanks very much!

closed 2016-03-27 by DanTup

XECDesign, 2016-03-27:
You're welcome. Glad it worked.