raspberrypi/firmware #581: MMAL renderer represents ARGB alpha black instead of transparent

zillevdr, 2016-03-30:
I will render a OSD as ARGB. The transparent part of the picture is black. The layer behind is'nt visible. 

format->encoding = MMAL_ENCODING_ARGB;

is set. What can I do to that the renderer use the alpha value from picture?

6by9, 2016-03-30:
Currently nothing. It really supports XRGB, but there isn't a format for that. It only supports a global alpha.
I did look briefly at supporting ARGB with per pixel alpha, but never completed it. I'll see if I can find time to resurrect it.

You can use the DispmanX API directly to support ARGB - that's all that MMAL video_render is wrapping with some buffer management.

zillevdr, 2016-03-31:
OK, thank you!

Ruffio, 2016-07-29:
@zillevdr can this be closed?

zillevdr, 2016-07-30:
I think yes. Now I use DispmanX API.

closed 2016-07-30 by zillevdr