raspberrypi/firmware #583: README does not state where source code can be found

rebroad, 2016-04-09:
The current README file explains that this repository stores the pre compiled binaries. Could it be amended to include where to find the source code also?

6by9, 2016-04-09:
The firmware is closed source. You aren't going to find source code or the required tools to build custom versions.

rebroad, 2016-04-09:
Ah, ok. Well, it could mention that then.

cleverca22, 2016-04-10:
and name which parts are open-source and where (the kernel, modules, userland libs)

lurch, 2016-05-02:
It was a long time ago, but I wrote some details in https://github.com/raspberrypi/documentation/pull/153#issuecomment-71926226

Ruffio, 2016-06-30:
@rebroad can this issue be closed?

Ruffio, 2016-08-10:
@popcornmix as firmware is close source, therefore the soruce code can't be found anywhere, so this 'issue' should be closed.

closed 2016-08-10 by popcornmix