raspberrypi/firmware #586: Sound issues with 4.4.6-v7+ #874

marcust, 2016-04-12:
I'm kind of lost right now because I lost sound and I don't know why, the only thing that I remember changing is the firmware:
```
Linux 4.4.6-v7+ #874 SMP Mon Apr 11 19:20:24 BST 2016 armv7l GNU/Linux
```

The modules are loaded:
```
snd_bcm2835            24948  0
snd_pcm                73870  1 snd_bcm2835
snd_timer              19032  1 snd_pcm
snd                    51237  3 snd_bcm2835,snd_timer,snd_pcm
```

But I have nothing in /dev/snd
```
ls /dev/snd/
seq  timer
```

On other machines with older firmwares (#872) that looks similar, only that I have more devices. 

When I downgrade to 
```
Linux 4.1.21-v7+ #873 SMP Mon Apr 11 18:22:10 BST 2016 armv7l GNU/Linux
```
I get them back, so I guess 874 somehow looses audio? 
Or do I have to do something in my userland?

pelwell, 2016-04-12:
Try adding dtparam=audio=yes to your config.txt and rebooting. You can remove snd-bcm2835 from /etc/modules, as it no longer works without Device Tree and using DT will cause it to be loaded automatically.

popcornmix, 2016-04-12:
> Try adding dtparam=audio=yes 

Isn't the usual syntax ``dtparam=audio=on``? Is ``yes`` also valid?

marcust, 2016-04-12:
That works, thanks.

I used the ```yes``` version.

Is there anywhere where there are release notes for the firmware? I guess the question is could I have figured that out anywhere?

closed 2016-04-12 by marcust

popcornmix, 2016-04-12:
You should find this file: https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README
on your sdcard.
In general optional drivers are enabled through device-tree rather than using modprobe with more recent kernels.
Lots more info here: https://www.raspberrypi.org/forums/viewforum.php?f=107

pelwell, 2016-04-12:
Yes - ""yes", "y", "on", "true", and non-zero are all equivalent here, but "on" is more usual.