raspberrypi/firmware #587: Screen resolution at hdmi_mode 83

AXDOOMER, 2016-04-13:
Hi,

The Raspberry Pi has a bug. In /boot/config.txt, when the hdmi_mode is 83, it should output the video at 1600x900 like it is said in the documentation, but it outputs at 1280x960. On the forum, I saw other users who had the same problem with this particular resolution and they couldn't do anything, so we have to use another hdmi_mode. 

Here is a screenshot: http://i.imgur.com/XcRF3xh.jpg

The documentation confirms that hdmi_mode 83 of hdmi_group 2 (DMT) should be 1600x900, not 1280x960: https://www.raspberrypi.org/documentation/configuration/config-txt.md

I have a workaround: I create my own mode (mode 87) using: hdmi_cvt=1600 900 60 3, but the bug remains because mode 83 isn't good. 

Regards,
Alexandre-Xavier

popcornmix, 2016-04-13:
I've just tried and:
```
pi@domnfs:~ $ tvservice  -s
state 0x120006 [DVI DMT (83) RGB full 16:9], 1600x900 @ 60.00Hz, progressive
pi@domnfs:~ $ fbset 

mode "1600x900"
    geometry 1600 900 1600 900 16
    timings 0 0 0 0 0 0 0
    rgba 5/11,6/5,5/0,0/16
endmode
```
So it looks like DMT mode 83 is 1600x900.
What evidence do you have that it is not outputting 1600x900? 

Just what the monitor reports? I'm not sure I'd trust that.

My monitor (Dell P2415Q reports 1600x900@60Hz) for this mode.

AXDOOMER, 2016-04-13:
Did you test like this?

1- sudo nano /boot/config.txt
2- change hdmi_group and hdmi_mode so you have:
```
hdmi_group=2
hdmi_mode=83
```
3- Reboot and check what your monitor reports. 

My Raspberry Pi is connected to my monitor using a HDMI to VGA adapter. 

How do you explain that mode 83 is not outputting at 1600x900, but that mode 87 (using hdmi_cvt=1600 900 60 3) outputs at the right resolution?

Tonight, I'll test with the same commands that you used to see if I get the same output.

popcornmix, 2016-04-13:
> Did you test like this?

Yes. I'm using a HDMI connected monitor.
If you are going through a VGA adapter then I'd say it's very likely that your monitor is reporting the resolution wrongly (a lot of information is lost in converting to an analogue signal).
Try it with a HDMI monitor/TV.

Ferroin, 2016-04-13:
@popcornmix No, no data relating to the EDID is lost in the conversion from HDMI to VGA (or to DP, or to DVI), provided that the adapter being used is properly designed.  The signaling used to communicate the EDID is exactly identical between all modern external display connection standards, and is a simple I2C connection.  If you're losing data, then the converter you're using is not standards compliant or is very poorly designed, and thus shouldn't be used.

That said, it would not be unreasonable to expect the monitor itself to be reporting a bad EDID, I've seen quite a few VGA monitors that report completely bogus EDID data, which is why the DRM subsystem in Linux lets you load arbitrary EDID's to override monitor settings, and most Windows drivers these days let you use 'custom resolutions'.

popcornmix, 2016-04-13:
The monitor will be reporting what it thinks the resolution is based on the pixel clock (which is ambiguous) and the timings of the blanking periods. This is all exact with digital HDMI signals, but approximate with analogue VGA signals.

I imagine the "reduced blanking" of mode 83 is confusing the VGA monitor.

Ferroin, 2016-04-13:
Ah, I misunderstood what you meant.  I thought you were talking about the EDID data getting garbled (which can cause things like this to happen sometimes, because it can affect how the kernel tries to drive the output in some drivers), not the loss from the D/A process.

I'd be willing to bet you're right though about the 'reduced blanking' thing being part of the issue.

AXDOOMER, 2016-04-13:
I'll try with "hdmi_cvt=1600 900 60 3  0 0 1" then. The last number is a Boolean that activates reduced blanking, so I'll see if it is the cause of the problem.

popcornmix, 2016-04-13:
Note that timings for DMT modes and CVT modes are not identical.
DMT mode 83 uses a 108MHz pixel clock
hdmi_cvt=1600 900 60 3 0 0 1 uses a 97.75MHz pixel clock
hdmi_cvt=1600 900 60 3 0 0 0 uses a 118.25MHz pixel clock

AXDOOMER, 2016-04-13:
hdmi_cvt=1600 900 60 3 0 0 1 still works. 

I set my Pi to mode 83 and then used "tvservice  -s" and "fbset", they both reported 1600x900. So the problem should be my monitor or my HDMI to VGA adapter. 

This is what I have: https://www.adafruit.com/products/1151

There are a lot of warnings about HDMI to VGA adapters, but it mostly concern power usages. The problem may be my monitor that gets the resolution wrong when it tries to guess it.

popcornmix, 2016-04-13:
Do you actually see a difference in the display between mode 83 and ``hdmi_cvt=1600 900 60 0 0 0``?
It is possibly just a bug in what the monitor display as the resolution and may not affect the pixels at all.

AXDOOMER, 2016-04-14:
You mean ```hdmi_cvt=1600 900 60 3 0 0 0``` ? You forgot a number. 

```
hdmi_cvt=1600 900 60 3 0 0 0
hdmi_cvt=1600 900 60 3 0 0 1
```

These two are the same, they render correctly, but mode 83 makes a black bar at the bottom of the screen and the image goes beyond the right side of my screen: http://i.imgur.com/XcRF3xh.jpg (In the screenshot, you can see there is a black bar at the bottom and my widgets are missing from the upper-right because they are outside of my screen)

popcornmix, 2016-04-14:
I've double checked the timings with the DMT spec and the behaviour on my Dell monitor. All is correct.
I believe the issue lies with your VGA monitor or VGA adaptor, so there is nothing we can fix.
If you can demonstrate the issue on a HDMI device that would be interesting, but I suspect that won't be possible.

AXDOOMER, 2016-04-14:
I'll try with another monitor, so I'll know if it is the adapter or the monitor.

Ruffio, 2016-06-29:
@AXDOOMER have you checked with another monitor and this issue can be closed?

AXDOOMER, 2016-06-29:
I have created my own video mode and it worked.

Ruffio, 2016-06-29:
Ok, then please close the issue. Thanks.

closed 2016-06-29 by popcornmix

ncraun, 2016-07-28:
> If you can demonstrate the issue on a HDMI device that would be interesting, but I suspect that won't be possible.

Hello, I have the same issue as AXDOOMER, but mine occurs with no VGA involved. tvservice -s confirms I have the mode set to 83 with resolution 1600x900, and so does fbset, but the monitor thinks the signal is 1280x960, and not 1600x900, and so the image is distorted and doesn't fill the screen.

I am connected to the monitor with an HDMI to DVI adapter.  The monitor is an Acer X203H bd. I have used this monitor on my x86 machines using the same adapter and the monitor displays 1600x900 correctly.

The really weird part is if I enable the Experimental OpenGL driver by enabling the vc4-kms-v3d dtoverlay, then the monitor and pi displays at 1600x900 correctly! However, AXDOOMER's custom HDMI mode does not work for me, it results in a black screen.

I have attached a few images to show what I'm talking about along with my edid dump from the monitor. 

EDID dump from monitor: Sorry it's a zip, but github wouldn't let me add a .dat file to the comment.
[edid.dat.zip](https://github.com/raspberrypi/firmware/files/387689/edid.dat.zip)

Display when using the experimental OpenGL Driver: [image]

Display when using Mode 83 on default: 
[image]

ncraun, 2016-07-28:
I got AXDOOMER's hdmi_cvt setting to work, so my monitor now works correctly with the pi. My immediate problem is solved, so you can close the issue if you want. I'm still curious why mode 83 wouldn't work with a nonvga 1600x900 monitor though.