raspberrypi/firmware #588: No ATAGs ?

uski, 2016-04-17:
Hi,

I'm not sure if this is a kernel issue, or a firmware issue. It's one of the two, and I can't diagnose to decide by myself.

I'm having trouble using an initrd. It appears that the kernel cannot find the ATAGs from the bootloader : [    0.383541] No ATAGs?

I'm using the Archlinux ARM distribution with the latest kernel.

Any idea ? From the source of the kernel, it looks like this dmesg message appears when the kernel cannot find the ATAGs passed from the bootloader. Which is not normal I believe... any clue ?

Thanks !

Debug info : 

Kernel version : 
Linux alarmpi 4.4.7-2-ARCH #1 Thu Apr 14 18:38:43 MDT 2016 armv6l GNU/Linux

Bootloader info : 
# /opt/vc/bin/vcgencmd version
Apr 12 2016 22:30:12
Copyright (c) 2012 Broadcom
version 124875c58e250c4dcee3c52e8971166f184fa2e0 (clean) (release)

0b72b8face7bff6ef577aac4c0b170ff  /boot/bcm2708-rpi-b-plus.dtb
e368495a3786a2788db80d9ae9cbb989  /boot/bcm2708-rpi-b.dtb
3437168bbe689b66a8cb692c4366012a  /boot/bcm2708-rpi-cm.dtb
0a844d5ceadba293d3945ef542851af9  /boot/bootcode.bin
446eae9a610bac2040c6488f4763f1ae  /boot/cmdline.txt
1c01b536d0024586358a1ecaf6460ac9  /boot/config.txt
3e43d1d38252d3aebb7b2d1cd27f1523  /boot/fixup.dat
16d628fe900bbc0fb0db860a2f597a45  /boot/fixup_cd.dat
d7f8bd1440faf25b687e9a1cdd8e723a  /boot/fixup_db.dat
1af61fe14b8f3d39ff39713583f43226  /boot/fixup_x.dat
45ca2b7213227a168644d0c52b773afc  /boot/initrd.gz
cc34cbceab36ffb0828a2942d9fea089  /boot/kernel.img
577b5b802b0bfbb602b1a9e9358cdb7c  /boot/start.elf
857cf92cd97be9dfd04f2e08bb0dcc53  /boot/start_cd.elf
382eb12f70d19ff90f87bb6c20f669c4  /boot/start_db.elf
69cd4fc1103eb1cddd25a75ad31e6f1b  /boot/start_x.elf

[root@alarmpi alarm]# cat /boot/config.txt
# See /boot/overlays/README for all available options

gpu_mem=64
initramfs initrd.gz followkernel
[root@alarmpi alarm]# cat /boot/cmdline.txt
root=/dev/mmcblk0p2 rw rootwait console=ttyAMA0,115200 console=tty1 selinux=0 plymouth.enable=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 kgdboc=ttyAMA0,115200 elevator=noop earlymodules=ili9325,fbcon ili9325 fbcon=font:VGA8x16

popcornmix, 2016-04-17:
The kernel uses either ATAGs or Device Tree, and in more recent builds we use Device Tree, hence the ``No ATAGs`` message. That is expected and not a sign of a problem.

Ruffio, 2016-06-29:
@popcornmix as there have been no further comments and it appears to be working as designed, I suggest that we close this issue.

closed 2016-06-29 by popcornmix