raspberrypi/firmware #590: A change around 4.4.6 causes SPI LCD to go blank (white screen)

alexruffell, 2016-04-21:
Somewhere after e484afecfa0ae694965ba33ed40bb864cd8f3aa6 update, my two RPIs (r2 & r3) running standard Jessie stopped displaying anything on their WaveShare 4" SPI LCD screens. The screens went completely white. By regressing to an older version of the RPI firmware the screen starts working again. Bare with me for the little technical details I can provide...

I was using WaveShare overlays https://github.com/swkim01/waveshare-dtoverlays and very basic settings in a few other files as detailed in swkim01's github page.

popcornmix, 2016-04-22:
Do you have SPI enabled? E.g. ``dtparam=spi=on`` in config.txt?
See:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=144087
https://www.raspberrypi.org/forums/viewforum.php?f=107
https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README

popcornmix, 2016-04-22:
If https://github.com/swkim01/waveshare-dtoverlays is providing custom dtb files, they will likely need updating for 4.4 kernel.

alexruffell, 2016-04-22:
Thanks! I opened an issue in the repository for the overlays. To answer your previous question, yes SPI is enabled and all my other settings seem to still be in places. When I downgrade firmware, it works without further action so I am guessing it is indeed a compatibility issue between the new firmware and the overlays.

pelwell, 2016-04-22:
Try these files:
[waveshare32b.dtbo](https://drive.google.com/file/d/0B8VsfKAD4-NOUEhUeEZNcHh1VVU/view?usp=sharing)
[waveshare35a.dtbo](https://drive.google.com/file/d/0B8VsfKAD4-NOMU5Pcy1hRWJmTmM/view?usp=sharing)

You may also get away with renaming your existing "name-overlay.dtb"s to "name.dtbo"s.

alexruffell, 2016-04-23:
Thank you! Renaming the file worked! Both RPIs are up and running now on the latest kernel. I also commented out dtparam=spi=on as I believe you mentioned somewhere that it was not needed given the overlay file takes care of it - and everything still works fine.

closed 2016-04-23 by alexruffell