raspberrypi/firmware #591: SET_OVERSCAN arguments are no longer the same as /boot/config.txt

Ealdwulf, 2016-04-26:
If you use the firmware interface, as we do here: https://github.com/KanoComputing/kano-settings/blob/master/overscan/overscan.c
to set the overscan settings, it used to be the case that the individual arguments set the bottom, top, left right distance, as they do in /boot/config.txt. But on the most recent raspbian, and also the most recent firmware from rpi-update, if you set only one argument and leave the others zero, once it gets above some threshhold, the other arguments also increase. I'm not sure when this changed.

 I thought this might be the intended behavior, except that /boot/config.txt does not seem to do the same. What happens is that if you increase 'left' then 'right' is the only distance which doesn't change, etc.

To reproduce:
* Download https://github.com/KanoComputing/kano-settings/blob/master/overscan/overscan.c
* `gcc -Wall -std=c99 -o overscan overscan.c`
* `for ((i=0;i<100; i++)) ; do ./overscan $i 0 0 0   ; done`

This means that SET_OVERSCAN can not be used any more to give a live demonstration of what setting the config entries will do.

Ealdwulf, 2016-04-26:
(Updated description to correct a link)

popcornmix, 2016-04-27:
You say "no longer". Can you point to an older firmware that behaved differently?
Are you using ``disable_overscan=1`` in config.txt?
That would likely make behaviour more consistent, otherwise the default is overscan=48,48,48,48.
Note that aspect ratio is framebuffer is preserved. That is why "right" parameter may change when "top" changes.

Ealdwulf, 2016-04-29:
Yes: version 2fe636ae3f5e82c6d0b31e32fa524a07821c440a allowed each setting to be set individually.

We are using `disable_overscan=1`. However setting it to 0 doesn't seem to make any difference.

If we set `overscan_left=0 overscan_right=0 overscan_top=100 overscan_bottom=0 disable_overscan=1`, using SET_OVERSCAN, this is the result:
[image]
But after these are written back to config.txt and we reboot, the result is different:
[image]
Changing disable_overscan to 0 gives the same result.
(That is using firmware efdcf1683e7c696e53fa28ab13c973cb32f097920) 

I don't imagine that many people set the overscan differently for the different sides, but I guess that it's not intended for SET_OVERSCAN to behave differently to the functionality on boot. I guess we (kano) don't mind that much which behavior is used. However, we are emulating the overscan setting in some other apps which have their own dispmanx layer (we don't have overscan_scale=1 currently) so it would be useful to know how the final offsets are calcuated - expecially if you change the firmware to do the same aspect ratio calculation after a reboot.

Ruffio, 2016-06-29:
@Ealdwulf do you experience this with latest firmware? I see a clear difference between setting disable_overscan or not

zeekoe, 2016-06-30:
Hello!

I'm not sure if this is the correct location to report, but it seems my problem is related. I am using an old DVD-player-panel with composite input with the following configuration:

framebuffer_width=480
framebuffer_height=236

disable_overscan=1

overscan_top=-63
overscan_bottom=-50
overscan_left=15
overscan_right=17

It worked perfectly in Wheezy, but it seems that top + bottom is totally ignored in Jessie.

It's on the Raspberry used in my homebuilt coffee machine. Now it looks like this:

[image]

While it used to be like this:

[image]
http://koffie.ronaldteune.nl/

I just did an apt-get update + upgrade, but no change. After the upgrade I did not try changing top + bottom, and also did not try the dynamic overscan setting script. Before this upgrade but after the jessie upgrade a month ago, I tried the dynamic script and also manual change + reboot extensively. In no way I could get my top and bottom black borders to change.

popcornmix, 2016-07-01:
What do you expect:
```
overscan_top=-63
overscan_bottom=-50
```
to do? overscan settings describe how many pixels to not use on each side of display.
They should never be negative - that doesn't mean anything.

Technically, if ``disable_overscan=0`` then you may (depending on display mode) have a default positive overscan, which negative overscan settings can reduce, but it would always be recommended to use ``disable_overscan=1`` and just use positive settings.

There was a bug long ago where negative overscan settings did something incorrect, and it looks like your settings were relying on that. But they make no sense without the bug in place.

Start with:
```
disable_overscan=1
overscan_top=0
overscan_bottom=0
overscan_left=0
overscan_right=0
```
and if there are pixels that should be present then gradually increase the numbers (positive only).

Ealdwulf, 2016-07-04:
@Ruffio For me, the behavior is the same irrespecive of what disable_overscan is set to. I guess it depends on the EDID information from your display. 

In case it's not clear - the issue I reported above is not related to the use of negative overscan values.

zeekoe, 2016-07-04:
@popcornmix: thanks very much for looking into this.
@Ealdwulf: sorry for accidentally hijacking your thread.
Continued in https://github.com/raspberrypi/firmware/issues/627 because of new issue.

Ruffio, 2016-08-07:
@Ealdwulf as I read it, you reference an issue with another program KanoComputing ( https://github.com/KanoComputing/kano-settings/blob/jessie/overscan/overscan.c). I think, you should report your issue there...

Ealdwulf, 2016-08-16:
@Ruffio We maintain that (although it's forked from somewhere else). We could fix it there. However we'd have to reimplement the transform between the values in config.txt and the raw offsets. Would it be possible to post the snippet of code which implements that in the firmware, @popcornmix, to save us reverse engineering it?

JamesH65, 2017-12-21:
@Ealdwulf Is this still outstanding? Or can it be closed?

jtanderson312, 2017-12-21:
@JamesH65 Still outstanding

The response to modifying the overscan_* variables either through the vcio mailbox, or the config.txt in /boot did change.   Setting even a positive value to overscan_left makes the screen push left and down from the top of the screen with the newer versions of Raspbian.

I was using the image from raspbian-2015-02-17 (YES, ancient)  When modifying the overscan_left value, it would just push/pull the screen's left edge.  A negative in overscan_left would allow the framebuffer edge to be moved towards the edge of the screen more, and a positive value would allow it to be pushed to towards the center of the monitor.  

Even if negative values are not to be allowed in the future, something is still wrong with the overscan_* values.   If the modifying these values are giving expected results, a bit of documentation or clarification could help us all out.

JamesH65, 2018-06-29:
@jtanderson312 Can you test with the latest Raspbian, and report back if still an issue?

Otherwise, this issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

jtanderson312, 2018-07-16:
I spent a bit of time with the lastest Raspbian image and have determined that the problem still exists.   I used the realtime tweaker of overscan in https://github.com/KanoComputing/kano-settings/tree/master/overscan to test changing the overscan values.

What I have determined is that the newer versions of Raspbian maintain the geometry (width/height) of the frame buffer and offset using the overscan values.   I remember in the old versions that "worked" modified the size of the frame buffer.    By doing this, it will change the aspect ratio of the frame buffer to stretch/shrink the display area to fit  the monitor.    

Honestly, I'm can see why this was modified from the original code.   Maintaining the aspect ratio of the frame buffer seems to be the problem.    I'm not sure this can or will be worked out correctly.

Let me know if I can help with any more clarification.

Thank you for revisiting this problem!  I appreciate it.

pelwell, 2018-07-16:
The aspect-ratio-maintenance feature was noticed by @spl237 a week or so ago, but I didn't hear the outcome of the discussion with @popcornmix.

spl237, 2018-07-16:
No real conclusions drawn from my point of view - it seems that there is no trivial way to use the mailbox parameters to set overscan in real-time to match the effect of the disable_overscan parameter. As it was a nice-to-have rather than anything vital, I've dropped the relevant change.

For what it's worth, I think it would be better if the effect of disable_overscan could be accurately duplicated by using the individual values for left, top etc, but it's not a disaster from my POV if it can't.

closed 2025-02-14 by popcornmix