raspberrypi/firmware #594: vcdbg --help could be better

dlech, 2016-05-03:
Here is the output of `vcdbg --help`

```
Usage: vcdbg [options] command arguments
 
Where options is one of:
  -o, --output filename   Open filename instead of using the videocore memory
  -f, --follow            Output new data every 5 seconds as it arrives
  -h, --help              Print this information
 
Use the command help to get a list of available commends
or help followed by a command name to get help on a 
particular command.
```

1. How do you "Use the command help to get a list of available commends"? The exact command should be presented verbatim so we don't have to guess. i.e. `sudo vcdbg help`. Or at least put `help` in quotes.
2. "commends" is misspelled - should be "commands".
3. `vcdbg --version` does not work, so I don't know which version I am using - if that matters

closed 2019-07-24 by popcornmix

shib, 2019-07-29:
[https://github.com/raspberrypi/firmware/issues/1212](url)
Can I ask the question above?

pelwell, 2019-07-29:
What is the point of posting a comment on a 3-year-old issue with a link to another new issue _in the same repo_? Don't you understand that the same people will see both messages and think you are just impatient and annoying?