raspberrypi/firmware #597: feature request: Add simplefb device tree node to get display before vc4/fb_bcm2708 load

anholt, 2016-05-09:
In today's pull request to the kernel I've included the vc4 change to add support for transition from simplefb to vc4.  With a similar change to fb_bcm2708, the firmware could start adding a node to the DT same as u-boot does, and then we would have dmesg output starting from early in the boot process.

References:

vc4 commit for handoff: upstream b3a15f6d55fb584dd4d8baac5d1b6a398720620c or downstream 76d865faade400e7a1842bd04a17358cda0e0091

simplefb node binding: Documentation/devicetree/bindings/display/simple-framebuffer.txt

u-boot simplefb node addition implementation: http://git.denx.de/?p=u-boot.git;a=blob;f=common/fdt_support.c;h=42e5d8a1d2f697ae9509d1767fce7a7a28a00a2b;hb=HEAD#l1545

pelwell, 2016-05-10:
This is probably a dumb question, but what part does simple-framebuffer play in the fb_bcm2708 -> vc4 handover? Or are you suggesting that the firmware always creates the framebuffer, configures a simple-framebuffer, and then the main display driver (either fb_bcm2708 or vc4) takes over?

The flow is a bit convoluted as it stands, with the firmware using module parameters to tell fb_bcm2708 the dimensions and format the frame buffer should be, which then asks the firmware to allocate it, so a simplification here sounds like a good idea.

anholt, 2016-05-10:
> Or are you suggesting that the firmware always creates the framebuffer, configures a simple-framebuffer, and then the main display driver (either fb_bcm2708 or vc4) takes over?

This, exactly.

popcornmix, 2016-05-10:
> The flow is a bit convoluted as it stands, with the firmware using module parameters to tell fb_bcm2708 the dimensions and format the frame buffer should be, which then asks the firmware to allocate it, so a simplification here sounds like a good idea.

Note that fb_bcm2708 can reconfigure the size and format of the framebuffer (with fbset for example), so I'm not sure this can be simplified much.

pelwell, 2016-05-10:
> Note that fb_bcm2708 can reconfigure the size and format of the framebuffer (with fbset for example), so I'm not sure this can be simplified much.

I understand that, but the data flow will be more logical.