raspberrypi/firmware #599: Transmission on serial port /dev/ttyAMA0 sporadically stops.

dimovnike, 2016-05-13:
When actively using RTS line, transmission sporadically stops until port is closed and reopened. 

Happens on firmware:
```
/opt/vc/bin/vcgencmd version
Feb 29 2016 14:23:50 
Copyright (c) 2012 Broadcom
version dea971b793dd6cf89133ede5a8362eb77e4f4ade (clean) (release)
```
and kernel `4.1.18-v7+`

pelwell, 2016-05-13:
We've now moved on to the 4.4 kernel - can you update your firmware and confirm that the problem still exists?

dimovnike, 2016-05-13:
version 1bf9a9a77026af9128a339c82d72e331d3532ee4 is also affected. 
will update to last version and check it

dimovnike, 2016-05-13:
Last version 280c320a06a8ee014c9ea8df8c1350daa2c50d25 /  4.4.10-v7+ is affected too.

What else can I send you to help fixing this bug? Will a C file help that trigggers the bug help? However you will have to manually "operate" the RTS line fast. Also sometimes it takes a couple of minutes to reproduce it (sometimes it reproduces in 10 seconds).

dimovnike, 2016-05-13:
Will a coredump of the locked executable help? The executable only sends with write() and then waits with tcdrain(). The tcdrain() is not needed for he bug to appear but it helps reproducing it (the initial setup had no tcdrain() and the transmission was stopping too).

pelwell, 2016-05-13:
C code and clear instructions will definitely help.

I have an idea - why not modify your test program to use another GPIO as an RTS line, so that all one has to do is connect the two pins with a patch cable.

Rather than a core dump, you could use the sysrq feature to dump the thread state.
```
echo t > /proc/sysrq-trigger
```
See https://www.kernel.org/doc/Documentation/sysrq.txt.

dimovnike, 2016-05-13:
Hi, here is the thread state attached. The locked process is gprs_simemu.

[threadstate.txt](https://github.com/raspberrypi/firmware/files/263685/threadstate.txt)

pelwell, 2016-05-13:
How do you enable the RTS0 and CTS0 functions on GPIOs 16 and 17?

What is your wiring?

dimovnike, 2016-05-13:
I enable RTS/CTS using this code (also please see the wiring in comments):

```
uint8_t pio_alt_tx = RPI_BPLUS_GPIO_J8_08; //     = 14,  ///< B+, Pin J8-08
uint8_t pio_alt_rx = RPI_BPLUS_GPIO_J8_10; //     = 15,  ///< B+, Pin J8-10
uint8_t pio_alt_cts = RPI_BPLUS_GPIO_J8_36;   //     =  16, ///< B+, Pin J8-36,
uint8_t pio_alt_rts = RPI_BPLUS_GPIO_J8_11;   //     = 17,  ///< B+, Pin J8-11

bcm2835_gpio_fsel(pio_alt_rts, BCM2835_GPIO_FSEL_ALT3);
bcm2835_gpio_fsel(pio_alt_cts, BCM2835_GPIO_FSEL_ALT3);

//set alternate functions for TX / RX
bcm2835_gpio_fsel(pio_alt_tx, BCM2835_GPIO_FSEL_ALT0);
bcm2835_gpio_fsel(pio_alt_rx, BCM2835_GPIO_FSEL_ALT0);

```

dimovnike, 2016-05-13:
Please also note that the RTS line works well until the lock (it really stops transmission when high).

dimovnike, 2016-05-13:
**First I have to rectify, i was taliking about the CTS (input) line (pin 36). ** Sorry for this misunderstanding.

Here is the code that reproduces the bug (but I did not have time to implement a secondary thread that would operate the CTS line).  It has to be compiled (instructions are in the file gprs_simemu.c, search by keyword README).

Note that the CTS line allows transmission when not connected (if unsure, connect to ground to allow transmission). To stop transmission, it has to be connected to 3.3V (pin 1). 

In my tests, i start the binary with the CTS disconnected then I start donnecting the CTS line (pin 36) and after some (sometimes 20-30 ) tries the binary stops transmitting. 

[rpi_sreial_bug.zip](https://github.com/raspberrypi/firmware/files/263743/rpi_sreial_bug.zip)

dimovnike, 2016-05-16:
Can someone confirm that they reproduced the bug? Is there anything else that I can add to help fixing this bug?

pelwell, 2016-05-16:
Patience - it's on my list. I just have one or two things to do first after the weekend.

dimovnike, 2016-05-16:
Ok, thanks!

pelwell, 2016-05-18:
I've been able to reproduce the stall, and the good news is that while it's in that state I can write characters directly to the data FIFO and, provided CTS is low, they are transmitted, i.e. this isn't a hardware fault.

dimovnike, 2016-05-18:
This is good news, thanks.

dimovnike, 2016-06-18:
Any news about this? Anything I can help with?

Ruffio, 2016-08-07:
@pelwell anything on this issue?

pelwell, 2016-08-07:
I was able to reproduce the issue, but never got enough time to find the bug.

JamesH65, 2017-12-21:
@pelwell Is this fixed by those recent (ish) changes to the UART stuff that fixed the BT stack?

pelwell, 2017-12-21:
I think the "cts-event-workaround" patch (see https://github.com/raspberrypi/linux/issues/1280 for details) is likely to have solved the lockup problem - the other patches were to fix data loss of various kinds.

@dimovnike Please retest if you can.

JamesH65, 2018-06-29:
Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

Closing this issue as questions answered/issue resolved.

closed 2018-06-29 by JamesH65