raspberrypi/firmware #600: Problems with overlay filenames now without -overlay in the name

msperl, 2016-05-14:
Hi!

With the last major release of the firmware there was a change that renamed:
mcp2515-can0-overlay.dtb to mcp2515-can0.dtb.

Any user that still uses the old valid:
`dtoverlay=mcp2515-can0-overlay,oscillator=16000000,interrupt=16`
would still use the old overlay file (which does not get removed by rpi-update and may be way out of date) and then - for some changes it would fail.

The old dt overlay did not load because of the missing "&spidev0" alias.
this was introduced with: 666d717a56639ca89b6bba18a325b1c530683fac

```
root@rasp2:~# vcdbg log msg 2>&1 | grep mcp2515
001557.525: Failed to load overlay 'mcp2515-can0-overlay'
```

The only solutions was:
* moving to the name mcp2515-can0-overlay

Probably `rm -f /boot/overlay/*-overlay.dtb` would also work.
(Anyway: i did both to be on the save side)

It took me more than 1 hour to figure out why all my dt modifications I made were not accepted - others will fall into the same trap!

I guess this will produce headache with several people and their "documentation" that still uses the "-overlay" names - think of the forum that does not allow modifications of posts to update such info...

You may want to think about how to solve that.

This may require that the firmware would be testing if both files exist, and if the *.dtbo file is newer than the *-overlay.dtb file then use that one.

At the very least this may mean headaches for vendors supporting their product!

pelwell, 2016-05-14:
'-overlay. names have never been recommended - they just happen to have worked previously. I have repeatedly told people not to do it, including when I  posted in the Device Tree forum about the upcoming changes. On top of that, the 4.4 kernel has been available for testing for months. I'm surprised you, of all people, missed all of those.

msperl, 2016-05-14:
I actually got asked by some CAN hardware vendor, so I investigated and found this issue.

Myself I prefer upstream and do not use the overlays all that often.

I also know that both options were equally working, but the "-overlay" closer resembled what was on disk, so I guess more people are using this as it seemed natural to map configs to files.

I understand that you want to make this consistency move in the right (recommended) direction,
but I just wanted to point out that this is biteing a few people who are not updating the firmware regularly, but wanted the latest updates because of this big raspbian update.

I guess cleaning up the old files might be an idea to help people avoid this situation - or trigger some warning - however the firmware can do this...

Possibly a simple motd change for any overlay load issue could get created after boot?
That may be a good thing in general to give people hints about something going wrong...

Just thoughts....

msperl, 2016-05-14:
Yes - I also remember the history behind -overlay:
* initially
  * dts originally sitting in arch/arm/boot/dts and not the overlay subdirectory
  * dtbs sitting in /boot directly
* then the move to overlay directory for both cases - but unfortunately no rename at the same time
  * with that also came a change in the dtb extension to dtbo for the binaries (maybe also separate - cant remember that detail)
* now the rename, but unfortunately leaving the old -overlay.dtb files in place...

Ruffio, 2016-07-24:
@msperl can this issue be closed?

Ruffio, 2016-07-29:
@pelwell  please consider to close this issue. Thanks.

closed 2016-07-30 by msperl