raspberrypi/firmware #604: missing some optional kernel feature to support container

vanloswang, 2016-05-22:
On raspberry 3 running fedora 23, following below setups.
1. wget https://raw.githubusercontent.com/dotcloud/docker/master/contrib/check-config.sh and chmod it to be able to execute.
2. modprobe configs
3. execute check-config.sh and the infomations is as follow:
info: reading kernel config from /proc/config.gz ...

Generally Necessary:
- cgroup hierarchy: properly mounted [/sys/fs/cgroup]
- CONFIG_NAMESPACES: enabled
- CONFIG_NET_NS: enabled
- CONFIG_PID_NS: enabled
- CONFIG_IPC_NS: enabled
- CONFIG_UTS_NS: enabled
- CONFIG_DEVPTS_MULTIPLE_INSTANCES: enabled
- CONFIG_CGROUPS: enabled
- CONFIG_CGROUP_CPUACCT: enabled
- CONFIG_CGROUP_DEVICE: enabled
- CONFIG_CGROUP_FREEZER: enabled
- CONFIG_CGROUP_SCHED: enabled
- CONFIG_CPUSETS: enabled
- CONFIG_MEMCG: enabled
- CONFIG_KEYS: enabled
- CONFIG_MACVLAN: enabled (as module)
- CONFIG_VETH: enabled (as module)
- CONFIG_BRIDGE: enabled (as module)
- CONFIG_BRIDGE_NETFILTER: enabled (as module)
- CONFIG_NF_NAT_IPV4: enabled (as module)
- CONFIG_IP_NF_FILTER: enabled (as module)
- CONFIG_IP_NF_TARGET_MASQUERADE: enabled (as module)
- CONFIG_NETFILTER_XT_MATCH_ADDRTYPE: enabled (as module)
- CONFIG_NETFILTER_XT_MATCH_CONNTRACK: enabled (as module)
- CONFIG_NF_NAT: enabled (as module)
- CONFIG_NF_NAT_NEEDED: enabled
- CONFIG_POSIX_MQUEUE: enabled

Optional Features:
- CONFIG_USER_NS: missing
- CONFIG_SECCOMP: enabled
- CONFIG_CGROUP_PIDS: missing
- CONFIG_MEMCG_SWAP: missing
- CONFIG_MEMCG_SWAP_ENABLED: missing
- CONFIG_MEMCG_KMEM: missing
- CONFIG_BLK_CGROUP: enabled
- CONFIG_BLK_DEV_THROTTLING: enabled
- CONFIG_IOSCHED_CFQ: enabled
- CONFIG_CFQ_GROUP_IOSCHED: enabled
- CONFIG_CGROUP_PERF: missing
- CONFIG_CGROUP_HUGETLB: missing
- CONFIG_NET_CLS_CGROUP: enabled (as module)
- CONFIG_CGROUP_NET_PRIO: missing
- CONFIG_CFS_BANDWIDTH: missing
- CONFIG_FAIR_GROUP_SCHED: enabled
- CONFIG_RT_GROUP_SCHED: missing
- CONFIG_EXT3_FS: missing
- CONFIG_EXT3_FS_XATTR: missing
- CONFIG_EXT3_FS_POSIX_ACL: missing
- CONFIG_EXT3_FS_SECURITY: missing
    (enable these ext3 configs if you are using ext3 as backing filesystem)
- CONFIG_EXT4_FS: enabled
- CONFIG_EXT4_FS_POSIX_ACL: enabled
- CONFIG_EXT4_FS_SECURITY: enabled
- Network Drivers:
  - "overlay":
    - CONFIG_VXLAN: missing
- Storage Drivers:
  - "aufs":
    - CONFIG_AUFS_FS: missing
  - "btrfs":
    - CONFIG_BTRFS_FS: enabled (as module)
  - "devicemapper":
    - CONFIG_BLK_DEV_DM: enabled (as module)
    - CONFIG_DM_THIN_PROVISIONING: enabled (as module)
  - "overlay":
    - CONFIG_OVERLAY_FS: enabled (as module)
  - "zfs":
    - /dev/zfs: missing
    - zfs command: missing
    - zpool command: missing

From the above informations, some optional kernel feature is missing, maybe ovs and cadivisor will work abnomal.Do we need to fix it?

popcornmix, 2016-05-23:
See: https://github.com/raspberrypi/linux/pull/1020 https://github.com/raspberrypi/linux/pull/1027
Some optional features have not been included as they impact performance in the kernel for all users (very few will actually be using docker).
You'll have to describe what you think the problem is. The features are optional for a reason - they don't have to be enabled to use docker.

Ferroin, 2016-05-23:
I'm pretty certain that the HugeTLB cgroup stuff is pointless on a Pi (I'm pretty certain that none of the SoC's used support hugepages)

The perf cgroup stuff is not likely to be used even by the people who would be using Docker.

The memcg stuff takes up significant space when enabled for accounting.

The other cgroup stuff probably should be enabled though (especially the scheduler stuff, that's _very_ useful outside of container usage).

The user namespace stuff is a potential security issue (it enables suage of containers, but it also exposes an insane amount of things to regular users that they normally wouldn't (and in some cases, arguably shouldn't) see).  It's worth pointing out though that some apps other than Docker need to be run as root to work correctly without this (Chromium for example can't do proper sandboxing of it's internal processes without either user namespace support, being run SUID root, or having CAP_SYS_ADMIN set).

I don't know that the VXLAN stuff would work with the Pi's Ethernet adapter.

The EXT3 stuff is bogus, the ext4 driver provides all the support that is needed.

AUFS is a __huge__ module, and isn't upstream either,

ZFS is entirely third-party, and I don't think that it will run as well as you would think on a Pi (even with all the features turned off, it's pretty resource intensive).

vanloswang, 2016-05-24:
@popcornmix @Ferroin Also, cgroup cpusets is not mounted in kernel 4.4.11-v7+ #887, so kubernetes will not work normally.
https://github.com/kubernetes/kubernetes/issues/26038
I test cpusets will kernel 4.4.11-v7+ #887 and fedora 23 armv7l edition, it appear once after i add following configs in /etc/cgconfig.conf.

mount {
        cpuset  = /cgroup/cpuset;
        cpu     = /cgroup/cpu;
        cpuacct = /cgroup/cpuacct;
        memory  = /cgroup/memory;
        devices = /cgroup/devices;
        freezer = /cgroup/freezer;
        net_cls = /cgroup/net_cls;
        blkio   = /cgroup/blkio;
}

But after I reboot my rpi3, the cpusets will nerver be mounted. So I confused, why cpuset not mounted automatic? how to mount it?

vanloswang, 2017-01-13:
turn to Pine A64.

closed 2017-01-13 by vanloswang

DieterReuter, 2017-01-13:
@vanloswang we're running k8s on the Pi3 successfully and we do have CGROUP support included in our kernel as well. See latest blog post https://blog.hypriot.com/post/setup-kubernetes-raspberry-pi-cluster/.

If you need some special feature you do have to compile your own kernel, that's not too complicated at all.

alexellis, 2017-08-22:
@popcornmix @vanloswang any chance of reconsidering this? Given that my blog post on a cluster with Raspberry Pi spent most of the day at number 1 on Hacker News for Docker Swarm - it would be good to get Kubernetes working too. https://blog.alexellis.io/your-serverless-raspberry-pi-cluster/

joejulian, 2019-03-17:
> very few will actually be using docker

Which is guaranteed because the needed kernel features are disabled.

Most users, rather than trying to do things like build a raspberry pi docker swarm, or kubernetes cluster and meeting this deficiency - especially since the error message from runc doesn't have any actionable text - will give up and build on NUCs instead.

Personally, I feel that ensuring these use cases will fail is doing a disservice to the community.

JamesH65, 2019-03-18:
Notwithstanding you are reponding to a very old closed thread, adding this support means an overhead for every Raspbian user, not just those who want to use docker. Since the number of Raspbian users not needing to use Docker vastly exceeds the number who will want to use Docker, the decision not to include does seem valid.

pelwell, 2019-03-18:
There are several distributions available for the Pi with some of the more power-user-focused features enabled - OpenSUSE, Arch, etc. Perhaps one of them would be a better fit for your needs?

JamesH65, 2019-03-18:
Or, it's pretty easy to build you own kernel  on the Pi, that might be a useful approach.  https://www.raspberrypi.org/documentation/linux/kernel/README.md

alexellis, 2019-03-18:
I am currently running Kubernetes on the Raspberry Pi 3, it would be awful for the community if support was deliberately removed.

https://github.com/alexellis/k8s-on-raspbian

pelwell, 2019-03-18:
I am not aware of any features being deliberately removed, but sometimes changes to the kernel config files means features get accidentally dropped, and sometimes newer kernel versions require additional options to be enabled in order to continue to support features that have worked previously.

If you are currently running Kubernetes on RPi3 running Raspbian, which Linux version is that? On which version does it no longer work?