raspberrypi/firmware #605: Lirc used simultaneously with bluetooth causes hci0 errors

jsayres, 2016-05-23:
I've got a Raspberry Pi 3 with Raspbian and the latest firmware. I'm trying to set it up to receive commands over bluetooth, translate them to IR signals, and send them out using LIRC.

While receiving bluetooth signals on their own, everything works great. Using LIRC on its own works great as well. But the instant I use LIRC to send while simultaneously receiving bluetooth signals, the system becomes unstable. When I had firmware version 4.1.x, the Pi would completely crash, but since updating to 4.4, it now just seems to limit itself to corrupting the bluetooth connection. Specifically, when using dmesg after the issue rears its head, I can see there are a bunch of "frame reassembly failed" errors on hci0. Resetting the bluetooth connections allows things to work again, but only for around 10-20 seconds each time.

I'm a Pi n00b, as well as a mere Linux beginner, so I'm sure I've left out some rather crucial information. (And hopefully I'm posting this in the correct place! But given the behavior change caused by the firmware upgrade, I don't think I'm too far off base.) So please let me know whatever additional information is needed to help or, if this isn't the correct place to raise this issue, where I should go instead.

pelwell, 2016-05-23:
lirc_gpio generates 10s of thousands of interrupts per second. The UART used for Bluetooth has small FIFOs by modern standards (only 16 characters), and because of a shortage of pins there is no flow control. Problems with the firmware and BlueZ stack have so far prevented us from using the H5 protocol (which designed to cope with data loss) - that's a problem I'd like to solve, but until then the best we can hope for is not crashing under heavy CPU load and/or high data throughput.

jsayres, 2016-05-23:
For my purposes, I only need to send out the IR commands about 5x per second, and they take a maximum of 100ms to complete. So if I could alternate every 100ms between reading the bluetooth data and ignoring it, that would work for me. For the period where I'd be ignoring it, data loss is fine.

I guess the question is, can I temporarily "suspend" the bluetooth connection and simply allow it to drop packets during the time LIRC is doing its work?

JamesH65, 2017-12-21:
@pelwell How about this one?

pelwell, 2017-12-21:
Without flow control there is always a possibility of data loss under load, but we have recently fixed a number of problems in the UART driver that resulted in data loss. It is worth retesting with a recent kernel (November onwards - the current Raspbian kernel should be fine).

JamesH65, 2017-12-21:
@jsayres Are you able to recheck this with the latest kernel?

JamesH65, 2018-06-29:
Please update to the latest kernel which may contain a fix for this issue.

Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

closed 2018-06-29 by JamesH65