raspberrypi/firmware #606: Kernel Panic on fedora

sedrubal, 2016-05-28:
I did [this instructions](https://fedoraproject.org/wiki/Raspberry_Pi) to install Fedora on Raspberry PI. But even on the first boot there is a kernel panic. I tried `Fedora 23` and `Fedora 24 Beta` and `raspberry firmware 1.20160506` (with kernel 4.4.9) and current master (with kernel 4.4.11). It didn't changed anything.

After 3 seconds after mounting root readonly and devtmpfs:

```kernel
[...]
Freeing unused kernel memory: 300K (c07bb000 - c081a000)
[...]
hub 1-1:1.0 3 ports detected
Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000004

CPU: 0 PID: 1 Comm: init Not tainted 4.4.9+ #884
Hardware name: BCM2708
[<c0016c9c>] (unwind_backtrace) from [<c0013c20>] (show_stack+0x20/0x24)
[<c0013c20>] (show_stack) from [<c02e319c>] (dump_stack+0x20/0x28)
[<c02e319c>] (dump_stack) from [<c00d7040>] (panic+0x8c/0x1f4)
[<c00d7040>] (panic) from [<c00243c8>] (do_exit+0x4c/0xc0)
[<...>] (do_exit) ...
[<...>] (do_group_exit) ...
[<...>] (get_signal) ...
[<...>] (do_signal) ...
[<...>] (do_work_pending) ...
---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000004]
```

I did set `SELINUX=0` and touched `/.autorelabel` but I think it seems not to be a SELinux problem...

popcornmix, 2016-05-31:
This isn't a firmware issue. You should report this to the fedora maintainers.

closed 2016-05-31 by popcornmix

sedrubal, 2016-05-31:
But it's [`boot/kernel.img`](boot/kernel.img) which has a kernel panic I think. Why do you think it's a problem of fedora arm?

popcornmix, 2016-05-31:
See: http://www.tldp.org/LDP/intro-linux/html/sect_04_02.html

> The kernel, once it is loaded, finds init in sbin and executes it.

init is part of the distribution (i.e. fedora) and that is what the panic is about. It is getting killed. I don't know why, but it it is something the fedora people will know more about.

zardoz99, 2016-07-01:
I've done some research on this today. It appears that the binaries supplied with the Fedora ARM images are not recognized by the kernel from this repository. The panic is, I believe, due to the inability to execute the /lib/systemd/systemd by the kernel. This is now now what /sbin/init links to. This may be due to missing configuration options when the kernel is built but I have not gone that deep yet.

Geoff160, 2016-07-25:
Any new news regarding this one?

clivem, 2016-07-25:
> The other possibility is that the Fedora ARM images are not suitable for anything lower than a Pi 2. I'm using a B+. 

> Any new news regarding this one?

What sort of news do you want? Confirmation that a armv7hl systemd binary isn't going to work on a B+?

Ruffio, 2016-07-25:
@Geoff160 @popcornmix wrote: 

> it it is something the fedora people will know more about.

'

zardoz99, 2016-07-25:
I figured out that ARM 7 was a minimum requirement very quickly. It was a pity that the Fedora page has a reference to the Raspberry Pi as well as PI2 under the Prerequisites section. It will only work on the later models.

MitchDresdner, 2016-09-28:
Well, you get the rights of first discovery, but I got same panic trying to install on Pi Zero, armv61.

MitchDresdner, 2016-09-29:
Should add that it boots perfectly on Pi 2 Arm7, but i'll be giving up on the server edition and try a desk top instead. Creating and acceptable Root password wore me out, complaining about using a dictionary word, when the only comparison to real a word is that they both have letters. The Root password would need to be a cryptographic key which could be later entered by a machine if at all.  It's so safe that not even an administrator can log in, if it didn't support network connections it would even be safer.