raspberrypi/firmware #609: camera gets jammed up after several weeks of recording

AndySchroder, 2016-06-02:
I'm having issues where the camera gets jammed up after several weeks of recording. This is with the v1 camera. I'm not yet sure if the problem happens with the v2 camera. I record an image every 1 second at the camera's maximum resolution of 2592x1944. I am using picamera. At first I thought the issue was with picamera because restarting the python interpreter would resolve the problem. After putting in some logic to restart python when this happens, I then still regularly reach a situation on multiple raspberry pis where the camera cannot be used by either raspistill or picamera without performing a reboot.

More details and some outputs of **vcdbg reloc** at various times can be found at this issue: https://github.com/waveform80/picamera/issues/242 .

I'm pretty confident there is a problem with the camera module and/or firmware because a reboot is required for both raspistill and picamera, although there may also be a different problem with picamera that required the python interpreter to be restarted, or the same problem may just be triggering an earlier symptom in picamera. It's hard to tell without first getting the camera module and/or firmware working without ever requiring a reboot.

I am using raspberry pi 2s. I did run rpi-update a few weeks ago and the problem still persisted. I have not yet done extensive testing on raspberry pi 3.

6by9, 2016-06-02:
Do you actually have a simple test case that you can share?

Please add ```start_debug=1``` to /boot/config.txt, and post the output from ```sudo vcdbg log assert``` when it fails. ```sudo vcdbg malloc``` may also show something.

Your vcdbg reloc outputs all look normal. "Block" is just referencing that there are 2 gaps in the heap, so 3 blocks of contiguous allocation. The largest free contiguous block is also being reported.

Ruffio, 2016-07-23:
@AndySchroder any news on the start_debug=1 ? Else please close this issue if is has been solved.

Ruffio, 2016-08-07:
@AndySchroder Anything? Can this issue be closed?

AndySchroder, 2016-08-08:
Sorry for the delayed reply. As mentioned it takes several weeks for the problem to occur. I had posted some more information here, but have not yet had time to do everything you asked.

https://github.com/waveform80/picamera/issues/242#issuecomment-232094063

I had a power failure so the raspberry pi was forced to reboot since the comments in the above link, so my test was restarted.

I can't say that the problem is gone yet, because it takes so long to reproduce and is slightly random.

Ruffio, 2016-08-08:
@AndySchroder but can't you provide a testcase that can demonstrate the issue? Have you added the start_debug ?

AndySchroder, 2016-08-08:
One question I have about a test case is would you prefer it to use raspistill or picamera? I was going to try and create a test case with bash and raspistill, but it will take weeks waiting for me to make sure that the test case actually reproduces the problem. I have a limited number of man time, pi's, and cameras to test with and there are different things I am trying.

Ruffio, 2016-08-08:
What suits you best

Ruffio, 2016-08-08:
but it sounds like a memory leak if it take that long time to reproduce. Maybe you could take a look at the memory after some weeks?

JamesH65, 2017-12-21:
@AndySchroder Is there anything to report on this? Can it be closed?

AndySchroder, 2017-12-29:
I never got to developing the extended test case with multiple raspberry pi's. I ended up abandoning the raspberry pi platform for my image capturing needs. In addition to this problem, I started having a number of other intermittent system crashes, and failed hardware of the raspberry pi platform. Standard IP cameras using RTSP seem to be much more robust and have the benefits of PoE, nice enclosures, and low light performance, at the expense of a little bit lower resolution and no ability to receive RAW images.

So, I no longer will be troubleshooting this problem.

closed 2017-12-29 by pelwell