raspberrypi/firmware #611: GPU starts to act up when "allocs" reaches ~32800

loganmc10, 2016-06-06:
I am trying to run an OpenGL ES 2 application (mupen64plus + GlideN64, an N64 emulator).

You can see the bug report I filed with them here:

https://github.com/gonetz/GLideN64/issues/1015

When I run "sudo vcdbg reloc stats" I can see that every single time the "allocs" get to around 32800, the textures start to get messed up, and "alloc_fails" starts to climb, with no new "allocs" happening.

Also, I use raspi2png to grab screenshots and when this happens it stops working with this message: 
"vc_dispmanx_snapshot() failed".

If I close mupen64plus and re-open it, everything starts to work normally again.

loganmc10, 2016-06-06:
If I add "start_debug=1" and do "sudo vcdbg log assert" after this starts happening, I see no error.

popcornmix, 2016-06-06:
Sounds like a memory leak in the arm side code. E.g. textures not being deleted.

Note: All gl calls must come from the same thread. I have seen a bug where texture deletes were being done from a different thread. That violates the GL API, and on the Pi the texture deletes will fail. On some platforms it may work which is why a bug like this may not have been spotted.

pelwell, 2016-06-06:
Here's a tiny snippet of the firmware source:
```
KHRONOS_SERVER_MEMPOOL_HANDLES=32768
```
That is the maximum number of allocations. Raising that limit is possible but it is a build-time change and will be lost memory for all users.

loganmc10, 2016-06-06:
@popcornmix In that case would a GL Error of some kind be raised? I don't get any errors when I run the application and have it check glGetError()

popcornmix, 2016-06-06:
The context is attached the the thread, and the context holds the last glGetError, so there is nowhere for the error to go when you call gl from a thread without a valid context. So I think the answer is no.

loganmc10, 2016-06-06:
Ok I'll check with the developer, I assume it's not a threading issue, I think the application is just allocating a lot of textures...

One last question, is there a way to check the current number of allocations? the "allocs" field in "vcdbg reloc stats" seems to be a running total

loganmc10, 2016-06-06:
Actually a better question. Is there a way to check the current number of allocations via C++? Perhaps via some "vc_dispmanx_" call or something?

popcornmix, 2016-06-06:
I'd suggest:
``vcgencmd cache_flush && sudo vcdbg reloc 2>&1 | grep used | wc -l``
to get the number of allocations. There is no C++ api for this - it is purely a debug feature and not something the arm should have access to.
32K does sound like an awful lot of textures, which I'd suspect was a leak.
Perhaps the developer could identify what a sensible number of textures should be?
Posting the ``vcdbg reloc`` output may confirm if they are definitely textures. The allocated size may give a clue as to what they may contain.
We could make a test build with, say a 128K limit, but I suspect that will just runs for a bit longer before hitting the limit.

loganmc10, 2016-06-06:
Yes they are definitely textures the ```vcdbg reloc``` output is a never ending list of
```
[29874] 0x2c551520: used  16K (refcount 1 lock count 0, size    12288, align 4096, data 0x2c552000, d1rual) 'Texture blob'
[29873] 0x2c555540: used  608 (refcount 1 lock count 0, size      540, align    4, data 0x2c555560, d0rual) 'GLXX_TEXTURE_T'
```

I don't know a lot about N64 emulation, but I believe what is happening is that it is generating some duplicate textures, the particular game I am having issues with is Worms Armageddon, which has a dynamic background (leaves blowing around and such). I think it is generating a new texture for every "leaf" even though they are all the same. That is just my guess though.

The emulator implements a texture cache, so these textures are kept in memory in case they are ever needed again, and it has a function to flush the cache when it hits a size limit (500MB), but because these are all tiny textures, I think it's hitting the 32768 limit before the 500MB. I'll probably just modify the code so that it makes sure the texture cache doesn't have too many (over ~32000) items in it.

loganmc10, 2016-06-07:
So it looks like it takes 2 allocs per textures, so its crapping out at around 15000 textures. I'll close this issue, but it would be nice is this sort of limitation was documented somewhere. I tested this same application with anholts new VC4 driver, and it doesn't have this issue (although it runs much slower).

loganmc10, 2016-07-09:
So the application has been improved and doesn't use as many textures now, but it still uses a lot.  @pelwell 32768 seems quite low, other GPU's don't have this limitation, I know many users use this same application and allocations can go up to 128000 without issue. The fact that each texture takes 2 allocations means that the real maximum is around 16000.

What is the disadvantage of raising that number? It would provide a real performance boost for this application, since it maintains a texture cache. Right now I had to modify the code so that once it reaches 15000 textures it wipes the texture cache and starts over, which obviously leads to a performance drop as it has to do a bunch of glTex calls to reload all those textures.

pelwell, 2016-07-09:
The issue is one of memory usage. Each handle occupies 8 bytes, so 32K handles is 256KB RAM, which is quite a bit to statically allocate, regardless of whether it is used, if you only have 16MB assigned to the VPU.

loganmc10, 2016-07-09:
How do other drivers do this? This issue isn't present in the open source driver, and I assume it isn't present in many other mobile GPU's (like Android devices) since this application is used quite a bit on those platforms and they have no issues.

pelwell, 2016-07-11:
Other drivers run on the ARM, which has an MMU; the VPU, which runs the standard OpenGL driver,  doesn't have an MMU and therefore needs to use a lot of tricks to avoid memory fragmentation.

loganmc10, 2016-07-11:
Ok makes sense. It would be nice to see that number increased. I assume if someone has gpu_mem set to 16 they aren't really using it for anything anyway so using an extra 512KB (or whatever it might be) wouldn't make a difference, unless it prevented it from booting or something.

popcornmix, 2016-07-12:
> Right now I had to modify the code so that once it reaches 15000 textures it wipes the texture cache and starts over

This seems inefficient. Can't the loaded textures be stored in a queue, and just drop the oldest texture when the length reaches 15000? Seems unnecessary to drop all textures at this point.

loganmc10, 2016-07-12:
That's what i did at first:

https://github.com/gonetz/GLideN64/commit/8c632ca0adcd085c2888bc77479ae0ece7316441

But the developer of the application rewrote it to just wipe the whole cache:

https://github.com/gonetz/GLideN64/commit/803eb1d3814e469b811f3b60b5df35d8197b71f6

So yes I would agree, I was just skipping some of the back story for you guys

Ruffio, 2016-07-23:
@loganmc10 so this issue can be closed now?

loganmc10, 2016-07-23:
Well I would like to see the maximum number of allocations raised, but if that is not going to happen then yes, it can be closed.

Ruffio, 2016-07-23:
@popcornmix what are your saying about this?

popcornmix, 2016-07-23:
mupen64plus is the only app I've heard of that hits this limit and doubling the number of allocations will lose 256K for every Pi user. And I suspect that mupen64plus will still hit the issue just after a longer period. So no plans to change this.

Note: we are switching to an [arm side GL driver](https://www.raspberrypi.org/blog/another-new-raspbian-release/) which won't have this limit and is an alternative option for mupen64plus.

closed 2016-07-23 by popcornmix