raspberrypi/firmware #612: hello-teapot drawing with no textures

atic-atac, 2016-06-10:
Running the latest firmware as of today, and trying the hello-teapot demo.

It seems on some occasions when trying to execute this demo, the textures fail to operate, but no errors are returned in the program.

Any clues ?

popcornmix, 2016-06-10:
What do you have gpu_mem set to? Does it help to increase it (e.g. to 128M).

atic-atac, 2016-06-10:
already set gpu_mem=512

atic-atac, 2016-06-10:
it looks like port settings changed for port 221 (egl_render output) never gets triggered

popcornmix, 2016-06-10:
So from boot, does it always work the first time?
If you keep running/stopping it how often does it fail?

atic-atac, 2016-06-10:
It's pretty sporadic, but it looks like it does work first time.

atic-atac, 2016-06-10:
second time usually fails

atic-atac, 2016-06-10:
in fact, after the first time, you are pretty lucky for it to work again.

popcornmix, 2016-06-10:
It's worked ten times in a row for me.

Did it work reliably with an older firmware?

What does:
```
vcgencmd version
vcgencmd get_config int
uname -a
```

atic-atac, 2016-06-10:
# vcgencmd version
Jun  8 2016 16:08:27 
Copyright (c) 2012 Broadcom
version 1c8638223821fe0ed16e20bd8e948b6cc037fa4a (clean) (release)

atic-atac, 2016-06-10:
# vcgencmd get_config int
arm_freq=900
audio_pwm_mode=1
config_hdmi_boost=5
disable_commandline_tags=2
disable_l2cache=1
disable_splash=1
enable_uart=1
force_eeprom_read=1
force_pwm_open=1
framebuffer_depth=32
framebuffer_ignore_alpha=1
framebuffer_swap=1
hdmi_force_cec_address=65535
hdmi_group=2
hdmi_mode=82
init_uart_clock=0x2dc6c0
lcd_framerate=60
over_voltage_avs=0x1b774
pause_burst_frames=1
program_serial_random=1
sdram_freq=450
second_boot=1
temp_limit=85

atic-atac, 2016-06-10:
# uname -a
Linux raspberrypi 4.4.13-v7+ #893 SMP Wed Jun 8 14:39:18 BST 2016 armv7l GNU/Linux

popcornmix, 2016-06-10:
It's working for me (with same kernel/firmware).
Has it worked for you reliably with a older firmware version?
Does it fail with a clean raspbian install?

atic-atac, 2016-06-10:
I can't say I tried with earlier firmware.

I'd have to take some time to install a clean install though.

popcornmix, 2016-06-10:
You can try:
```
vcgencmd set_logging level=0xc0
/opt/vc/src/hello_pi/hello_teapot/hello_teapot.bin
sudo vcdbg log msg
```
which gets some gpu side logging out. Any obvious difference in log between good and bad run?

atic-atac, 2016-06-10:
there's a lot of stuff in there, not sure exactly what I'm looking for.

popcornmix, 2016-06-10:
Run a diff would be my suggestion. Or use a paste site and post links to them.

atic-atac, 2016-06-10:
yeah, but it seems as though the threading shows some different behaviour when fill/empty happens so there's some bits in the diff that would need skipping over.

Ruffio, 2016-06-29:
@atic-atac has this been resolved or would you link to at paste site?

atic-atac, 2016-06-29:
Not sure why it doesn't work with teapot, but my application works fine. So close it.

closed 2016-06-29 by atic-atac