raspberrypi/firmware #613: DTB: set the ethernet local-mac-address alias in the firmware

capnm, 2016-06-10:
See:
https://github.com/anholt/linux/commit/6a93792774fc72861b7e8efaa3545a88272b4413

 ARM: bcm2835: dt: Add the ethernet to the device trees

The hub and the ethernet in its port 1 are hardwired on the board.

Compared to the adapters that can be plugged into the USB ports, this
one has no serial EEPROM to store its MAC. Nevertheless, the Raspberry Pi
has the MAC address for this adapter in its ROM, accessible from its
firmware.

U-Boot can read out the address and set the local-mac-address property of the
node with "ethernet" alias. Let's add the node so that U-Boot can do its
business.

Model B rev2 and Model B+ entries were verified by me, the hierarchy and
pid/vid pair for the Version 2 was provided by Peter Chen. Original
Model B is a blind shot, though very likely correct.

Acked-by: Stephen Warren <swarren@wwwdotorg.org>

Ruffio, 2016-06-29:
Is this a request or a reminder for a coming PR?

Ruffio, 2016-08-10:
@capnm Is this a request or a reminder for a coming PR?

popcornmix, 2016-08-10:
The linked commit is in the upstream 4.8 source and in the rpi-4.8.y tree:
https://github.com/raspberrypi/linux/tree/rpi-4.8.y
https://github.com/raspberrypi/linux/commit/6a93792774fc72861b7e8efaa3545a88272b4413

stapelberg, 2017-02-04:
I find this issue a bit confusingly worded, so hopefully the following helps to clarify:

* Currently, the raspberry pi firmware bootloader (this repository) sets the `smsc95xx.macaddr` kernel module parameter on the kernel command line.
* The upstream `smsc95xx` driver does not support this functionality, hence a custom patch is necessary.
* Said patch has been rejected by upstream 7 years ago: http://netdev.vger.kernel.narkive.com/Fi2roLel/patch-smsc95xx-add-module-parameter-to-override-mac-address
* In the meantime, the upstream `smsc95xx` driver learnt to respect the MAC address from the device tree, if present.
* The u-boot bootloader correctly sets the MAC address in the device tree.

It would be great if the raspberry pi firmware bootloader could be changed to also correctly set the MAC address in the device tree, so that when using an upstream kernel, users don’t need to apply patches to avoid having a new random MAC address on every boot.

Could this functionality be added please?

Thanks!

stapelberg, 2017-02-16:
@popcornmix Any news on this? Is there any way I can help?

pelwell, 2017-02-16:
@popcornmix is probably waiting for me to deal with this one since it falls more into my area. 

Although those commits are present in the tree, they currently won't affect the DTBs used by the standard Pi configurations since the firmware loads bcm27*.dtb rather than bcm283*.dtb.

I'm happy to modify the firmware to look for the node with the `ethernet` label and, if found, set the `local-mac-address` property and suppress the command line parameter; this will allow testing with suitably hacked DTBs. We can then consider changing the standard DTBs in order to drop the downstream smsc95xx patch.

stapelberg, 2017-02-16:
@pelwell Thanks for your reply.

On my various Raspberry Pi 3s, I use an upstream kernel and set `device_tree=bcm2837-rpi-3-b.dtb`. I’m assuming your planned modification would cover my use-case as well as allow you guys to eventually drop the smsc95xx patch.

I’m happy to help test a modified firmware. Thanks in advance for your time!

pelwell, 2017-02-17:
[Here](https://drive.google.com/file/d/0B_P-i4u-SLBXUlpFM0lmN3JWLTA/view?usp=sharing)'s a test firmware (elfs and dats). I've pushed updates to the RPi DTBs on 4.9 and 4.10 that includes a node for the SMSC. As outlined above, the new firmware will detect the `ethernet` label, populate `local-mac-address` and suppress `smsc.95xx.macaddr=...`.

Assuming testing goes well we'll drop the downstream driver patch. The new firmware also strips out some legacy parameters from the command line.

Thanks for the prod - this has been a useful exercise.

stapelberg, 2017-02-20:
Thanks for the test firmware!

I copied https://people.debian.org/~stapelberg/raspberrypi3/2016-11-24-raspberry-pi-3-stretch-PREVIEW.img onto an SD card and dropped your elfs/dats onto the boot partition. The Linux 4.8.7 upstream DTB file already included the ethernet label:
```
$ dtc -I dtb -O dts /media/sdb1/bcm2837-rpi-3-b.dtb 2>&- | tail -4
	aliases {
		ethernet = "/soc/usb@7e980000/usb1@1/usbether@1";
	};
};
```

For your convenience, here’s the full DTB I’m using: [bcm2837-rpi-3-b.dtb.gz](https://github.com/raspberrypi/firmware/files/788194/bcm2837-rpi-3-b.dtb.gz) (text form: 
[bcm2837-rpi-3-b.dtb.txt.gz](https://github.com/raspberrypi/firmware/files/788196/bcm2837-rpi-3-b.dtb.txt.gz))

Unfortunately, the smsc95xx driver does not pick up the the MAC address, and the kernel cmdline parameter is still present:
```
root@rpi3:~# cat /proc/cmdline
earlyprintk console=ttyAMA0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 dma.dmachans=0x7f35 bcm2709.boardrev=0xa02082 bcm2709.serial=0x95d0f70b bcm2709.uart_clock=48000000 smsc95xx.macaddr=B8:27:EB:D0:F7:0B vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  console=ttyS0,115200 root=/dev/mmcblk0p2 elevator=deadline fsck.repair=yes rootwait
root@rpi3:~# ifconfig eth0
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether ea:9f:52:39:38:d6  txqueuelen 1000  (Ethernet)
[…]
```

I tried adding `loglevel=5` to get some more debug output, but I see no additional messages on the serial port. I tried compiling `vcgencmd` in the hope that it might avail some debug logs, but it doesn’t seem to work on arm64 (or I’m doing something wrong):
```
# ../../../../../../bin/vcgencmd --helplease/host_applications/linux/apps/gencmd#
VCHI initialization failed
```

Am I doing something wrong? Is there anything else you need from me to take a look at this issue?

pelwell, 2017-02-20:
I think we're seeing the difference between a label (which becomes a symbol in __symbols__) and an alias. Can you confirm that there is no ethernet property in the __symbols__ node (you can look in /proc/device-tree/__symbols__)?

If I'm right (actually, even if I'm wrong) the correct thing to do is to look for the alias instead - it is more portable.

stapelberg, 2017-02-20:
Oddly enough, I don’t have a symbols file at all. I can display the alias, though:
```
# ls -l /proc/device-tree/symbols
ls: cannot access '/proc/device-tree/symbols': No such file or directory
# cat /proc/device-tree/aliases/ethernet 
/soc/usb@7e980000/usb1@1/usbether@1
```

Googling for `"/proc/device-tree/symbols"` yields merely one result. Are device tree symbols maybe a Raspberry Pi-specific kernel concept?

pelwell, 2017-02-20:
They aren't exactly Pi-specific, but they do depend on a patched dtc which isn't (yet) part of a standard kernel tree.

I'll get you an updated firmware ASAP.

pelwell, 2017-02-21:
[This firmware](https://drive.google.com/open?id=0B_P-i4u-SLBXMk9mcnRYUlk4RlU) (just elfs and dats again) looks for an alias, falling back to a label if it doesn't find the alias.

stapelberg, 2017-02-21:
The 170221 firmware files you provided work for me: the ethernet interface comes up with the correct MAC address and the smsc95xx.macaddr kernel parameter is no longer present. I also verified the MAC address stays the same after rebooting.

Please let me know once this change made it into the repository and I’ll package a new firmware snapshot for Debian.

Thank you very much!

pelwell, 2017-02-27:
This change has now made it to the firmware repo used by rpi-update (https://github.com/Hexxeh/rpi-firmware/commit/411288a7e31067fffa1b955a30b133bf2902bb54). It is only a matter of time before it hits the Raspbian raspberrypi-bootloader package.

stapelberg, 2017-03-17:
Thank you very much. I can confirm that recent firmware is indeed setting the MAC address properly with the upstream kernel. This issue can be closed now.

closed 2017-03-17 by pelwell