raspberrypi/firmware #615: vcgencmd get_throttled on RPi3 returns unexpected value

Terminus-IMRC, 2016-06-11:
When I do ```vcgencmd measure_temp; vcgencmd get_throttled; vcgencmd measure_temp``` on RPi3, it sometimes output

```
temp=39.2'C
throttled=0x50000
temp=38.6'C
```

and sometimes
```
temp=38.6'C
throttled=0x50005
temp=38.6'C
```
(The over-temperature bit is on though the actual temperature is much lower than 85'C).

I'm using the latest firmware.

Is this reproducible on other RPi3? (Sorry but I have only one RPi3 now.)
I confirmed that this is NOT reproducible on RPi2B.

popcornmix, 2016-06-11:
Actually bit 2 means you are currently throttled (turbo disabled). The forum post was incorrect there.
That may be either due to low voltage or over temperature.
If the low voltage bit is set and you are throttled then that implies over voltage.

So, for your results you are currently throttled due to low voltage. I suspect you also have a red/orange warning square on screen. You need a better power supply (temperature is not a problem).

popcornmix, 2016-06-11:
I've edited the forum post to match the behaviour more accurately.

closed 2016-06-11 by Terminus-IMRC

Terminus-IMRC, 2016-06-11:
Note: The forum post is this: https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=147781&start=50#p972790