raspberrypi/firmware #616: Visibility in the source version used for building the binaries

petrosagg, 2016-06-16:
The source package from raspbian [1] points to this repo to build the libraspberrypi0 package etc. I'm trying to create a slightly modified version of the libraspberrypi0 package and to do that I have to build from source, which is in raspberrypi/userland.

The question is how do I know on which commit of the userland I should apply the patch on. Currently the commits to this repo sometimes reference PRs from the related repos, but that's not easy to automate.

Can there be a file that lists the commit hashes of the sources used to build the binaries included in this repo? That would make creating patched packages way easier.

[1] http://archive.raspbian.org/raspbian/pool/firmware/r/raspberrypi-firmware-nokernel/raspberrypi-firmware-nokernel_1.20160523-1~nokernel2.tar.gz

lurch, 2016-08-12:
I'd assume that the binaries in the firmware repo are always built from the latest code in the userland repo, so maybe you could match things up by date?

petrosagg, 2016-08-12:
Yeah, that's my current approach, but it has a few issues. What happens when there are multiple commits in a day?

How about creating a git tag every time a commit is built and published to the repos? Then I could match the using the date plus the autoincrement at the end `20160523-1`

JamesH65, 2017-12-21:
@XECDesign Any thoughts on this, or close as won't fix?

XECDesign, 2017-12-21:
The firmware and kernel repos are currently tagged with the corresponding version number in Raspbian. I don't know of a way to tell which userland version is used though, so I'd have to throw that over the fence.

lurch, 2017-12-21:
I guess this is a question for @popcornmix then? (he seems to be The Builder of the Firmware)

@petrosagg I realise this is an old issue, but bear in mind that there's a couple of linux-side ARM binaries whose source isn't in the [userland](https://github.com/raspberrypi/userland/) repo (see  https://github.com/raspberrypi/userland/issues/56 ), which may or may not be an issue for your use-case.