raspberrypi/firmware #617: config.txt preprocessor

xcvista, 2016-06-18:
Introducing finer grained conditional syntax into `config.txt`. This will allow any settings be made model-specific. Support of defining new variables optional. By the way, compared to the existing syntax, this offers much finer grained control and basic hardware detection, especially useful when compute modules are used with a customization product.

Some variables and functions that came to mind (using C preprocessor syntax):

```C
// CPU: Raspberry Pi Generation
// 2835 = Raspberry Pi 1 (BCM2835)
// 2836 = Raspberry Pi 2 (BCM2836)
// 2837 = Raspberry Pi 3 (BCM2837)
#define CPU 2836

// MODEL: Raspberry Pi Model
// 0 = Model Zero
// 1 = Model A(+)
// 2 = Model B(+)
// 3 = Compute Module
#define MODEL 0x02

// SERIAL: Serial number
#define SERIAL 0x123456

// MAC: MAC address (as string)
// Helpful if cmdline.txt can be absorbed into config.txt
#define MAC "00:14:cf:95:3d:6c"

// GPIO: main GPIO connectors pins count
// 26: Pi 1 Rev 1 (P1)
// 34: Pi 1 Rev 2 (P1+P5)
// 40: Pi 1+/2/3/Zero
// 200: Pi CM
#define GPIO 40

// HAT(): Detect HAT
// -1 = HAT unsupported (Pi 1 Models A & B, CM)
// 0 = HAT not detected
// 1 = HAT detected
#define HAT() ((GPIO == 40) ? I2C_VC_DETECT(0x50) : -1)

// CAMERA(): Detect camera
// 0 = no camera
// 1 = revision 1 camera
// 2 = revision 2 camera
#define CAMERA()

// CAMERA_NOIR(): NoIR camera
// -1 = no camera
// 0 = normal camera
// 1 = NoIR camera
#define CAMERA_NOIR()

// DISPLAY(): available displays (bitmap)
// 0x1: HDMI output
// 0x2: Components
// 0x4: DSI
#define DISPLAY()

// CSI(): CSI link status bitmap
// 0x01: CSI bus 0 link 0
// 0x02: CSI bus 0 link 1
// 0x10: CSI bus 1 link 0
// 0x20: CSI bus 1 link 1
// 0x40: CSI bus 1 link 2
#define CSI()

// DSI(): DSI link status bitmap
// 0x01: DSI bus 0 link 0
// 0x02: DSI bus 0 link 1
// 0x10: DSI bus 1 link 0
// 0x20: DSI bus 1 link 1
// 0x40: DSI bus 1 link 2
// 0x80: DSI bus 1 link 3
#define DSI()

// I2C_ARM_DETECT(addr): Function detecting if something exists on ARM I2C bus at given address
#define I2C_ARM_DETECT(addr)

// I2C_VC_DETECT(addr): Function detecting if something exists on VC I2C bus at given address
#define I2C_VC_DETECT(addr)

// I2C_EEPROM_READ(device, addr): Read a byte from an 16-bit EEPROM attached to ARM I2C bus
#define I2C_EEPROM_READ(device, addr)

// HAT_EEPROM_READ(device, addr): Read a byte from an 16-bit EEPROM attached to VC I2C bus
#define HAT_EEPROM_READ(device, addr)
```

This means users can write device-specific `config.txt`, even perform a little hardware probing so overlays can be inserted conditionally. Although if the C preprocessor syntax is used here like the example below we may need to move comments to C99 comments too (starting with `//` instead of `#`)

```C
#if CPU == 2835 && MODEL != 0 // Pi 1
arm_freq=950 // overclock...
#elif CPU == 2837 // Pi 3
core_freq=250 // /dev/ttyS0 fix
#endif

#if I2C_ARM_DETECT(0x68) // Found an RTC at 0x68, assume DS3231.
dtoverlay=i2c_rtc,ds3231
#endif

#if I2C_ARM_DETECT(0x3b) // Found WM8804, one of the two main components of Cirrus Logic Audio Card, and is visible on I2C bus
dtoverlay=rpi-cirrus-wm5102
#endif
``` 

Also introduce an overlay that can result in its arguments appear somewhere in the filesystem, one line per argument:

```
dtoverlay=sysinfo,gpio=$GPIO,camera=$CSI()
```

results in this being appended to `/proc/cpunfo` (or somewhere in `/sys`) for a Pi 2 with a camera:

```
gpio=40
camera=48
```

grigorig, 2016-06-22:
This sounds over the top. The config.txt syntax is already too complicated.

MilhouseVH, 2016-06-22:
I agree extending the syntax further seems over the top.

Apart from the hardware probing (which might be better left to device tree?) can't most of the model detection be handled already with existing `config.txt` conditional filter syntax ie. `[pi3]` (see "Conditional Filters" section in this [blog post](https://www.raspberrypi.org/documentation/configuration/config-txt.md)).

I also believe overlays can now be loaded form userland, so maybe that's another solution - load overlays once the system has booted, using whatever hardware detection is deemed appropriate?

xcvista, 2016-06-22:
@grigorig @MilhouseVH I think the reason the existing syntax is being complicated is because of its use of `[all]`. This makes the syntax a bit confusing for programmers that is familiar with the concept of opening and closing blocks,with opening and closing of blocks matched. This `config.txt` syntax (indented to make blocks more clear):

```
[rpi2]
    dtparams=i2c_arm=on
    [edid=foobar]
        hdmi-turbo=1
[all]
arm-freq=950
```

maps to this syntax:
```C
#if RPI2()
    dtparams=i2c_arm=on
    #if EDID() == "foobar"
        hdmi-turbo=1
    #endif
#endif
arm-freq=950
```

This sudden termination of blocks makes it feel over the top. Also it makes the "else" clause a bit difficult to implement. This C-like syntax intend to make the block matching more clearer and explicit, and making the "else" clauses much easier to write.

JamesH65, 2017-12-21:
It seems extremely unlikely we will be changing the config parser at this point given its been in place for some years, so closing this as a won't fix.

closed 2017-12-21 by JamesH65