raspberrypi/firmware #619: Wrong pts from deinterlacer image_fx

zillevdr, 2016-06-25:
Hello,

image_fx used as deinterlacer for a DVB stream set wrong pts. Frames out with 50Hz image_fx set the diff between frames to 20000. The diff must be 1800. It`s like the wrong clocksource used. With clock_gettime() the diff is 20000. 
From [https://en.wikipedia.org/wiki/Presentation_timestamp](https://en.wikipedia.org/wiki/Presentation_timestamp):
"Presentation time stamps have a resolution of 90kHz". It should be now_diff * 90 / 1000 = 1800.

Regards zillevdr.

popcornmix, 2016-06-25:
No, the openmax/mmal api specifies time in microseconds.
The correct offset for 50Hz frames is 20000.
This link you have referenced is for "an MPEG transport stream or MPEG program stream". openmax covers many more use cases than that.
If your program wants to work with a 90kHz time base then scale the numbers.

zillevdr, 2016-06-25:
Hello,

in MMAL_BUFFER_HEADER_T *buffer are a pointer to the next buffer. It`s possible addition both pts and division by 2. I think this is for all cases correct.

Regards zillevdr.

popcornmix, 2016-06-25:
> in MMAL_BUFFER_HEADER_T *buffer are a pointer to the next buffer. It`s possible addition both pts and division by 2. I think this is for all cases correct.

I don't know what you are asking here.
But we are not changing the MMAL header structure. The current headers are used by many applications.

zillevdr, 2016-06-25:
There is no need to change the header structure. In an interlaced frame are two progressive frames. The pts for the second out frame is interpolated. Now the pts for the second out frame is an addition from first out frame pts (== in frame pts) and a value of microseconds. The pts must be only in the middle of the pts from the 2 interlaced (in) frames. It's possible to addition the pts from the current interlaced frame to the pts from the next interlaced frame and division by 2 for the pts for the second out frame? 
Sorry for my english.

popcornmix, 2016-06-25:
I don't understand the problem. The API for MMAL says that timestamps are in microseconds.
If you respect the API You will get the timestamp + half the frame interval for the interpolated frame.

That is a better choice than using the timestamp of the next frame to determine the interpolated timestamp where frames can be missing (e.g. live TV) or frames have unknown timestamps.

With bob deinterlace there may not even be a next frame - each input frame produces two output frames. With your scheme we couldn't output the second frame until another input frame arrived, just to get the timestamp.

Ruffio, 2016-08-07:
@zillevdr can this issue be closed?

zillevdr, 2016-08-07:
Yes, closed.

closed 2016-08-07 by zillevdr