raspberrypi/firmware #620: Raspbian Lite 4.4 error on RPi 3: No SDIO Drive strength init done for chip 43430 rev 1 pmurev 24

Ruffio, 2016-06-26:
On a fresh image of Raspbian Lite 4.4, where I only added wlan0 info, I see this error on dmseg:

[    0.000000] Linux version 4.4.11-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #888 SMP Mon May 23 20:10:33 BST 2016
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 8 MiB at 0x3a800000
...
[    3.988455] systemd-udevd[134]: starting version 215
[    4.415906] gpiomem-bcm2835 3f200000.gpiomem: Initialised: Registers at 0x3f200000
[    4.425428] bcm2835-wdt 3f100000.watchdog: Broadcom BCM2835 watchdog timer
[    4.682434] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
**[    4.706857] brcmfmac: brcmf_sdio_drivestrengthinit: No SDIO Drive strength init done for chip 43430 rev 1 pmurev 24**
[    4.707520] usbcore: registered new interface driver brcmfmac
[    4.872449] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Dec 15 2015 18:10:45 version 7.45.41.23 (r606571) FWID 01-cc4eda9c
[    4.886381] cfg80211: World regulatory domain updated:
[    4.886394] cfg80211:  DFS Master region: unset
[    4.886395] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[    4.886407] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[    4.886414] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[    4.886420] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[    4.886430] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[    4.886439] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[    4.886449] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[    4.886455] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[    4.886463] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
**[    4.926629] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code**
[    4.928195] random: nonblocking pool is initialized
[    5.055443] systemd-journald[132]: Received request to flush runtime journal from PID 1
**[    6.015288] brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists
[    6.015307] brcmfmac: brcmf_add_if: ignore IF event**
[    6.020744] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    6.020795] brcmfmac: power management disabled
[    6.370131] uart-pl011 3f201000.uart: no DMA platform data
[    6.643428] Adding 102396k swap on /var/swap.  Priority:-1 extents:3 across:200700k SSFS
[    6.861447] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    6.861669] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready

Is that expected?

Ruffio, 2016-06-28:
I guess '[ 4.926629] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code'  is coming from upstream, as it is also reported here: https://bugzilla.kernel.org/show_bug.cgi?id=103201

pelwell, 2016-07-11:
The SDIO drive strength message is not applicable to BCM43438. A Broadcom-supplied patch that effectively removes that warning went into the rpi-4.4.y tree just after the last Raspbian releases were built.

Ruffio, 2016-07-11:
@pelwell so you are saying that this is fixed next time Raspbian is realeased? So I will be able to do a rpi-update and then it is fixed?

Currently the issue still remains:
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.4.14-v7+ #896 SMP Sat Jul 2 15:09:43 BST 2016 armv7l GNU/Linux

Ruffio, 2016-07-23:
Confirmed.

closed 2016-07-23 by Ruffio