raspberrypi/firmware #621: Raspbian Lite 4.4 error on RPi 3:  Cannot add dependency job for unit display-manager.service, ignoring: Unit display-manager.service failed to load: No such file or directory.

Ruffio, 2016-06-26:
On a fresh image of Raspbian Lite 4.4, where I only added wlan0 info, I see this error on dmseg:

[    0.000000] Linux version 4.4.11-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #888 SMP Mon May 23 20:10:33 BST 2016
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 8 MiB at 0x3a800000
...
[    2.532820] NET: Registered protocol family 10
[    2.538353] systemd[1]: Inserted module 'ipv6'
[    2.545235] systemd[1]: Set hostname to <raspberrypi>.
[    2.609348] usb 1-1.1: new high-speed USB device number 3 using dwc_otg
[    2.729520] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00
[    2.733761] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    2.740831] smsc95xx v1.0.4
[    2.802837] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:d2:14:c2
[    2.899205] usb 1-1.3: new low-speed USB device number 4 using dwc_otg
**[    2.910088] systemd[1]: Cannot add dependency job for unit display-manager.service, ignoring: Unit display-manager.service failed to load: No such file or directory.**
[    2.917728] systemd[1]: Starting Forward Password Requests to Wall Directory Watch.
[    2.922759] systemd[1]: Started Forward Password Requests to Wall Directory Watch.
[    2.927445] systemd[1]: Starting Remote File Systems (Pre).
[    2.937609] systemd[1]: Reached target Remote File Systems (Pre).
[    2.942312] systemd[1]: Starting Encrypted Volumes.

Is that expected?

Ruffio, 2016-07-24:
Hmm, it looks like systemd is always looking for a Display manager, but because there isn't one on Lite, the above error is thrown.

Here are links to bug filed at Debian regarding this issue: 
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=755359
https://lists.debian.org/debian-qt-kde/2015/08/msg00463.html
https://lists.debian.org/debian-qt-kde/2015/01/msg00057.html

Should the necessary files be created and shipped with Lite, so we avoid this error message or can systemd get configured to disregard the 'missing' file(s)?

ncguk, 2016-07-24:
Off the top of my head, `sudo systemctl mask display-manager.service` should disable it.

Ruffio, 2016-07-24:
Thanks, however:

pi@raspberrypi:~ $ sudo systemctl mask display-manager.service
Failed to execute operation: File exists

IMHO there shouldn't be any errors in dmesg when running a fresh image. If 'they' are fixed (essential errors) or masked (non-essential) that is fine by me, but I don't think there should be any logged.

ncguk, 2016-07-24:
That's odd. I've used the command many times on fresh Raspbian Lite installs and it's always worked for me (I realise this is massively unhelpful). Perhaps `sudo -s` followed by `systemctl mask display-manager.service` will produce different results.

AFAICR, there is no file in the Raspbian Lite distro with the name 'display-manager.service', but it is asked for by another service file. Editing that request out of that service file could produce problems down the road if a user decides to install something that will require that service. As all the 'mask' command does is create a symlink called display-manager.service to /dev/null in /etc/systemd/system, it is much easier to revert.

Whether the file that calls for display-manager.service is superfluous in Raspbian-Lite and would be correctly installed by any software that required it I don't know.

Ruffio, 2016-07-24:
My bad. As an experiment I have installed LXDE on the Lite, so that must be the reason there is a file. 
But shouldn't the error disappear then? I get the error now as before I installed LXDE, so it doesn't change the fact that a fresh image do create the error line in dmesg.

Ruffio, 2016-07-24:
So I guess that we can't get rid of the error as the mask will block users from getting the error. Right?

ncguk, 2016-07-24:
I dislike spurious errors in logs, but I suspect this is completely harmless and in terms of support fixing the problem may make matters worse. That's for the Raspbian maintainers to decide.

closed 2016-07-24 by Ruffio