raspberrypi/firmware #623: pixel_rep have no effect in hdmi_timings (config.txt)

ajefr, 2016-06-29:
Hello,
 I tried differents value for pixel_rep property and it have absolutely no effect on screen.
`
hdmi_timings=<h_active_pixels> <h_sync_polarity <h_front_porch> <h_sync_pulse> <h_back_porch> <v_active_lines> <v_sync_polarity> <v_front_porch> <v_sync_pulse> <v_back_porch> <v_sync_offset_a> <v_sync_offset_b> <pixel_rep> <frame_rate> <interlaced> <pixel_freq> <aspect_ratio>
`
 Pixel is still one pixel wide.
 Any workaround to 2X or 4X pixels like in CEA mode 10 -> 15

 Another issue is that the pi is not booting in CEA mode 65 only in DMT 87

Thanks

popcornmix, 2016-06-29:
Can you explain why you are using pixel_rep rather than just halving/quartering the width?

Can you show your ``config.txt`` settings that fail to boot in CEA mode 65?

ajefr, 2016-06-29:
Hello.
Thanks for your message.
I'm using pixel_rep because I need to use a high pixel clock for a low resolution output.
As low pixel clock is fixed to three or four values and high pixel clock is adjustable to what ever you want, it's possible to get exact timing for screen output.
I will post my config.txt once back home.
Bye

Ruffio, 2016-07-05:
@ajefr can you provide your config.txt file?

ajefr, 2016-07-06:
Hello,
 Sorry for delay,
 this config.txt is booting :
`
dtparam=audio=on

enable_dpi_lcd=1
display_default_lcd=1
framebuffer_width=640
framebuffer_height=480
disable_overscan=1

#dpi_output_format=24853
hdmi_timings=640 1 8 2 8 480 1 15 10 15 0 0 0 60 0 19200000 6

dpi_group=2
dpi_mode=87
hdmi_force_hotplug=1
force_hdmi_hotplug=1
dtparam=spi=off
dtparam=i2c_arm=on
dtoverlay=pi3-disable-bt-overlay

dtparam=i2c1=on
dtparam=i2c_arm=on

gpu_mem_256=128
gpu_mem_512=256
gpu_mem_1024=256
overscan_scale=1

`

and if you replace mode and group by :
`
dpi_group=1
dpi_mode=65
`
Pi is not booting anymore.

I will provide another example for my pixel_rep problem.
Bye

JamesH65, 2017-12-21:
@ajefr Is this still an issue or can the topic be closed?

closed 2018-06-29 by JamesH65