raspberrypi/firmware #624: pi 2 crash 'Bad mode in undefined instruction handler detected'

xlazom00, 2016-06-30:
Jun 29 19:00:17 pi4 kernel: [ 2467.716481] Bad mode in undefined instruction handler detected
Jun 29 19:00:17 pi4 kernel: [ 2467.716718] Internal error: Oops - bad mode: 0 [#1] SMP ARM
Jun 29 19:00:17 pi4 kernel: [ 2467.716868] Modules linked in: rpcsec_gss_krb5 bcm2835_wdt bcm2835_gpiomem uio_pdrv_genirq uio snd_bcm2835 snd_pcm_oss snd_mixer_oss snd_pcm snd_seq_dummy snd_seq_oss snd_seq_midi snd_rawmidi snd_seq_midi_event snd_seq snd_seq_device snd_timer snd ipv6
Jun 29 19:00:17 pi4 kernel: [ 2467.717528] CPU: 3 PID: 1194 Comm: VCHIQ completio Not tainted 4.4.14-v7+ #895
Jun 29 19:00:17 pi4 kernel: [ 2467.717706] Hardware name: BCM2709
Jun 29 19:00:17 pi4 kernel: [ 2467.717797] task: 934a78c0 ti: 9aaca000 task.ti: 9aaca000
Jun 29 19:00:17 pi4 kernel: [ 2467.717918] PC is at 0x75ab1d24
Jun 29 19:00:17 pi4 kernel: [ 2467.718014] LR is at _raw_spin_unlock_irqrestore+0x3c/0x70
Jun 29 19:00:17 pi4 kernel: [ 2467.718130] pc : [<75ab1d24>]    lr : [<805b9014>]    psr: 600a0017
Jun 29 19:00:17 pi4 kernel: [ 2467.718130] sp : 9aacbdb0  ip : 808614f8  fp : 200a0010
Jun 29 19:00:17 pi4 kernel: [ 2467.718347] r10: 808618c4  r9 : 8085cdc0  r8 : 8085cdc0
Jun 29 19:00:17 pi4 kernel: [ 2467.718449] r7 : a00a0017  r6 : 934a70fc  r5 : 00000001  r4 : 76e07c78
Jun 29 19:00:17 pi4 kernel: [ 2467.718573] r3 : 63d8ff0c  r2 : 00000008  r1 : 00000000  r0 : 934a70fc
Jun 29 19:00:17 pi4 kernel: [ 2467.718698] Flags: nZCv  IRQs on  FIQs on  Mode ABT_32  ISA ARM  Segment user
Jun 29 19:00:17 pi4 kernel: [ 2467.718832] Control: 10c5387d  Table: 1349c06a  DAC: 00000055
Jun 29 19:00:17 pi4 kernel: [ 2467.718945] Process VCHIQ completio (pid: 1194, stack limit = 0x9aaca210)
Jun 29 19:00:17 pi4 kernel: [ 2467.719074] Stack: (0x9aacbdb0 to 0x9aacc000)
Jun 29 19:00:17 pi4 kernel: [ 2467.719163] bda0:                                     934a70fc 00000000 00000008 63d8ff0c
Jun 29 19:00:17 pi4 kernel: [ 2467.719318] bdc0: 76e07c78 00000001 934a70fc a00a0017 8085cdc0 8085cdc0 808618c4 200a0010
Jun 29 19:00:17 pi4 kernel: [ 2467.727706] bde0: 808614f8 9aacbdb0 805b9014 75ab1d24 600a0017 ffffffff 00000004 934a6d40
Jun 29 19:00:17 pi4 kernel: [ 2467.736813] be00: 9aacbe44 9aacbe10 8004d5fc 805b8fe4 9c98f800 808618c4 9aacbe3c 934a7100
Jun 29 19:00:17 pi4 kernel: [ 2467.746668] be20: 934a6d48 00000001 934a6d40 00000000 9dab6608 9dab6600 9aacbe6c 9aacbe48
Jun 29 19:00:17 pi4 kernel: [ 2467.754781] be40: 8004d838 8004d420 0000a65d 9dab65fc 00000001 9aacbe74 ffffffff 00000001
Jun 29 19:00:17 pi4 kernel: [ 2467.764794] be60: 9aacbeb4 9aacbe70 800966a4 8004d7e4 00000000 934a7100 934a7100 0170a000
Jun 29 19:00:17 pi4 kernel: [ 2467.776351] be80: 9c89fc00 00000f2c 73e3ed8c 00000000 00000001 0170af2c 0170af2c 8000fd08
Jun 29 19:00:17 pi4 kernel: [ 2467.786091] bea0: 00000001 00000000 9aacbf4c 9aacbeb8 80098414 8009655c 00001000 00000008
Jun 29 19:00:17 pi4 kernel: [ 2467.797268] bec0: 73e3eda0 805b4c00 9aacbf24 9aacbed8 80049918 800d95b0 9aacbf0c 9aacbee8
Jun 29 19:00:17 pi4 kernel: [ 2467.810740] bee0: 800436cc 8000a918 9db30018 934a78c0 1e761000 9dbd1720 73e3ed8c 9d1c5a80
Jun 29 19:00:17 pi4 kernel: [ 2467.821923] bf00: c014c407 00000008 9aaca000 00000000 9aacbf7c 9aacbf20 801693b8 803cf92c
Jun 29 19:00:17 pi4 kernel: [ 2467.835368] bf20: 805b4c00 00000000 00000001 00000081 0170af2c 8000fd08 9aaca000 00000001
Jun 29 19:00:17 pi4 kernel: [ 2467.848940] bf40: 9aacbfa4 9aacbf50 80098f54 800982e4 00000001 00000000 ffffffff 73e3ed8c
Jun 29 19:00:17 pi4 kernel: [ 2467.862079] bf60: 9d1c5a80 80169604 fffffdfd 9d1c5a81 9aacbfa4 00000001 7fffffff 00000001
Jun 29 19:00:17 pi4 kernel: [ 2467.876428] bf80: 7fffffff 0170af14 000000f0 8000fd08 9aaca000 00000000 00000000 9aacbfa8
Jun 29 19:00:17 pi4 kernel: [ 2467.890287] bfa0: 8000fb40 80098ed4 00000001 7fffffff 0170af2c 00000081 00000001 00000000
Jun 29 19:00:17 pi4 kernel: [ 2467.904291] bfc0: 00000001 7fffffff 0170af14 000000f0 73e6f260 73e3ed8c 73e6f20c 73e5ee9c
Jun 29 19:00:17 pi4 kernel: [ 2467.917287] bfe0: 000000f0 73e3ed08 76ea3c45 76ea6456 400a0030 0170af2c 00000000 00000000
Jun 29 19:00:17 pi4 kernel: [ 2467.931406] Code: ed2d1c5c bf088b02 3ffff1b2 e1f0f8df (b08544fe)

Ruffio, 2016-06-30:
Could you provide more details? How to reproduce?

xlazom00, 2016-06-30:
I can't reproduce it sorry
I just found this in kern.log

Ruffio, 2016-06-30:
How did it happen? How is your boot configuration? Was it a normal boot or what happened? More details are needed else it can be difficult/imposible to recreate

pelwell, 2016-06-30:
When I see "undefined instruction handler" I think of inadequate power supplies. How are you powering your Pi2, and what is connected?

xlazom00, 2016-07-01:
Desktop power supply :)

Ruffio, 2016-07-01:
@xlazom00 I'm not a developer, I just help gathering information.
Are you able to reproduce it? How?
Do you have any usb devices attached to your Pi? If you have too many devices connected/devices that draw too much power from the Pi (usb), this can happen. If you experience it again then try with a good external power supply, and report back.

I'm just a mortal, so I can't close issue, but you can do it yourself if you are satisfied...

xlazom00, 2016-07-01:
As I can't reproduce it 
I am closing this bug

closed 2016-07-01 by xlazom00