raspberrypi/firmware #627: Re-allow negative overscan

zeekoe, 2016-07-04:
Split from https://github.com/raspberrypi/firmware/issues/591#issuecomment-230018884
My display has a physical size of 480*236 pixels. Therefore my config starts with:
```
framebuffer_width=480
framebuffer_height=236
```
to have my framebuffer as closely related to the physical size as possible.

To compensate for black borders, I used this:
```
disable_overscan=1
overscan_top=-63
overscan_bottom=-50
overscan_left=15
overscan_right=17
```
In Debian Wheezy, this worked like a charm.
After updating to Jessie, I have black borders that I cannot get back.
I counted my vertical amount of pixels today from a digital photograph and can confirm it's 232+/-10 pixels.

Quoting @popcornmix:
> What do you expect:
> 
> overscan_top=-63
> overscan_bottom=-50
> 
> to do? overscan settings describe how many pixels to not use on each side of display.
> They should never be negative - that doesn't mean anything.
(...)
> There was a bug long ago where negative overscan settings did something incorrect, and it looks > like your settings were relying on that. But they make no sense without the bug in place.

For my display and also for my Raspberry pi until one month ago, it seemed a perfectly meaningful setup (as can be seen in https://github.com/raspberrypi/firmware/issues/591#issuecomment-229771024)

> Start with:
> 
> disable_overscan=1
> overscan_top=0
> overscan_bottom=0
> overscan_left=0
> overscan_right=0
> 
> and if there are pixels that should be present then gradually increase the numbers (positive only).

With this setup, I have the same output as with the left + right set to zero: so indeed they made no sense. However, I also have the same black borders, and since I cannot go negative, there's no way to get them back. It looks the same as with the negative values:
[image]

This made me start to experimenting. Setting my height to 360 almost fills the entire height - leaving small black borders at left and right. So if all else fails, I can set it to 360ish, adjust overscan until it fits, and then rescale my UI to stretch, then let the display stretch it down again.

However, I'd really like to have a physical pixel be a physical pixel. Any chances that negative overscan could be reimplemented - perhaps not meaningful on most devices, but at least on some? 
Can I use an old firmware, and are there risks in that? Can I compile the bug back in, or is it all closed source? (it looks like the latter)

popcornmix, 2016-07-04:
When running what do ``tvservice -s`` and ``fbset`` report?

zeekoe, 2016-07-04:
for the big screen:
```
pi@koffie /boot $ fbset
mode "480x360"
    geometry 480 360 480 360 16
    timings 0 0 0 0 0 0 0
    rgba 5/11,6/5,5/0,0/16
endmode
pi@koffie ~ $ tvservice -s
state 0x40001 [NTSC 4:3], 720x480 @ 60.00Hz, interlaced
```
and for the original, small one:
```
pi@koffie ~ $ fbset
mode "480x236"
    geometry 480 236 480 236 16
    timings 0 0 0 0 0 0 0
    rgba 5/11,6/5,5/0,0/16
endmode
pi@koffie ~ $ tvservice -s
state 0x40001 [NTSC 4:3], 720x480 @ 60.00Hz, interlaced
```

Ruffio, 2016-07-19:
@popcornmix any chances that negative overscan could be reimplemented?

zeekoe, 2016-11-20:
Or perhaps someone can tell us what version still has negative overscan, and if we can use it without problems?

JamesH65, 2017-12-21:
@zeekoe Its been a while since this was updated, and I am with @popcornmix that negative offsets make no sense, so I doubt we will be reintroducing the bug that allowed them in the first place. It ought to be possible to get round the issue (which I am still not sure about) simply by ensuring the FB matches the display, so I am bit confused.

zeekoe, 2017-12-22:
Hi @JamesH65,
Thanks for coming back on this.
I think the problem is the odd resolution of this display: 480x236 (or 234, can't remember). I can get the screen filled with pixels if I use a more regular resolution, but in no way I can then get a 1-to-1 pixel ratio. I tried 480x472 (twice the height), but then I have similar issues. It just seems the screen expects a certain width/height input ratio. 
The 480x236 is a common resolution on thrift store portable DVD players (I even saw a DVB-T player with a screen of this res!), probably all having this behaviour.
I did match the FB to the display, right? Or is there something else I can try?
Does introducing negative overscan introduce problems for you? Because in this use case (of me and a few other persons) it has proven to be a solution to our problem. If it breaks nothing, can't it just be reintroduced? Or is that too pragmatic? :)

Ronald

JamesH65, 2018-06-29:
@zeekoe Are you still interested/seeing in this issue?

AFACT, -ve overscan  values are simply cover up an issue elsewhere, i.e. the inability to actually correctly work with 480x236 displays. That's what needs to be fixed.

zeekoe, 2018-06-29:
Yep, still using my machine on a near-daily basis. So the negative overscan has not been re-enabled, and there is a chance that 480x236 will be supported? Here in NL, all cheap 7 inch dvd/dvb/etc screens at thrift stores seem to be that resolution.