raspberrypi/firmware #629: Unresolved imports "libbrcmGLESv2.so" and "libbrcmEGL.so" in libopenmaxil.so

umgfoin, 2016-07-06:
One of the recent commits broke several applications trying to load libopenmaxil.so:

readelf libopenmaxil.so -d
Dynamic section at offset 0x581c contains 34 entries:
  Tag        Type                         Name/Value
 0x00000001 (NEEDED)                     Shared library: [libdl.so.2]
 0x00000001 (NEEDED)                     Shared library: [libbrcmGLESv2.so]
 0x00000001 (NEEDED)                     Shared library: [libbrcmEGL.so]
 0x00000001 (NEEDED)                     Shared library: [libbcm_host.so]
 0x00000001 (NEEDED)                     Shared library: [libm.so.6]
 0x00000001 (NEEDED)                     Shared library: [libvchiq_arm.so]
 0x00000001 (NEEDED)                     Shared library: [libvcos.so]
 0x00000001 (NEEDED)                     Shared library: [libpthread.so.0]
 0x00000001 (NEEDED)                     Shared library: [librt.so.1]
 0x00000001 (NEEDED)                     Shared library: [libc.so.6]
 0x0000000e (SONAME)                     Library soname: [libopenmaxil.so]

best regards,
++umgfoin

6by9, 2016-07-06:
Sounds like #625. Library names have been updated.

popcornmix, 2016-07-06:
How did you upgrade?
Did you rebuilding raspistill yourself?
Have you rebuild all of userland repo?
Have you updated firmware (rpi-update)?

umgfoin, 2016-07-06:
> How did you upgrade?

pacman -Syu
> Did you rebuilding raspistill yourself?

No. For me it's breaking kodi-rpi.
> Have you rebuild all of userland repo?

No.
> Have you updated firmware (rpi-update)?

No. ArchLinux updates firmware via pacman.
The break appeared with package raspberrypi-firmware-20160703-1
Latest reverts in raspberrypi-firmware-20160705-1 didn't fix the problem.

popcornmix, 2016-07-06:
Report output of:
```
uname -a
vcgencmd version
ldd `which raspistill`
ls -l /opt/vc/lib
```

Kiina, 2016-07-06:
I think he is saying it isn't raspistill that broke but kodi-rbp. I have the same problem:

`Unable to load libopenmaxil.so, reason: libbrcmGLESv2.so: cannot open shared object file: No such file or directory`

```
uname -a
Linux raspberrypi 4.4.14-2-ARCH #1 SMP Sat Jul 2 20:41:34 MDT 2016 armv7l GNU/Linux
```

```
ls -l /opt/vc/lib
total 2584
-rw-r--r-- 1 root root 201852 Jul  6 00:53 libEGL.so
-rw-r--r-- 1 root root 243424 Jul  6 00:53 libEGL_static.a
lrwxrwxrwx 1 root root     12 Jul  6 12:36 libGLESv1_CM.so -> libGLESv2.so
-rw-r--r-- 1 root root 105572 Jul  6 12:35 libGLESv2.so
-rw-r--r-- 1 root root 100008 Jul  6 00:53 libGLESv2_static.a
-rw-r--r-- 1 root root  99200 Jul  6 00:53 libOpenVG.so
-rw-r--r-- 1 root root  78532 Jul  6 00:53 libWFC.so
-rw-r--r-- 1 root root  96040 Jul  6 00:53 libbcm_host.so
-rw-r--r-- 1 root root 361236 Jul  6 00:53 libcontainers.so
-rw-r--r-- 1 root root  15392 Jul  6 00:53 libdebug_sym.so
-rw-r--r-- 1 root root  12486 Jul  6 00:53 libdebug_sym_static.a
-rw-r--r-- 1 root root  48984 Jul  6 00:53 libdtovl.so
-rw-r--r-- 1 root root  13548 Jul  6 00:53 libkhrn_client.a
-rw-r--r-- 1 root root 448544 Jul  6 00:53 libkhrn_static.a
-rw-r--r-- 1 root root  14108 Jul  6 00:53 libmmal.so
-rw-r--r-- 1 root root  68856 Jul  6 00:53 libmmal_components.so
-rw-r--r-- 1 root root  81928 Jul  6 00:53 libmmal_core.so
-rw-r--r-- 1 root root  88628 Jul  6 00:53 libmmal_util.so
-rw-r--r-- 1 root root  44428 Jul  6 00:53 libmmal_vc_client.so
-rw-r--r-- 1 root root  32974 Jul  6 12:37 libopenmaxil.so
-rw-r--r-- 1 root root   3264 Jul  6 00:53 libvcfiled_check.a
-rw-r--r-- 1 root root  30036 Jul  6 00:53 libvchiq_arm.so
-rw-r--r-- 1 root root 136502 Jul  6 00:53 libvchostif.a
-rw-r--r-- 1 root root  28462 Jul  6 00:53 libvcilcs.a
-rw-r--r-- 1 root root 173108 Jul  6 00:53 libvcos.so
-rw-r--r-- 1 root root  21164 Jul  6 00:53 libvcsm.so
-rw-r--r-- 1 root root  31078 Jul  6 00:53 libvmcs_rpc_client.a
drwxr-xr-x 2 root root   4096 Jul  6 10:08 pkgconfig/
drwxr-xr-x 2 root root   4096 Jul  6 10:08 plugins/
```

popcornmix, 2016-07-06:
Ah - there were some new ``.so`` [files](https://github.com/Hexxeh/rpi-firmware/commit/6a1575e1d59d3c253aeb5553631610c80a7d595d) that were missed (bah - ``.gitignore``) from last firmware update.

They are pushed now. I'm not sure how they get into next ``pacman -Syu`` - that is a question for Arch maintainers.

umgfoin, 2016-07-06:
>  I'm not sure how they get into next pacman -Syu - that is a question for Arch maintainers.

Great, I'll point them to this thread.
Thanks!

umgfoin, 2016-07-06:
> I'm not sure how they get into next pacman -Syu - that is a question for Arch maintainers.

Arch-alarm package fixed with [commit ebe24a9](https://github.com/archlinuxarm/PKGBUILDs/commit/ebe24a9d134922a71106354f61b34b7b2d560e09).

closed 2016-07-06 by umgfoin