raspberrypi/firmware #630: Raspbian Lite 4.4 on RPi 3: Wifi changes regulatory domain multiple times during boot when configured in wpa_supplicant.conf

Ruffio, 2016-07-11:
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.4.14-v7+ #896 SMP Sat Jul 2 15:09:43 BST 2016 armv7l GNU/Linux

I have changed the wpa to below, where country DK is set as regulatory domain:

pi@raspberrypi:~ $ sudo cat /etc/wpa_supplicant/wpa_supplicant.conf
**country=DK**
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP = netdev
update_config=1
network={
    ssid="XXX"
    psk="XXX"
}

It looks like the wifi is changing regulatory domain several times during boot, sometime to unset, sometime to DE and sometime to the correct DK:

Here is part of my dmesg:
**[    5.387541] cfg80211: World regulatory domain updated:
[    5.387554] cfg80211:  DFS Master region: unset
[    5.387567] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[    5.387585] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[    5.387599] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[    5.387611] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[    5.387632] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[    5.387672] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[    5.387686] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[    5.387698] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[    5.387710] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)**
[    5.481267] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[    5.602819] systemd-journald[139]: Received request to flush runtime journal from PID 1
[    6.424771] brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists
[    6.424798] brcmfmac: brcmf_add_if: ignore IF event
[    6.430232] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    6.430277] brcmfmac: power management disabled
[    6.980087] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    6.980221] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
**[    7.009238] cfg80211: Regulatory domain changed to country: DK
[    7.009253] cfg80211:  DFS Master region: ETSI
[    7.009258] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[    7.009266] cfg80211:   (2402000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[    7.009274] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[    7.009282] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[    7.009288] cfg80211:   (5490000 KHz - 5710000 KHz @ 160000 KHz), (N/A, 2700 mBm), (0 s)
[    7.009301] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)**
[    8.054473] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
[    8.054488] brcmfmac: brcmf_cfg80211_scan: scan error (-11)
[    8.270289] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
**[    8.291739] cfg80211: Regulatory domain changed to country: DE
[    8.291754] cfg80211:  DFS Master region: ETSI
[    8.291759] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[    8.291767] cfg80211:   (2400000 KHz - 2483500 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[    8.291775] cfg80211:   (5150000 KHz - 5250000 KHz @ 80000 KHz, 200000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[    8.291782] cfg80211:   (5250000 KHz - 5350000 KHz @ 80000 KHz, 200000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[    8.291788] cfg80211:   (5470000 KHz - 5725000 KHz @ 160000 KHz), (N/A, 2698 mBm), (0 s)
[    8.291794] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)**
[   12.324228] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
**[   12.324257] cfg80211: World regulatory domain updated:
[   12.324266] cfg80211:  DFS Master region: unset
[   12.324276] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[   12.324290] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   12.324311] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   12.324323] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[   12.324338] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[   12.324358] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[   12.324371] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[   12.324383] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[   12.324396] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[   12.363960] cfg80211: Regulatory domain changed to country: DK
[   12.363984] cfg80211:  DFS Master region: ETSI
[   12.363994] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[   12.364008] cfg80211:   (2402000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   12.364024] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[   12.364057] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[   12.364077] cfg80211:   (5490000 KHz - 5710000 KHz @ 160000 KHz), (N/A, 2700 mBm), (0 s)
[   12.364090] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
[   13.833626] cfg80211: Regulatory domain changed to country: DE
[   13.833639] cfg80211:  DFS Master region: ETSI
[   13.833645] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[   13.833653] cfg80211:   (2400000 KHz - 2483500 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   13.833661] cfg80211:   (5150000 KHz - 5250000 KHz @ 80000 KHz, 200000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[   13.833668] cfg80211:   (5250000 KHz - 5350000 KHz @ 80000 KHz, 200000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[   13.833675] cfg80211:   (5470000 KHz - 5725000 KHz @ 160000 KHz), (N/A, 2698 mBm), (0 s)
[   13.833681] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)**

chrisbgp, 2016-07-11:
Having the same problem since the last upgrade...

Linux pi 4.4.14-2-ARCH #1 SMP Sat Jul 2 20:41:34 MDT 2016 armv7l GNU/Linux

linux-firmware 20160516.80d463b-1
raspberrypi-firmware 20160706-1
raspberrypi-firmware-bootloader 20160706-1
raspberrypi-firmware-bootloader-x 20160706-1
raspberrypi-firmware-examples 20160706-1
raspberrypi-firmware-tools 20160706-1

Bus 001 Device 006: ID 0cf3:9271 Atheros Communications, Inc. AR9271 802.11n

...which is using ath9k

Ruffio, 2016-07-27:
Hmmm, seams to be an upstream issue then? Anybody can conform this?

I set the CRDA at /etc/defaults/crda to DK and also
sudo -i
echo "options cfg80211 ieee80211_regdom=DE"  >  /etc/modprobe.d/cfg80211.conf
exit

but nothing has changed. It is still switching around :-(
(http://askubuntu.com/questions/621452/something-keeps-changing-my-wireless-regulatory-domain)

> pi@raspberrypi:~ $ dmesg |grep "cfg80211"
> [    4.345460] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
> [    4.396900] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
> [    4.396931] cfg80211: World regulatory domain updated:
> [    4.396940] cfg80211:  DFS Master region: unset
> [    4.396949] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
> [    4.396963] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
> [    4.396976] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
> [    4.396988] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
> [    4.397003] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
> [    4.397025] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
> [    4.397041] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
> [    4.397054] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
> [    4.397066] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
> [    4.478371] cfg80211: Regulatory domain changed to country: DK
> [    4.478395] cfg80211:  DFS Master region: ETSI
> [    4.478404] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
> [    4.478419] cfg80211:   (2402000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
> [    4.478446] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
> [    4.478473] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
> [    4.478487] cfg80211:   (5490000 KHz - 5710000 KHz @ 160000 KHz), (N/A, 2700 mBm), (0 s)
> [    4.478499] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
> [    7.501814] cfg80211: Regulatory domain changed to country: DE
> [    7.501830] cfg80211:  DFS Master region: ETSI
> [    7.501836] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
> [    7.501844] cfg80211:   (2400000 KHz - 2483500 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
> [    7.501852] cfg80211:   (5150000 KHz - 5250000 KHz @ 80000 KHz, 200000 KHz AUTO), (N/A, 2000 mBm), (N/A)
> [    7.501859] cfg80211:   (5250000 KHz - 5350000 KHz @ 80000 KHz, 200000 KHz AUTO), (N/A, 2000 mBm), (0 s)
> [    7.501865] cfg80211:   (5470000 KHz - 5725000 KHz @ 160000 KHz), (N/A, 2698 mBm), (0 s)
> [    7.501871] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)

And why does it ends by switching to DE? Where does that come from? Is it hardcoded somewhere?

Ruffio, 2016-07-27:
Also tried:
pi@raspberrypi:/etc/default $ export COUNTRY=DK
pi@raspberrypi:/etc/default $ crda
Failed to set regulatory domain: -28

tuxen, 2016-07-28:
> I set the CRDA at /etc/defaults/crda to DK and also
 sudo -i
 echo "options cfg80211 ieee80211_regdom=**DE**" > /etc/modprobe.d/cfg80211.conf
 exit

shouldn't the DE in bold be DK too, or is it just a type error?

Ruffio, 2016-07-28:
@tuxen it is a typing error here, I did correct on my RPi.

Ruffio, 2016-07-28:
Even without any of my attempts it switches to DE

tuxen, 2016-07-28:
thought so =) 
I have not setup wireless on my RPi. not that I am expecting different results.
odd that it likes DE so much. maybe as you say; its hardcoded somewhere.

clivem, 2016-07-28:
> Even without any of my attempts it switches to DE

I have vague recollections of crda setting regulatory based on system timezone. eg. /etc/localtime.

Ruffio, 2016-07-29:
I tried to copy copy Copenhagen to /etc/locatime and rebooted, but thats doesn't make any difference:
sudo cp /usr/share/zoneinfo/Europe/Copenhagen /etc/localtime

Still changing from DK and back to DE, so to begin with it is correct (DK) but then for some reason it changes to DE. Any other place where DE is configured by default?

> [    5.461802] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: May 27 2016 00:13:38 version 7.45.41.26 (r640327) FWID 01-df77e4a7
> [    5.483335] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
> [    5.572622] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
> [    5.572653] cfg80211: World regulatory domain updated:
> [    5.572662] cfg80211:  DFS Master region: unset
> [    5.572671] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
> [    5.572696] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
> [    5.572721] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
> [    5.572734] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
> [    5.572768] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
> [    5.572792] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
> [    5.572824] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
> [    5.572838] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
> [    5.572850] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
> [    5.674970] cfg80211: Regulatory domain changed to country: DK
> [    5.674992] cfg80211:  DFS Master region: ETSI
> [    5.675002] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
> [    5.675017] cfg80211:   (2402000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
> [    5.675033] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
> [    5.675047] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
> [    5.675060] cfg80211:   (5490000 KHz - 5710000 KHz @ 160000 KHz), (N/A, 2700 mBm), (0 s)
> [    5.675073] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
> [    5.812286] systemd-journald[141]: Received request to flush runtime journal from PID 1
> [    6.483514] brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists
> [    6.483527] brcmfmac: brcmf_add_if: ignore IF event
> [    6.487171] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
> [    6.487199] brcmfmac: power management disabled
> [    7.419978] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
> [    7.420235] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
> [    8.217504] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
> [    8.241718] cfg80211: Regulatory domain changed to country: DE
> [    8.241733] cfg80211:  DFS Master region: ETSI
> [    8.241738] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
> [    8.241746] cfg80211:   (2400000 KHz - 2483500 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
> [    8.241754] cfg80211:   (5150000 KHz - 5250000 KHz @ 80000 KHz, 200000 KHz AUTO), (N/A, 2000 mBm), (N/A)
> [    8.241761] cfg80211:   (5250000 KHz - 5350000 KHz @ 80000 KHz, 200000 KHz AUTO), (N/A, 2000 mBm), (0 s)
> [    8.241768] cfg80211:   (5470000 KHz - 5725000 KHz @ 160000 KHz), (N/A, 2698 mBm), (0 s)
> [    8.241774] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)

clivem, 2016-07-29:
> Any other place where DE is configured by default?

@Ruffio I also have another vague recollection (from trying to help a user with similar problems a year ago), that when WPA supplicant connects to an AP, the reg domain can be set to match that of the AP it is associating with during the association. Never did really understand how/why, but I wonder if your AP is hard-coded to DE. Just throwing the thought out there.....

lfoerster, 2016-07-31:
Same issue here ! Without setting anything wlan stick comes up with reg.domain set to CN (TP-Link Atheros based stick) 
Setting /etc/defaults/crda to DE shows reg.domain set to country to 98. Weired.
`root@lcdpi:/etc# iw reg get
country 98: DFS-UNSET
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 25), (N/A, 20), (N/A), NO-IR
        (2474 - 2482 @ 8), (N/A, 20), (N/A), NO-OFDM, NO-IR
        (5170 - 5250 @ 80), (N/A, 20), (N/A), NO-IR
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, NO-IR
        (5735 - 5835 @ 80), (N/A, 20), (N/A), NO-IR
        (57240 - 59400 @ 2160), (N/A, 0), (N/A)
        (59400 - 63720 @ 2160), (N/A, 0), (N/A)
root@lcdpi:`
Even setting it in wpa_supplicant.conf did NOT work.
I ended up editing /etc/rc.local and set the iw command _sudo iw reg set DE_ there.
That finally did the trick but is of course not a good solution !
There is definitely something wrong in the crda package.

clivem, 2016-07-31:
Guys, I really do suspect that this might be because of what the AP is broadcasting as the country code. I know that Sky broadband customers in the UK, get given an AP/Router that broadcasts DE as the regulatory country....

eg. I'm in the UK, (GB), my neighbours who are using Sky as ISP, all appear to have AP's that are broadcasting DE as the regulatory country.....

```
sudo iw dev wlan0 scan | grep -e SSID -e Country
	SSID: SKY1E929
	Country: DE	Environment: Indoor/Outdoor
	SSID: SKY77B78
	Country: DE	Environment: Indoor/Outdoor
	SSID: SKYEB806
	Country: DE	Environment: Indoor/Outdoor
	SSID: SKY22D06
	Country: DE	Environment: Indoor/Outdoor
	SSID: SKY1D45D
	Country: DE	Environment: Indoor/Outdoor
	SSID: SKYE55A8
	Country: DE	Environment: Indoor/Outdoor
```

Ruffio, 2016-08-01:
Sorry for my late response, but I have been without Internet connection during the weekend and still is :-(

@clivem I don't know where to see if my AP is configured to DE, it is not an open source AP.

'''
pi@raspberrypi:~ $ sudo iw reg set DK
pi@raspberrypi:~ $ dmesg
...
[    6.317397] brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists
[    6.317413] brcmfmac: brcmf_add_if: ignore IF event
[    6.321186] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    6.321214] brcmfmac: power management disabled
[    7.279890] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    7.280029] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    8.136495] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[    8.159917] cfg80211: Regulatory domain changed to country: DE
[    8.159932] cfg80211:  DFS Master region: ETSI
[    8.159937] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[    8.159945] cfg80211:   (2400000 KHz - 2483500 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[    8.159953] cfg80211:   (5150000 KHz - 5250000 KHz @ 80000 KHz, 200000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[    8.159960] cfg80211:   (5250000 KHz - 5350000 KHz @ 80000 KHz, 200000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[    8.159967] cfg80211:   (5470000 KHz - 5725000 KHz @ 160000 KHz), (N/A, 2698 mBm), (0 s)
[    8.159973] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
[   79.941538] cfg80211: Current regulatory domain intersected:
[   79.941564] cfg80211:  DFS Master region: ETSI
[   79.941573] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[   79.941588] cfg80211:   (2402000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   79.941603] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[   79.941617] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[   79.941630] cfg80211:   (5490000 KHz - 5710000 KHz @ 160000 KHz), (N/A, 2698 mBm), (0 s)
[   79.941643] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
'''

That is not what I expected. I expected, to see: cfg80211: Regulatory domain changed to country: DK

Mayby you are right regarding that it is the AP that sets it to use DE, because right after the link becomes ready (look above), the cfg80211 changes to DE.
Eailier I tried to search for files containing '=DE' and I also did the above. That caused the below in dmesg. Is that expected? I only have a USB stick connected (haven't touched it), but is 
dwc_otg 3f980000.usb the onboard wifi?

'''
[ 2783.890791] dwc_otg 3f980000.usb: Invalid offset (0xffffffff)
[ 2783.890828] hsic_connect_show(ad9fd810) -> platform_dev ad9fd800, otg_dev ad031800
[ 2783.890858] guid_show(ad9fd810) -> platform_dev ad9fd800, otg_dev ad031800
[ 2783.890887] mode_show(ad9fd810) -> platform_dev ad9fd800, otg_dev ad031800
[ 2783.897603] WARNING! power/level is deprecated; use power/control instead
[ 2783.913580] srpcapable_show(ad9fd810) -> platform_dev ad9fd800, otg_dev ad031800
[ 2783.913615] Core Global Registers
[ 2783.913621] GOTGCTL   @0xF3980000 : 0x001C0001
[ 2783.913625] GOTGINT   @0xF3980004 : 0x00000000
[ 2783.913630] GAHBCFG   @0xF3980008 : 0x00000031
[ 2783.913634] GUSBCFG   @0xF398000C : 0x20001700
[ 2783.913639] GRSTCTL   @0xF3980010 : 0x80000000
[ 2783.913643] GINTSTS   @0xF3980014 : 0x04000021
[ 2783.913648] GINTMSK   @0xF3980018 : 0xF301080E
[ 2783.913652] GRXSTSR   @0xF398001C : 0x000F0005
[ 2783.913656] GRXFSIZ   @0xF3980024 : 0x00000306
[ 2783.913661] GNPTXFSIZ @0xF3980028 : 0x01000306
[ 2783.913665] GNPTXSTS  @0xF398002C : 0x10080100
[ 2783.913670] GI2CCTL   @0xF3980030 : 0x00000000
[ 2783.913674] GPVNDCTL  @0xF3980034 : 0x00000000
[ 2783.913678] GGPIO     @0xF3980038 : 0x00000000
[ 2783.913683] GUID              @0xF398003C : 0x2708A000
[ 2783.913687] GSNPSID   @0xF3980040 : 0x4F54280A
[ 2783.913692] GHWCFG1   @0xF3980044 : 0x00000000
[ 2783.913696] GHWCFG2   @0xF3980048 : 0x228DDD50
[ 2783.913700] GHWCFG3   @0xF398004C : 0x0FF000E8
[ 2783.913705] GHWCFG4   @0xF3980050 : 0x1FF00020
[ 2783.913709] GLPMCFG   @0xF3980054 : 0x75736230
[ 2783.913714] GPWRDN    @0xF3980058 : 0x00000000
[ 2783.913718] GDFIFOCFG         @0xF398005C : 0x00000000
[... 338 lines cut ...]
[ 3329.432424] hptxfsiz_show(ad9fd810) -> platform_dev ad9fd800, otg_dev ad031800
[ 3329.432451] devspeed_show(ad9fd810) -> platform_dev ad9fd800, otg_dev ad031800
[ 3329.432464] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[ 3329.432497] Core is not in hibernation
[ 3329.432560] hnpcapable_show(ad9fd810) -> platform_dev ad9fd800, otg_dev ad031800
[ 3329.432587] HZ 100, MSEC_PER_JIFFIE 10, loops_per_jiffy 384000
[ 3330.399175] enumspeed_show(ad9fd810) -> platform_dev ad9fd800, otg_dev ad031800
[ 3330.399191] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[ 3330.399228] inv_sel_hsic_show(ad9fd810) -> platform_dev ad9fd800, otg_dev ad031800
[ 5511.467547] cfg80211: Current regulatory domain intersected:
[ 5511.467570] cfg80211:  DFS Master region: ETSI
[ 5511.467579] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[ 5511.467595] cfg80211:   (2402000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[... cut ...]
[ 5511.467625] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[ 5511.467638] cfg80211:   (5490000 KHz - 5710000 KHz @ 160000 KHz), (N/A, 2698 mBm), (0 s)
[ 5511.467651] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)	
'''

I have also found this thread, but havn't got the chance to try it out:
https://community.nxp.com/thread/352772

As soon as I get my internet connection back, I will try out some more testing and report back.

Ruffio, 2016-08-01:
Sorry, this is the correct link: http://fuzon.co.uk/phpbb/viewtopic.php?f=11&t=151&sid=b4aaa56c509e6d83ec743355d502d1ec&start=20

clivem, 2016-08-01:
> I don't know where to see if my AP is configured to DE, it is not an open source AP.

Just scan, `sudo iw dev wlan0 scan`,then look at the details shown for each AP, including yours and see if it has a country set...

I've got a feeling that when you connect to an AP via wpa_supplicant, some kind of callback takes place, probably via udev, (/lib/udev/rules.d/85-regulatory.rules change event calls setregdomain), if a country code is set on the AP. You could probably add some sort of logging to that change event, or a wrapper around setregdomain, so you can see when those events are being generated.

lfoerster, 2016-08-01:
Here its definitely not the case. 
The stick (TP-Link TL-WN722N) comes up with regulatory domain set to China (CN) and, weired enough, DFS set to FCC:
_root@lcdpi:/home/pi# dmesg | grep 802
[    3.937802] systemd[1]: Listening on udev Kernel Socket.
[   10.701188] cfg80211: Calling CRDA to update world regulatory domain
[   13.254861] cfg80211: World regulatory domain updated:
[   13.254904] cfg80211:  DFS Master region: unset
[   13.254956] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[   13.254980] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   13.254996] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   13.255012] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[   13.255063] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[   13.255086] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[   13.255103] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[   13.255118] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[   13.255133] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[   14.808116] ieee80211 phy0: Atheros AR9271 Rev:1
**[   14.808326] cfg80211: Calling CRDA for country: CN
[   15.266293] cfg80211: Regulatory domain changed to country: CN 
[   15.266364] cfg80211:  DFS Master region: FCC **
[   15.266379] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)_

Putting the "iw reg set DE" command into rc.local switches it to DE and ETSI:

_[   22.480363] cfg80211: Calling CRDA for country: DE
[   22.914043] cfg80211: Regulatory domain changed to country: DE
[   22.914081] cfg80211:  DFS Master region: ETSI_
root@lcdpi:/home/pi# iw reg get
country DE: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 20), (N/A), NO-OUTDOOR
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS
        (5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)
So as a conclusion it can be set to different values by the software and its definitely not predefined or somehow "burned in" to a EEPROM or so.
With nothing configured it stays in regulatory domain CN (China) and DFS FCC.

Changing the /etc/defaults/crda domain setting to DE sets the regulatory domain to county code 98 after reboot with no ISO name just the number (98 is reported as Iran...ooops). This is definitely wrong. Why is DE converted to "98" ?
If its preset like this in the /etc/defaults/crda btw. it can NOT be set with _iw reg set DE_ command to a different value. Its stays 98. the iw set command is just ignored.
After erasing DE in /etc/defaults/crda it then can be set with the "iw reg set xy" commad to different values again.
Thats why i ended up setting it in r.local. This finally did set the correct reg.domain as a quick and dirty workaround.

clivem, 2016-08-01:
> So as a conclusion it can be set to different values by the software and its definitely not predefined or somehow "burned in" to a EEPROM or so.

@lfoerster I think we are heading away from what the problem actually is now, but anyway..... I think setting regdomain from software/userspace can only make things more restrictive, with regard to regulatory restictions on channels and power, not less. It will not override what is set in the EEPROM. eg. I bought a wifi dongle from an American eBay vendor. The country was set to 'US' in the EEPROM. `iw reg set GB`, did not restore the availability of channels 12 and 13. ;) ISTR, that from userspace crda you can make it more restrictive than was originally programmed, but not less. 

@Ruffio Anyway, before I butt-out of this conversation, you can always monitor the udev events from userspace. Probably won't tell you what is generating the regdomain change events, but should at least make it easy to see the country code that is being sent to userspace crda (setregdomain) from the kernel. `sudo udevadm monitor --environment kernel`

Ruffio, 2016-08-02:
My whole point of creating this issue is that I can see that RPi3 using onboard wifi does three times of changing the regulatory domain. IMHO it should only do it one time, not three. It should be unnecessary to the two last changes and it takes time for the booting to complete. I worries me that the CRDA has been implemented but there is no guidance in how to set it correctly.

In this thread (http://raspberrypi.stackexchange.com/questions/40738/jessie-cfg80211-calling-crda-to-update-world-regulatory-domain) three valid suggestions are proposed and tried but none of them are working and I find no documentation on the official RPi Forum that explains what the correct way of doing it is.

I still haven't been able to test in depth yet, but will eventually get there...
@clivem thanks. As far as I understand it, your suggested monitor-line only shows what is happening from when it is executed and not during boot. Correct?

clivem, 2016-08-02:
@Ruffio Rasmus, I don't know if this is going to help you in any way, or muddy the waters because I'm using Fedora (F24) and my network connections are managed by NetworkManager, but anyway, I think this is working correctly..... (Or at least, the way I expect things to be working.) By which I mean, that World regulatory is "updated" followed by crda setting "GB" because of my London timezone.

NB. In the case of the Atheros dongle, it's CN, then GB. This is because the Atheros dongle has CN (China) programmed in hardware and the ath9k driver sets the regulatory based on that before it is changed again, based on timezone, via `/lib/udev/rules.d/85-regulatory.rules`.

Example 1. Using on-board Pi3B brcmfmac.
```
Feb 11 16:28:08 IQPlus3B1 kernel: usbcore: registered new interface driver brcmfmac
Feb 11 16:28:08 IQPlus3B1 kernel: cfg80211: World regulatory domain updated:
Feb 11 16:28:08 IQPlus3B1 kernel: cfg80211:  DFS Master region: unset
Feb 11 16:28:08 IQPlus3B1 kernel: cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
Feb 11 16:28:08 IQPlus3B1 kernel: cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
Feb 11 16:28:08 IQPlus3B1 kernel: cfg80211:   (2457000 KHz - 2482000 KHz @ 20000 KHz, 92000 KHz AUTO), (N/A, 2000 mBm), (N/A)
Feb 11 16:28:08 IQPlus3B1 kernel: cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
Feb 11 16:28:08 IQPlus3B1 kernel: cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
Feb 11 16:28:08 IQPlus3B1 kernel: cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
Feb 11 16:28:08 IQPlus3B1 kernel: cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
Feb 11 16:28:08 IQPlus3B1 kernel: cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
Feb 11 16:28:08 IQPlus3B1 kernel: cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
Feb 11 16:28:09 IQPlus3B1 kernel: brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: May 27 2016 00:13:38 version 7.45.41.26 (r640327) FWID 01-df77e4a7
Feb 11 16:28:09 IQPlus3B1 kernel: brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
Feb 11 16:28:09 IQPlus3B1 crda[276]: setting regulatory domain to GB based on timezone (Europe/London)
Feb 11 16:28:09 IQPlus3B1 kernel: cfg80211: Regulatory domain changed to country: GB
Feb 11 16:28:09 IQPlus3B1 kernel: cfg80211:  DFS Master region: ETSI
Feb 11 16:28:09 IQPlus3B1 kernel: cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
Feb 11 16:28:09 IQPlus3B1 kernel: cfg80211:   (2402000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
Feb 11 16:28:09 IQPlus3B1 kernel: cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
Feb 11 16:28:09 IQPlus3B1 kernel: cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
Feb 11 16:28:09 IQPlus3B1 kernel: cfg80211:   (5490000 KHz - 5710000 KHz @ 160000 KHz), (N/A, 2700 mBm), (0 s)
Feb 11 16:28:09 IQPlus3B1 kernel: cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
Feb 11 16:28:13 IQPlus3B1 NetworkManager[375]: <info>  [1455208093.5517] rfkill0: found WiFi radio killswitch (at /sys/devices/platform/soc/3f300000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/rfkill0) (driver brcmfmac_sdio)
Feb 11 16:28:13 IQPlus3B1 kernel: brcmfmac: power management disabled
Feb 11 16:28:15 IQPlus3B1 kernel: brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists
Feb 11 16:28:15 IQPlus3B1 kernel: brcmfmac: brcmf_add_if: ignore IF event
Feb 11 16:28:15 IQPlus3B1 kernel: brcmfmac: power management disabled
Feb 11 16:28:15 IQPlus3B1 dbus[363]: [system] Activating via systemd: service name='fi.w1.wpa_supplicant1' unit='wpa_supplicant.service'
Feb 11 16:28:16 IQPlus3B1 dbus[363]: [system] Successfully activated service 'fi.w1.wpa_supplicant1'
Feb 11 16:28:16 IQPlus3B1 audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=wpa_supplicant comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Feb 11 16:28:16 IQPlus3B1 wpa_supplicant[459]: Successfully initialized wpa_supplicant
Feb 11 16:28:16 IQPlus3B1 NetworkManager[375]: <info>  [1455208096.0965] supplicant: wpa_supplicant running
Feb 11 16:28:16 IQPlus3B1 kernel: brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists
Feb 11 16:28:16 IQPlus3B1 kernel: brcmfmac: brcmf_add_if: ignore IF event
Feb 11 16:28:16 IQPlus3B1 wpa_supplicant[459]: Could not read interface p2p-dev-wlan0 flags: No such device
Feb 11 16:28:17 IQPlus3B1 wpa_supplicant[459]: wlan0: Trying to associate with SSID 'BTHub4-FM2W'
Feb 11 16:28:17 IQPlus3B1 wpa_supplicant[459]: wlan0: Associated with 4c:5e:0c:6c:21:ac
Feb 11 16:28:17 IQPlus3B1 wpa_supplicant[459]: wlan0: WPA: Key negotiation completed with 4c:5e:0c:6c:21:ac [PTK=CCMP GTK=CCMP]
Feb 11 16:28:17 IQPlus3B1 wpa_supplicant[459]: wlan0: CTRL-EVENT-CONNECTED - Connection to 4c:5e:0c:6c:21:ac completed [id=0 id_str=]
```

Example 2, using mt7601u dongle.
```
Feb 11 16:28:14 IQDigi2B1 kernel: cfg80211: World regulatory domain updated:
Feb 11 16:28:14 IQDigi2B1 kernel: cfg80211:  DFS Master region: unset
Feb 11 16:28:14 IQDigi2B1 kernel: cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[... cut ...]
Feb 11 16:28:26 IQDigi3B1 wpa_supplicant[515]: wlan0: SME: Trying to authenticate with 4c:5e:0c:6c:21:ac (SSID='BTHub4-FM2W' freq=2412 MHz)
Feb 11 16:28:27 IQDigi3B1 wpa_supplicant[515]: wlan0: Trying to associate with 4c:5e:0c:6c:21:ac (SSID='BTHub4-FM2W' freq=2412 MHz)
Feb 11 16:28:27 IQDigi3B1 wpa_supplicant[515]: wlan0: Associated with 4c:5e:0c:6c:21:ac
Feb 11 16:28:27 IQDigi3B1 wpa_supplicant[515]: wlan0: WPA: Key negotiation completed with 4c:5e:0c:6c:21:ac [PTK=CCMP GTK=CCMP]
Feb 11 16:28:27 IQDigi3B1 wpa_supplicant[515]: wlan0: CTRL-EVENT-CONNECTED - Connection to 4c:5e:0c:6c:21:ac completed [id=0 id_str=]
```

Ruffio, 2016-08-02:
Looking at my AP, I do see the country set to DE even I live in DK, so that must be the reason why the driver changes. That should IMHO be configurable not to align. with AP:

BSS 8c:10:d4:9a:6f:46(on wlan0) -- associated
        TSF: 0 usec (0d, 00:00:00)
        freq: 2427
        beacon interval: 100 TUs
        capability: ESS Privacy SpectrumMgmt ShortSlotTime RadioMeasure (0x1511)
        signal: -10.00 dBm
        last seen: 0 ms ago
        SSID: xxx
        Supported rates: 1.0* 2.0* 5.5* 11.0* 18.0 24.0 36.0 54.0
        DS Parameter set: channel 4
        **Country: DE     Environment: Indoor/Outdoor**
                Channels [1 - 13] @ 20 dBm
        Power constraint: 0 dB
        TPC report: TX power: 16 dBm
        ERP: Barker_Preamble_Mode
        ERP D4.0: Barker_Preamble_Mode
        RSN:     * Version: 1
                 * Group cipher: CCMP
                 * Pairwise ciphers: CCMP
                 * Authentication suites: PSK
                 * Capabilities: 16-PTKSA-RC 1-GTKSA-RC (0x000c)
        Extended supported rates: 6.0 9.0 12.0 48.0
        BSS Load:
                 * station count: 7
                 * channel utilisation: 32/255
                 * available admission capacity: 0 [*32us]
        HT capabilities:
                Capabilities: 0x19ad
                        RX LDPC
                        HT20
                        SM Power Save disabled
                        RX HT20 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 4 usec (0x05)
                HT RX MCS rate indexes supported: 0-23
                HT TX MCS rate indexes are undefined
        HT operation:
                 * primary channel: 4
                 * secondary channel offset: no secondary
                 * STA channel width: 20 MHz
                 * RIFS: 0
                 * HT protection: non-HT mixed
                 * non-GF present: 1
                 * OBSS non-GF present: 1
                 * dual beacon: 0
                 * dual CTS protection: 0
                 * STBC beacon: 0
                 * L-SIG TXOP Prot: 0
                 * PCO active: 0
                 * PCO phase: 0
        Extended capabilities: Extended Channel Switching, BSS Transition, 6
        WPS:     * Version: 1.0
                 * Wi-Fi Protected Setup State: 2 (Configured)
                 * Response Type: 3 (AP)
                 * UUID: 89a5682a-45e7-18e8-3996-3d5f1f735d51
                 * Manufacturer: Sagemcom
                 * Model: Sagemcom
                 * Model Number: 123456
                 * Serial Number: 0000001
                 * Primary Device Type: 6-0050f204-1
                 * Device name: SagemcomAP
                 * Config methods: Display
                 * RF Bands: 0x3
                 * Unknown TLV (0x1049, 6 bytes): 00 37 2a 00 01 20
        WMM:     * Parameter version 1
                 * u-APSD
                 * BE: CW 15-1023, AIFSN 3
                 * BK: CW 15-1023, AIFSN 7
                 * VI: CW 7-15, AIFSN 2, TXOP 3008 usec
                 * VO: CW 3-7, AIFSN 2, TXOP 1504 usec

Regarding the error **[ 12.324228] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code**
I have found this: https://patchwork.kernel.org/patch/8336131/

clivem, 2016-08-03:
> Looking at my AP, I do see the country set to DE even I live in DK, so that must be the reason why the driver changes. That should IMHO be configurable not to align. with AP:

The whole point of the Linux kernel wifi code respecting 802.11d, if being broadcast by an AP, is to ensure radio regulatory compliance from the Linux side of things. I'd suggest that the problem is with the configuration of your AP, not the Linux wifi stack. ;)

Ruffio, 2016-08-03:
@clivem I agree and the one part, but how do you explain the brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code  before the Interface has connected to the AP? Where is it that the driver is looking for ISO3166 code and potentially misread it? As far as I can find out, that is in etc/wpa_supplicant/wpa_supplicant.conf and  /etc/default/crda where I already have configured it to be DK, so why does it throw that error?

clivem, 2016-08-03:
> how do you explain the brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code before the Interface has connected to the AP

Like the Atheros which implements a mapping table in the driver, the (BRCM) firmware is using different codes, to the actual 2 digit country codes. That patch you referenced implements a mechanism to lookup a country code in a driver and convert it to what they are actually using internally. (Although, I didn't notice any actual mapping table.)
Also, ISTR the driver code is throwing that warning/error when it receives any country code that isn't comprised of A-Z chars. It seems to happen after the world defaults are set from cfg80211. It's possibly just the reg_notifier callback into the driver after the world defaults are set from cfg80211 driver, and can probably be ignored........

Arend Van Spriel is probably the BRCM guy you need to start talking to, for definitive answers on anything concerning BRCM wifi. His email address is in that patch you referenced.

Ruffio, 2016-08-03:
@clivem thanks for your response, and please bear over with me if/when there is something I miss/don't understand.
As I'm developing an application to be runned on RPi, I really don't like when errors are thrown. In my (naive) world one should try to avoid errors and that is the reason why I'm having focus on this. In my world, if I configure it correctly, then there will (ideally) be no errors.

In your response I can't quite figure out if the driver can be configured/implemented differently in the various implementation. I have tried to find out how to configure it, but I can't find any documentation regarding this on the official RPi homepage. I know that it is not supposed to be a wiki with information of everything, and that is why I have tried to use configurations that other implementation have used.

Since noone can answer me how to configure it correctly so no errors are thrown, I guess that nobody really knows? I could understand it if I had configured it in a very special and never used way, but this is how it works out of the box, without any configuration.

Your suggestion regarding contacting Arend, that I will follow and update this issue. Hopefully soon :-) Thanks all for your contributions.

clivem, 2016-08-03:
@Ruffio Sorry, I'm not doing a very good job of explaining..... Even when I have an understanding in my own head, I'm not so good at putting it into words. Let me try again....

The error you see, "brcmf_cfg80211_reg_notifier: not a ISO3166 code", is because cfg80211 sets the WORLD '00' profile before a more specific country one, either because a country is programmed into the wifi device, or you are setting a default from userspace. That "default" WORLD profile has a country code of '00'. The brcmfmac reg_notifier callback checks for alpha chars, and "errors" on anything else, so it says '00'  is "not a ISO3166 code".
 
```
	/* ignore non-ISO3166 country codes */
	for (i = 0; i < sizeof(req->alpha2); i++)
		if (req->alpha2[i] < 'A' || req->alpha2[i] > 'Z') {
			brcmf_err("not a ISO3166 code\n");
			return;
		}
```

All this means, AFAICT, is that the country code of '00' isn't passed back to the firmware by the driver.

Later, you are setting a profile of 'DK'. This is "passed" by the driver to the device firmware.

I agree, that in an ideal world there would be no errors, but I think this error, that '00' isn't a country code and it is being ignored by the hardware, isn't really something you need to be concerned about.

Ruffio, 2016-08-03:
Thanks.

I wrote to Arend that quickly responded with the below. I guess there is nothing more to be done here as it is working as designed, even I don't agree on throwing an error if nothing can be done/configured to prevent this.

I can. This is not really a problem. The regulatory code in cfg80211 has
a world regulatory domain which is identified with "00". It notifies the
brcmfmac driver about this domain, but "00" is not a proper ISO3166 code
and as such there is no need to handle this in the driver. We could add
an exception for "00" in the driver, ie. ignore it without logging an
error message.

Regards,
Arend

closed 2016-08-03 by Ruffio