raspberrypi/firmware #631: Kernel trace in cfg80211

herrernst, 2016-07-23:
I'm seeing this in dmesg after reboot with the current kernel (don't remember having seen this before, but not sure):
```
[    8.874102] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   19.640719] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   19.647554] ------------[ cut here ]------------
[   19.652468] WARNING: CPU: 1 PID: 200 at net/wireless/sme.c:714 __cfg80211_connect_result+0x2cc/0x3e4 [cfg80211]()
[   19.662889] Modules linked in: binfmt_misc cpufreq_stats nfsd 8192cu snd_soc_wm8804_i2c cfg80211 snd_soc_wm8804 regmap_i2c rfkill snd_soc_bcm2835_i2s snd_soc_hifiberry_digi snd_soc_core snd_pcm_dmaengine snd_pcm snd_timer snd i2c_bcm2708 bcm2835_gpiomem bcm2835_wdt uio_pdrv_genirq uio ipv6
[   19.689154] CPU: 1 PID: 200 Comm: kworker/u8:3 Not tainted 4.4.13-v7+ #894
[   19.696095] Hardware name: BCM2709
[   19.699618] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[   19.705546] [<80018724>] (unwind_backtrace) from [<80014058>] (show_stack+0x20/0x24)
[   19.713380] [<80014058>] (show_stack) from [<80320744>] (dump_stack+0xd4/0x118)
[   19.720771] [<80320744>] (dump_stack) from [<80025300>] (warn_slowpath_common+0x98/0xc8)
[   19.728949] [<80025300>] (warn_slowpath_common) from [<800253ec>] (warn_slowpath_null+0x2c/0x34)
[   19.737888] [<800253ec>] (warn_slowpath_null) from [<7f1c9f94>] (__cfg80211_connect_result+0x2cc/0x3e4 [cfg80211])
[   19.748460] [<7f1c9f94>] (__cfg80211_connect_result [cfg80211]) from [<7f1a2580>] (cfg80211_process_wdev_events+0x160/0x1a0 [cfg80211])
[   19.763445] [<7f1a2580>] (cfg80211_process_wdev_events [cfg80211]) from [<7f1a25fc>] (cfg80211_process_rdev_events+0x3c/0x70 [cfg80211])
[   19.781204] [<7f1a25fc>] (cfg80211_process_rdev_events [cfg80211]) from [<7f19d28c>] (cfg80211_event_work+0x24/0x2c [cfg80211])
[   19.798082] [<7f19d28c>] (cfg80211_event_work [cfg80211]) from [<8003c644>] (process_one_work+0x154/0x458)
[   19.813165] [<8003c644>] (process_one_work) from [<8003c99c>] (worker_thread+0x54/0x500)
[   19.826749] [<8003c99c>] (worker_thread) from [<80042678>] (kthread+0xec/0x104)
[   19.836914] [<80042678>] (kthread) from [<8000fbc8>] (ret_from_fork+0x14/0x2c)
[   19.847050] ---[ end trace 33b07efe39ceebfd ]---
[   25.515799] ip_tables: (C) 2000-2006 Netfilter Core Team
```
I'm using a WiFi USB stick: Edimax Technology Co., Ltd EW-7811Un 802.11n Wireless Adapter [Realtek RTL8188CUS]
WiFi is still working though.
Kernel: Linux rpi2deb 4.4.13-v7+ #894 SMP Mon Jun 13 13:13:27 BST 2016 armv7l GNU/Linux

dlech, 2016-07-23:
Sounds like a side effect of https://github.com/raspberrypi/linux/pull/1489

tootai, 2016-10-29:
This bug is still existing in 4.4.26-v7+

Will this be corrected or is there a workaround ?

Daniel

tootai, 2016-10-29:
Just did a rpi-update which brought 4.4.27+ Problem is still there :(

JamesH65, 2017-12-21:
This is a linux driver issue, not firmware, (if it still exists on our current kernel), and Realtek drivers are undergoing major changes. If the issue still exists , please open a NEW issue in the Pi linux repo.

closed 2017-12-21 by JamesH65

jwulf, 2018-12-28:
Try this: https://www.raspberrypi.org/forums/viewtopic.php?t=61665#p458761