raspberrypi/firmware #632: Custom device_tree_address causes initramfs fail to load?

RomanSaveljev, 2016-07-24:
This may be a misleading title, but you know how debugging proprietary code goes.. I am happy to answer all the clarifying questions!

When I put `device_tree_address=0x100` to `config.txt`, the kernel log has a bunch of warnings like these:

```
[    0.057161] device-tree: Duplicate name in display, renamed to "#1"
[    0.057205] device-tree: Duplicate name in display, renamed to "#2"
[    0.057262] device-tree: Duplicate name in system, renamed to "#1"
[    0.057426] device-tree: Duplicate name in aliases, renamed to "#1"
[    0.057468] device-tree: Duplicate name in aliases, renamed to "#2"
[    0.057518] device-tree: Duplicate name in aliases, renamed to "#3"
[    0.057565] device-tree: Duplicate name in aliases, renamed to "#4"
```

The side effect of this is that custom initramfs does not get executed. Here is, for instance, the output of `vcdgb log msg`, when I have `initramfs initramfs.cpio.gz 0x02100000` in `config.txt`:

```
000945.477: HDMI:EDID error reading EDID block 0 attempt 0
000946.740: HDMI:EDID error reading EDID block 0 attempt 1
000947.999: HDMI:EDID error reading EDID block 0 attempt 2
000949.257: HDMI:EDID error reading EDID block 0 attempt 3
000950.516: HDMI:EDID error reading EDID block 0 attempt 4
000951.774: HDMI:EDID error reading EDID block 0 attempt 5
000953.033: HDMI:EDID error reading EDID block 0 attempt 6
000954.291: HDMI:EDID error reading EDID block 0 attempt 7
000955.550: HDMI:EDID error reading EDID block 0 attempt 8
000956.808: HDMI:EDID error reading EDID block 0 attempt 9
000957.830: HDMI:EDID giving up on reading EDID block 0
000973.033: HDMI:Setting property pixel encoding to Default
000973.053: HDMI:Setting property pixel clock type to PAL
000973.071: HDMI:Setting property content type flag to No data
000973.091: HDMI:Setting property fuzzy format match to enabled
001161.006: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
001161.031: hdmi: HDMI:>>>>>>>>>>>>>Rx sensed, reading EDID<<<<<<<<<<<<<
001161.415: hdmi: HDMI:EDID error reading EDID block 0 attempt 0
001162.684: hdmi: HDMI:EDID error reading EDID block 0 attempt 1
001163.948: hdmi: HDMI:EDID error reading EDID block 0 attempt 2
001165.211: hdmi: HDMI:EDID error reading EDID block 0 attempt 3
001166.474: hdmi: HDMI:EDID error reading EDID block 0 attempt 4
001167.738: hdmi: HDMI:EDID error reading EDID block 0 attempt 5
001169.001: hdmi: HDMI:EDID error reading EDID block 0 attempt 6
001170.265: hdmi: HDMI:EDID error reading EDID block 0 attempt 7
001171.528: hdmi: HDMI:EDID error reading EDID block 0 attempt 8
001172.792: hdmi: HDMI:EDID error reading EDID block 0 attempt 9
001173.820: hdmi: HDMI:EDID giving up on reading EDID block 0
001173.851: hdmi: HDMI: No lookup table for resolution group 0
001173.867: hdmi: HDMI: hotplug attached with DVI support
001173.925: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
001174.288: hdmi: HDMI:EDID error reading EDID block 0 attempt 0
001175.550: hdmi: HDMI:EDID error reading EDID block 0 attempt 1
001176.812: hdmi: HDMI:EDID error reading EDID block 0 attempt 2
001178.074: hdmi: HDMI:EDID error reading EDID block 0 attempt 3
001179.336: hdmi: HDMI:EDID error reading EDID block 0 attempt 4
001180.598: hdmi: HDMI:EDID error reading EDID block 0 attempt 5
001181.860: hdmi: HDMI:EDID error reading EDID block 0 attempt 6
001183.123: hdmi: HDMI:EDID error reading EDID block 0 attempt 7
001184.385: hdmi: HDMI:EDID error reading EDID block 0 attempt 8
001185.647: hdmi: HDMI:EDID error reading EDID block 0 attempt 9
001186.672: hdmi: HDMI:EDID giving up on reading EDID block 0
001187.154: hdmi: HDMI: hotplug deassert
001187.170: hdmi: HDMI: HDMI is currently off
001187.184: hdmi: HDMI: changing mode to unplugged
001187.206: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
001188.401: *** Restart logging
001189.413: Failed to open command line file 'cmdline.txt'
001438.503: Loading 'kernel7.img' to 0x8000 size 0x407460
001623.645: Loading 'initramfs.cpio.gz' to 0x40f460 size 0x303ec4
002179.079: Kernel trailer DTOK property says yes
002179.093: Kernel trailer DDTK property says yes
002182.593: Loading 'bcm2710-rpi-3-b.dtb' to 0x40f460 size 0x3ae9
002257.806: dtparam: uart0_clkrate=48000000
002339.610: dtparam: arm_freq=1200000000
002382.409: dtparam: core_freq=250000000
002393.741: dtparam: cache_line_size=64
002425.377: Device tree loaded to 0x100 (size 0x4029)
002427.673: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
005167.056: vchiq_core: vchiq_init_state: slot_zero = 0xfac80000, is_master = 1
005171.126: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
```

Looks like `0x40f460` is used as a scratch. Does loading `bcm2710-rpi-3-b.dtb` corrupt initramfs?

pelwell, 2016-07-24:
The clue is in this line:
```
002425.377: Device tree loaded to 0x100 (size 0x4029)
```
Although the kernel starts at 0x8000, something (the decompressor perhaps) uses the RAM from 0x4000. With a starting point of 0x100 that gives a maximum DTB size of 0x3f00. If the combined DTB exceeds that you will have to load it somewhere else.

RomanSaveljev, 2016-07-24:
Thanks @pelwell for a quick hint. I did another test. Instead of `kernel7.img` I gave it `uboot.bin`. I assume U-Boot (I let `mknlimg` on it) will see FDT the same as the real kernel would:

```
U-Boot 2016.07 (Jul 21 2016 - 12:03:50 +0000)

DRAM:  944 MiB
RPI 3 Model B (0xa02082)
MMC:   bcm2835_sdhci: 0
reading uboot.env

** Unable to read "uboot.env" from mmc0:1 **
Using default environment

In:    serial
Out:   lcd
Err:   lcd
Net:   Net Initialization Skipped
No ethernet found.
starting USB...
USB0:   Core Release: 2.80a
scanning bus 0 for devices... 3 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
       scanning usb for ethernet devices... 1 Ethernet Device(s) found
Hit any key to stop autoboot:  0 
U-Boot> md 0x100
00000100: edfe0dd0 29400000 38000000 bc390000    ......@)...8..9.
00000110: 28000000 11000000 10000000 00000000    ...(............
00000120: 6d060000 84390000 00000000 00000000    ...m..9.........
00000130: 00000000 00000000 01000000 00000000    ................
00000140: 03000000 08000000 d6050000 0000003b    ............;...
00000150: 00000004 03000000 04000000 00000000    ................
00000160: 01000000 03000000 04000000 0f000000    ................
00000170: 01000000 03000000 04000000 1b000000    ................
00000180: 01000000 03000000 1a000000 2c000000    ...............,
00000190: 6d637262 6d63622c 30313732 63726200    brcm,bcm2710.brc
000001a0: 63622c6d 3037326d 00000039 03000000    m,bcm2709.......
000001b0: 1f000000 37000000 70736152 72726562    .......7Raspberr
000001c0: 69502079 4d203320 6c65646f 52204220    y Pi 3 Model B R
000001d0: 31207665 7300322e 01000000 70736964    ev 1.2.s....disp
000001e0: 0079616c 03000000 04000000 3a060000    lay............:
000001f0: 10000000 03000000 04000000 2a060000    ...............*
U-Boot> fdt addr 0x100
U-Boot> fdt header
magic:                  0xd00dfeed
totalsize:              0x4029 (16425)
off_dt_struct:          0x38
off_dt_strings:         0x39bc
off_mem_rsvmap:         0x28
version:                17
last_comp_version:      16
boot_cpuid_phys:        0x0
size_dt_strings:        0x66d
size_dt_struct:         0x3984
number mem_rsv:         0x0

U-Boot> fdt print
/ {
        memreserve = <0x3b000000 0x04000000>;
        #address-cells = <0x00000001>;
        #size-cells = <0x00000001>;
        interrupt-parent = <0x00000001>;
...
        };
};
U-Boot> 
```

So, at least the U-Boot does not see it corrupted.

pelwell, 2016-07-24:
We can't change the kernel/decompressor memory usage, so it won't ever be possible to load a larger DTB at 0x100. The new stubs (source code available in the tools repo) include slots for requesting and returning load addresses for the kernel and DTB - these provide an alternative to the config.txt setting, and could also allow other "kernels" like U-Boot to read the DTB from a variable address.

RomanSaveljev, 2016-07-24:
Okay, I will try a different address and report. Thank you!

RomanSaveljev, 2016-07-25:
I have used a default address dynamically calculated by the bootloader:

```
device_tree_address=0x2fffc000
```

Now the kernel no longer complains and initramfs is executed. I will find some "safe" address. Thanks for your support again

closed 2016-07-25 by RomanSaveljev