raspberrypi/firmware #633: openvg : vguPolygon() can corrupt vertex coordinates on long paths.

paeryn, 2016-07-26:
Whilst testing something I noticed that when creating large paths with vguPolygon() (has happened with as low as 246 vertices in my original though in the test program below it starts breaking at about 512) it ends up with corrupt coordinates sort of like it's getting chunks of coordinate data out of order (and half way through a vertex so what should be a vertex's Y ends up as the X from the next etc.). This is repeatedly happening (always the same every run).

I've made a minimal test program that should draw a sine wave horizontally, but when creating the path with vguPolygon() you get half of it or so drawn horizontally (as it should), then the rest (or parts of) is drawn vertically. Creating the polygon directly with vgAppendPathData() on exactly the same data works fine so there's no problem with the length of the path.

[Sample program and screenshots](https://www.dropbox.com/s/3ltu07wp8wnnsiu/vgupolygon.tar.gz?dl=0) Screenshots are from the original bigger program because it saves png directly but the line function and drawing routines are the same, although the corruption happens at different offsets along the wave.

[sine_append.png](https://www.dropbox.com/s/ed277lkmjishj3w/sine_append.png?dl=0) is what it looks like on my RPi 3 when I create the path myself with vgAppendPathData()
[sine_vgu.png](https://www.dropbox.com/s/q7fkto3ekyerk5w/sine_vgu.png?dl=0) is what it looks like when vguPolygon() creates the path with the same data.

popcornmix, 2016-07-26:
Thanks for sample code. Can you try editing this line:
https://github.com/raspberrypi/userland/blob/master/interface/khronos/vg/vg_client.c#L5125
and making it ``ps += 2 * chunk_ps_count;``
and rebuild userland.

paeryn, 2016-07-26:
Cheers, I didn't spot that. It would fit into what I was seeing. 

I'll have a go at rebuilding it tonight (not done it before).

paeryn, 2016-07-26:
That was easy, I was expecting to have to configure a few things first before building...

I can confirm that it fixes the issue, thank you very much.

I'll have a trawl through to see if there's anything similar elsewhere (I know there are some cases where font paths can become corrupt, but I've never been able to reliably get it to happen). At least I know what to look for.

popcornmix, 2016-07-28:
This is now in latest rpi-update firmware.

Ruffio, 2016-08-08:
@paeryn have you tried to upgrade to latest firmware and can confirm that this issue has been resolved? Then please close this issue. Thanks.

paeryn, 2016-08-08:
Sorry, I had been using my locally compiled version with the fix. I've just done an rpi-update and it's working correctly also.

closed 2016-08-08 by paeryn