raspberrypi/firmware #634: Dispmax transformations not working

kapetanos, 2016-07-27:
I am trying to use transformations when adding a new dispmanx element but 90/270 transformations don't seem to be working (180 works).
I noticed that if I configure display_rotate=1 or display_rotate=3 in config.txt, transformations are working.
The issue seems to be happening when display_rotate=0 or display_rotate=2.

Is this a limitation or a bug?

popcornmix, 2016-07-27:
Transposed images require allocation of an off screen buffer, so make sure your gpu_mem is sufficient. Otherwise provide a simple demo app (e.g. based on hello_dispmanx) that shows the problem and I'll have a look.

kapetanos, 2016-07-27:
I am running my tests on RPI3 with Raspbian, 128 mb gpu memory.
Firmware version:
Jun 20 2016 18:13:27 
Copyright (c) 2012 Broadcom
version 460b39c674ef2ac89360e15331f97a2cb1388a44 (clean) (release)

You can check my modified version of hello_dispmanx [here](https://gist.github.com/kapetanos/6f59876176e2cf7fb2850d863f4a1b42). It accepts the rotation as an argument (0, 1, 2, 3). Changes are: FillRect is slightly modified so the orientation of the rectangles is visible on the screen (white lines) and the positioning of the element depending on rotation selected.

All rotations work only when display_rotate=1 or display_rotate=3.

When display_rotate=0 or display_rotate=2,  1 / 3 rotations do not work (nothing shows up on screen).

popcornmix, 2016-07-27:
Yes, looking more closely, this is expected behaviour. To support transposed dispmanx elements you need to add ``dispmanx_offline=1`` to config.txt.
That creates the required buffers needed for the transposer hardware and should make your example work.

Ruffio, 2016-08-07:
@kapetanos have you tried @popcornmix advise? Can this issue be closed? Please do. Thanks.

kapetanos, 2016-08-07:
@popcornmix Thanks!

closed 2016-08-07 by kapetanos