raspberrypi/firmware #635: RPi3, serial driver loaded even when setting enable_uart=0 in /boot/config.txt

Ruffio, 2016-07-27:
In dmseg, I get:
[ 0.288469] Console: switching to colour frame buffer device 228x61
**[ 0.304732] Serial: 8250/16550 driver, 1 ports, IRQ sharing disabled**
[ 1.244267] bcm2835-rng 3f104000.rng: hwrng registered

That is a long (compared to my total of 7.5 seconds) so why is it loading when I'm using the following?

/boot/config.txt:
enable_uart=0
dtoverlay=pi3-disable-bt

/boot/cmdline.txt:
dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait noswap logo.nologo quiet noatime

services disabled by:
sudo systemctl disable bluetooth
sudo systemctl disable hciuart
sudo systemctl disable ModemManager.service

Ruffio, 2016-08-02:
Is it possible to get an indication if the serial driver should load with enable_uart=0 ?
If it is working as designed, then I'm happy to close this issue. Are there any other way (besides rebuilding) to don't get that driver loaded?

Ruffio, 2016-11-22:
@popcornmix any idea if the Serial driver should be loaded even if enable_uart=0 in config.txt?
IMHO I see no reason, but there could be an explanation to this...

pelwell, 2016-11-22:
enable_uart applies to the "primary" UART - the one with the /dev/serial0 alias, that would have a console running it. On a Pi3 that is normally ttyS0 because ttyAMA0 is tied up driving the Bluetooth modem. By applying the pi3-disable-bt overlay you are actually swapping the UARTs back again, so that enable_uart applies to ttyAMA0 instead. If you remove pi3-disable-bt and keep enable_uart=0 (which is  the default while ttyS0 is the primary) you will lose ttyS0, but leave you with ttyAMA0. With `dtparam=uart0=off` you can disable /dev/ttyAMA0, but the driver is still initialised by the kernel.

Unfortunately you will then find that it isn't the serial port that causes the delay in the dmesg log but the RNG initialising - you can't rush entropy.

Ruffio, 2016-11-22:
ok, thanks.

closed 2016-11-22 by Ruffio