raspberrypi/firmware #639: Bootloader-level JTAG flag in config.txt?

xcvista, 2016-08-07:
**TL; DR:** By adding `arm_jtag=1` to `config.txt` or when the kernel file is missing on the SD card, JTAG on the ARM core is enabled before control is handed to the kernel.

This may be a weird idea, but I think it can be useful for newbies in bare metal programming. This allows JTAG being used before a working bare metal kernel is written into the SD card, enabling debugging the earliest parts of the kernel.

The missing-kernel JTAG allows developing kernels without having to switch in and out SD cards (wearing down the card and the connector) or a working bootloader (so serial loading can work - in fact for Pi 3 serial loading may never work due to the UART timer scaling issue.) Kernels can be injected directly into the RAM and booted in-place using JTAG.

Additionally a new flag `arm_autorun` can be introduced. When disabled (default to `arm_autorun=1`) the ARM core will not be released from reset by the GPU after the kernel (if any) is loaded and JTAG is required to kickstart the kernel. This allows JTAG being attached and stabilized even before the first line of code is executed on the ARM core, making early-kernel debugging even easier.

popcornmix, 2016-08-08:
Have you tried ``enable_jtag_gpio=1``?
That sets the alt-function of the gpio pins into jtag mode and should allow connection of an arm debugger.

xcvista, 2016-08-11:
@popcornmix Thanks for this info. However what about that missing-kernel scenario or the disabling of autoboot?

popcornmix, 2016-08-11:
I would suggest you create a kernel.img/kernel7.img files that just has a single instruction that branches back to itself. That should get the arm enabled and doing something safe and allow you to attach the debugger and load whatever code you want.

closed 2016-08-11 by xcvista