raspberrypi/firmware #641: OMX_EmptyThisBuffer hangs

atic-atac, 2016-08-12:
Just opening a new case after looking at issue #449.

It seems I can get the above call to hang when using dispmanx submit() calls, but I can't get it to hang when using submit_sync() calls.

I can try and get to a test case, but maybe there's someone who knows why the above could hang ?

atic-atac, 2016-08-12:
O.k. I have a theory that there is multiple submit() calls outstanding and some queue gets full.

I can workaround the issue. So closing.

atic-atac, 2016-08-12:
I'm going to reopen as I suspect it's worth tracking this so I can understand things.

popcornmix, 2016-08-12:
Well, you'll need to post some test code that shows the problem if you want any feedback.

Make sure you are not calling block GPU calls from GPU callbacks (e.g. dispmanx/openmax) which would be a recipe for deadlocks.

atic-atac, 2016-08-12:
O.k. I'm not going to have time to follow this up at the moment. Maybe reopen later, but for now closing.

closed 2016-08-12 by atic-atac