raspberrypi/firmware #642: Feature Request: A switch to give Linux the control of the GPIO expander on Pi 3

imzyxwvu, 2016-08-17:
I need to turn off the PWR LED since it is really bright at night, but it is connected to the GPIO expander and there is no way to control it in userspace. I don't need the undervoltage warning and the HDMI detection because I have hdmi_force_hotplug=1 in config.txt.

6by9, 2016-08-17:
There's already a mailbox service to control the GPIO expander that can be exercised with https://github.com/6by9/rpi3-gpiovirtbuf (discussion started at https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=109137&start=100#p989832).
It doesn't currently allow control of direction on those GPIOs, and the PWR LED is really a hardware input for reading the undervoltage state. It just so happened that setting it as an output allowed the state to be overridden. There had been a suggestion that the function might be added, but not been done yet that I'm aware of.

6by9, 2016-09-02:
See the end of https://github.com/raspberrypi/linux/issues/1332
Confirmed that the power LED can be disabled via that pin config blob. 
https://github.com/6by9/rpi3-gpiovirtbuf does then allow you to change the state.

popcornmix, 2016-09-02:
> and the PWR LED is really a hardware input for reading the undervoltage state

Just to clarify that forcing the PWR LED will disable the under-voltage detection, so ensure  your power supply is good before doing this.

imzyxwvu, 2016-09-02:
@popcornmix will disabling the PWR LED on a Pi2 disable the under-voltage protection?

popcornmix, 2016-09-02:
Yes, by default the PWD LED is an input with the state of the LED generated by the under-voltage detection circuitry.
If you override this as an output (which is safe) the Pi can no longer detect under-voltage, so no warning symbol on screen, no ``vcgencmd get_throttled`` bit set and no disabling of turbo mode.

imzyxwvu, 2016-09-03:
Solved

closed 2016-09-03 by imzyxwvu