raspberrypi/firmware #643: Can Pi2 UART Clock as Pi3

Cheong2K, 2016-08-17:
I know that, Pi3 bcm2708.uart_clock=48000000, Pi2 bcm2708.uart_clock=3000000.
Is it possible to make the clock of Pi2 the same as Pi3? I want this because I made a HAT, and my users need to put init_uart_clock in the config.txt, the HAT now is not fully plug-&-play.

pelwell, 2016-08-22:
I've dropped a patch into the firmware build machine and a new release should pop out of the other end in the next few days. It will initially be available via rpi-update, but if there are no reports of adverse effects it will be included in a subsequent Raspbian release.

popcornmix, 2016-08-23:
This is in latest ``rpi-update`` firmware.

Cheong2K, 2016-08-23:
Awesome, thank you!

Cheong2K, 2016-09-05:
With Pi Zero, after using `apt-get upgrade`, it will change the UART back to 3 MHz.

pelwell, 2016-09-05:
Yes it will, until we make the next Raspbian release - which should be in the next few weeks. Until then, you need to follow any `apt-get upgrade` with an `rpi-update`.

Cheong2K, 2016-09-06:
Thanks, got it!

pelwell, 2016-10-27:
Please close if appropriate.

closed 2016-10-28 by Cheong2K

0xygen8, 2017-04-18:
I have the same question but the other way around: I'm using a Pi2 to communicate with a device running at 300-e-7-1 I found that when I switched to a Pi3 300bps is no longer a possible baud rate (only garbled data) I tried switching the UART (pi3-disable-bt) but that has no effect. Is there a way to solve this? (I'm using Raspbian Jessie lite dd 2017-04-10)