raspberrypi/firmware #644: Encoding Frame number in OMX_IndexConfigBrcmPortStats

F5OEO, 2016-08-21:
Encoding in H264, SEI Message enable
OMX_IndexConfigBrcmPortStats.nFrameCount could not be used to count Video frame number : at each time there is a BUFFERFLAG_CODECCONFIG, framenumber is incremented even it is not a video frame.

Incrementing a local counter at each ENDOFFRAME and comparing with OMX_IndexConfigBrcmPortStats.nFrameCount does not fit : difference is the number of buffer with CODECCONFIG.

Is it normal that a CODECCONFIG is count as a frame ?

popcornmix, 2016-08-22:
@6by9 may be able to describe what the intended behaviour is.

6by9, 2016-08-22:
OMX_IndexConfigBrcmPortStats was only ever a debug feature to profile how the encoder was performing. Neither inline headers nor motion vectors were in used in production, so behaviour with either of those enabled is undefined. It's also a custom Broadcom extension, so doesn't have to conform to any spec.

The code is simply counting the number of buffers that go out with the ENDOFFRAME flag set. Headers  should have ENDOFFRAME | CODECCONFIG set, so they are included in the count.
It differs from stats.nBufferCount in that that counts every single buffer that is sent, as the encoder can fragment the H264 output frames.
Your comment 
> Incrementing a local counter at each ENDOFFRAME and comparing with OMX_IndexConfigBrcmPortStats.nFrameCount does not fit : difference is the number of buffer with CODECCONFIG.

sounds odd, but that may be because you are branching in your code to handle CODECCONFIG being set before you then look at ENDOFFRAME for your count. It also seems to contradict your first paragraph.

Technically with respect to the H264 stream the headers are part of the framed data, they just don't contain an image frame, so I would expect them to be included in the count.

6by9, 2016-08-24:
> As I read in OpenMaxIL document, A frame is normally an optionnal header+image. We speak about frame by second for example : 50 Frame Per second for example do not count header.

Those would be unencoded frames, which is not what we are discussing here.
We are discussing H264 encoded data, so a frame can be considered to be a NAL unit, and both the SPS and PPS headers are independent NAL units (perhaps they should be split into two separate buffers, as the IL spec does say "No buffer payload can contain data from two separate frames")
With H264 there are options for splitting the one image frame into multiple slices too, and from my (possibly inaccurate) knowledge each of those slices would show up with the ENDOFFRAME flag set as each is a new NAL unit. How would you want that counting?

> at each time there is a BUFFERFLAG_CODECCONFIG, framenumber is incremented even it is not a video frame.

vs
> Incrementing a local counter at each ENDOFFRAME and comparing with OMX_IndexConfigBrcmPortStats.nFrameCount does not fit : difference is the number of buffer with CODECCONFIG.

Those still sound contradictory to me.
First case you're saying that CODECCONFIG|ENDOFFRAME flags increment nFrameCount.
Second case (with clarification of checking each callback) you're saying that there is a difference between the number of callbacks counted (including those with CODECCONFIG set) and the value reported by nFrameCount.

From my point of view, this is a custom index that does what it does. It doesn't have to conform to any external specification, and I won't be looking to make any backward incompatible changes to it. @popcornmix may decide to do otherwise, but he can do that as a Raspberry Pi employee (I'm not).

F5OEO, 2016-08-24:
I try to be clearer with an example : here is a debug of what is happening with a encoded mutlislice H264, IDR every 5 frames 
You can see that :

- Each time there is an IDR Frame, there is CODECCONFIG ENDOFNAL which increment FrameNumber
- After that  ENDOFFRAME CODECCONFIG ENDOFNAL which **AGAIN** increment FrameNumber
- Even there are multislices (encapsulated in NAL), the frame count is not incremented

Few remarks : 

>  Those would be unencoded frames, which is not what we are discussing here.

Even on encoded stream there is a notion of FrameRate.

> so a frame can be considered to be a NAL unit

No , a frame could be separated in multiple NALU

> No buffer payload can contain data from two separate frames

Just say that we cannot have more that one frame in a IL buffer : normal , because FLAGS will be irrelevant 

> From my point of view, this is a custom index that does what it does. It doesn't have to conform to any external specification, and I won't be looking to make any backward incompatible changes to it.

@popcornmix 
Understand but by correcting the behaviour, it should simplify calculation of number of frame which is implemented "by hand". Usefull to have Frame number to calculate the time we are.

```
VideoStat : CODECCONFIG ENDOFNAL  ByteCount 47 Buffer 2 - Frame 0 = 2 Skip 0 Discard 0 Max Delta0:0 TIME 343002636
VideoStat : ENDOFFRAME CODECCONFIG ENDOFNAL  ByteCount 56 Buffer 4 - Frame 1 = 2 Skip 0 Discard 0 Max Delta0:0 TIME 1
VideoStat : SYNCFRAME ENDOFNAL  ByteCount 72 Buffer 6 - Frame 1 = 4 Skip 0 Discard 0 Max Delta0:0 TIME 100
VideoStat : SYNCFRAME ENDOFNAL  ByteCount 88 Buffer 8 - Frame 1 = 6 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : SYNCFRAME ENDOFNAL  ByteCount 474 Buffer 10 - Frame 1 = 8 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : ENDOFFRAME SYNCFRAME ENDOFNAL  ByteCount 1387 Buffer 12 - Frame 2 = 8 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : ENDOFNAL  ByteCount 1403 Buffer 14 - Frame 2 = 10 Skip 0 Discard 0 Max Delta0:0 TIME 44
VideoStat : ENDOFFRAME ENDOFNAL  ByteCount 1415 Buffer 16 - Frame 3 = 10 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : ENDOFNAL  ByteCount 1430 Buffer 18 - Frame 3 = 12 Skip 0 Discard 0 Max Delta0:0 TIME 53
VideoStat : ENDOFFRAME ENDOFNAL  ByteCount 2854 Buffer 20 - Frame 4 = 12 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : ENDOFNAL  ByteCount 2869 Buffer 22 - Frame 4 = 14 Skip 0 Discard 0 Max Delta0:0 TIME 37
VideoStat : ENDOFFRAME ENDOFNAL  ByteCount 4003 Buffer 24 - Frame 5 = 14 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : ENDOFNAL  ByteCount 4018 Buffer 26 - Frame 5 = 16 Skip 0 Discard 0 Max Delta0:0 TIME 27
VideoStat : ENDOFFRAME ENDOFNAL  ByteCount 4040 Buffer 28 - Frame 6 = 16 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : CODECCONFIG ENDOFNAL  ByteCount 4087 Buffer 30 - Frame 6 = 18 Skip 0 Discard 0 Max Delta0:0 TIME 32
VideoStat : ENDOFFRAME CODECCONFIG ENDOFNAL  ByteCount 4096 Buffer 32 - Frame 7 = 18 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : ENDOFNAL  ByteCount 4111 Buffer 34 - Frame 7 = 20 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : ENDOFFRAME ENDOFNAL  ByteCount 4818 Buffer 36 - Frame 8 = 20 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : CODECCONFIG ENDOFNAL  ByteCount 4865 Buffer 38 - Frame 8 = 22 Skip 0 Discard 0 Max Delta0:0 TIME 53
VideoStat : ENDOFFRAME CODECCONFIG ENDOFNAL  ByteCount 4874 Buffer 40 - Frame 9 = 22 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : ENDOFNAL  ByteCount 4889 Buffer 42 - Frame 9 = 24 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : ENDOFFRAME ENDOFNAL  ByteCount 5312 Buffer 44 - Frame 10 = 24 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : CODECCONFIG ENDOFNAL  ByteCount 5359 Buffer 46 - Frame 10 = 26 Skip 0 Discard 0 Max Delta0:0 TIME 31
VideoStat : ENDOFFRAME CODECCONFIG ENDOFNAL  ByteCount 5368 Buffer 48 - Frame 11 = 26 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : ENDOFNAL  ByteCount 5383 Buffer 50 - Frame 11 = 28 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : ENDOFFRAME ENDOFNAL  ByteCount 5406 Buffer 52 - Frame 12 = 28 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : CODECCONFIG ENDOFNAL  ByteCount 5453 Buffer 54 - Frame 12 = 30 Skip 0 Discard 0 Max Delta0:0 TIME 40
VideoStat : ENDOFFRAME CODECCONFIG ENDOFNAL  ByteCount 5462 Buffer 56 - Frame 13 = 30 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : ENDOFNAL  ByteCount 5477 Buffer 58 - Frame 13 = 32 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : ENDOFFRAME ENDOFNAL  ByteCount 5830 Buffer 60 - Frame 14 = 32 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : CODECCONFIG ENDOFNAL  ByteCount 5877 Buffer 62 - Frame 14 = 34 Skip 0 Discard 0 Max Delta0:0 TIME 37
VideoStat : ENDOFFRAME CODECCONFIG ENDOFNAL  ByteCount 5886 Buffer 64 - Frame 15 = 34 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : ENDOFNAL  ByteCount 5901 Buffer 66 - Frame 15 = 36 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : ENDOFFRAME ENDOFNAL  ByteCount 6189 Buffer 68 - Frame 16 = 36 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : CODECCONFIG ENDOFNAL  ByteCount 6236 Buffer 70 - Frame 16 = 38 Skip 0 Discard 0 Max Delta0:0 TIME 26
VideoStat : ENDOFFRAME CODECCONFIG ENDOFNAL  ByteCount 6245 Buffer 72 - Frame 17 = 38 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : ENDOFNAL  ByteCount 6260 Buffer 74 - Frame 17 = 40 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : ENDOFFRAME ENDOFNAL  ByteCount 6622 Buffer 76 - Frame 18 = 40 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : CODECCONFIG ENDOFNAL  ByteCount 6669 Buffer 78 - Frame 18 = 42 Skip 0 Discard 0 Max Delta0:0 TIME 51
VideoStat : ENDOFFRAME CODECCONFIG ENDOFNAL  ByteCount 6678 Buffer 80 - Frame 19 = 42 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : ENDOFNAL  ByteCount 6693 Buffer 82 - Frame 19 = 44 Skip 0 Discard 0 Max Delta0:0 TIME 0
VideoStat : ENDOFFRAME ENDOFNAL  ByteCount 7017 Buffer 84 - Frame 20 = 44 Skip 0 Discard 0 Max Delta0:0 TIME 0
```

F5OEO, 2016-09-07:
Any comment @6by9 @popcornmix  ?

6by9, 2016-09-07:
> From my point of view, this is a custom index that does what it does. It doesn't have to conform to any external specification, and I won't be looking to make any backward incompatible changes to it.

For my own sanity, I did a quick test. Hacked version of raspivid pushed to https://github.com/6by9/userland/tree/fw644. Log of the output at https://github.com/6by9/userland/blob/fw644/log.txt, by using the command
```raspivid -w 640 -h 480 -o /dev/null -ih -n -g 5 2> log.txt```
buffer_count is odd as it seems to be counting each buffer twice.
frame_count seems to be doing as I'd expect in counting buffers with ENDOFFRAME (0x04 in MMAL) set.
I haven't tried it with multiple slices per frame. I'll do a quick dig further to get suitable values in MMAL.

NB Use of this index is likely to give incorrect results if you have more than one buffer on the port. There is a reasonable chance that the port can have emitted an additional buffer(s) before your callback has woken up and queried the port stats, and now you're out of sync.

The original use was for regression testing. Set the pipeline running for N seconds, stop the pipe, query the number of frames and frame drops that occurred during that period, and check it was in tolerance. It was never intended as a dynamic value.

6by9, 2016-09-07:
Hmm, requesting two slices per frame does appear to spit the two NALs out in one buffer, with start codes at the start of each slice. I can think of at least one bit of code that may be dubious with that. Oh well.

JamesH65, 2017-12-21:
@6by9 Status? Closable?

6by9, 2017-12-21:
Certainly no intent of changing OMX_IndexConfigBrcmPortStats.

As to my comment over "two NALs out in one buffer, with start codes at the start of each slice", that's almost certainly the minimise fragmentation option being a touch over eager. Turn that off and I expect it'd go away, so again no real need to change anything.

Close away.

closed 2017-12-21 by JamesH65

F5OEO, 2017-12-21:
"Certainly no intent of changing OMX_IndexConfigBrcmPortStats" this is an easy solution. Issue is still there but @6by9 doesn't want to understand my explanations. Let it closed, you are the boss !

6by9, 2017-12-21:
Your request may still stand, and I understand what you're after, but there's also the possibility that others are using it as-is and any change will break their use case.

It's a proprietary extension that does what it does.